jeremiah-c-leary/vhdl-style-guide #161: Support blocks

ryanfitzsimon, 2018-09-03:
Block statements are not handled correctly.
The attached file apparently violates architecture_010 on line 8, but shouldn't.

[block.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/2344443/block.txt)

jeremiah-c-leary, 2018-09-12:
I'm guessing it thinks "end block BLK;" is the end of the architecture and throws the architecture_010 violation.

jeremiah-c-leary, 2018-09-13:
I added an initial check for blocks.  Please re-test and let me know if this is sufficient.

ryanfitzsimon, 2018-09-13:
Thanks Jeremiah, that looks like a great start. It seems to be working well against most of the code I'm running it against, but it has issues with multi-line constants that are defined within blocks.

I've just opened PR #165 to address this.

jeremiah-c-leary, 2018-09-13:
Hey Ryan,
I merged in PR #165.  I have not added any rules against blocks, but it looks like have several already I can re-purpose from existing rules.  I will update the sphinx documentation to show what rules I believe should be applied to blocks.  I would appreciate any feedback before I implement anything.

ryanfitzsimon, 2018-09-14:
Like I said, the changes you have already made appear to be enough to support blocks in the code I'm running vsg on.

That said, I can think of a few cases this won't cover.

Instantiations inside blocks won't be recognized correctly (see https://github.com/jeremiah-c-leary/vhdl-style-guide/commit/510ba10e8cb32a07c0a02ecc7a49fcefade8e0b1#r30500956).

The other case I can see being problematic is nested blocks. I expect this will be much trickier to support,
and should probably be tackled separately.

I'm sure there will be more edge cases, but for now I'm quite happy with this level of support.

jeremiah-c-leary, 2018-09-14:
I added a fix for instantiations inside of blocks.

As for nested blocks, I have the same issue with nested loops (issue #134 ).  It might require a different approach, or a method to keep track of nesting level.  hmmm...maybe that could be done with the variable dictionary I pass around when first processing a file.  The nesting is really just an indenting issue...I think.

I will try to find some examples of blocks around the internet to process and eventually create some rules.

Again, thanks for the input.  I also appreciate the minor enhancements you have made.  It has increased my knowledge of python.

Also, let me know if/when we can close this issue.  We can always open more.