jeremiah-c-leary/vhdl-style-guide #180: comment alignment in instantiations

csomers82, 2018-11-20:
If instantiation_023 is disabled, then comments in instantiations are ignored, rather then flagged us unaligned. Is this an issue, or more-so an opportunity for a local ruleset?

jeremiah-c-leary, 2018-11-21:
So **instantiation_023** and **component_019** check for and remove comments in instantiations and component declarations.  If these are disabled there is no check for comment alignment.

So there are two options:
  1)  Add a local rule based on **entity_018** for components and instantiations
  2)  Add two new rules to vsg

I could add the new rules and default them to *disabled*, but should I?  I am leaning towards adding one, but you could try out a locally modified entity_018 rule.

Let me know what you think about addiing it to the base rule set.

Is there a corollary rule to this?  All instantiation ports must have a comment.

csomers82, 2018-11-26:
I will try the a local rule 'instantiation_029'. 
If disabled rules still affect total run time, then perhaps a disabled by default rule is not favorable. Else, providing pre-made alternate rules would be a welcome advancement.

csomers82, 2018-11-26:
Considering a start to a new conversation, would a corollary be a new rule, or a property of the same rule?
I would think the latter, but understand that code complexity may be a challenge in implementation.

csomers82, 2018-11-26:
>2. Add two new rules to vsg
I added a local rule and to worked beautifully.
I only needed to add a single local rule however, just for the instantiations. Is this a bug?

jeremiah-c-leary, 2018-11-27:
so a single local rule for instantiations fixed components also?

I will work on adding the rule and see if I get the same behavior.

jeremiah-c-leary, 2018-11-27:
I added component_020 to detect and fix comments at the end of component declarations.

jeremiah-c-leary, 2018-11-27:
I just release version 0.25 with the two new rules.  Please let me know if these address the issue and if the issue can be closed.

csomers82, 2018-11-29:
Yes, the issue is resolved. Thank you.