jeremiah-c-leary/vhdl-style-guide #181: provision for coarse, in-file-granularity VSG enable/disable

csomers82, 2018-11-26:
It would be convenient if there was a way to hide passages of code from VSG, like a comment switch to turn off VSG function until another comment switch (to turn VSG back on) is read.  

I am finding a particular whitespace rule to have some complex considerations and am not sure whether the rules in place can be navigated. The source code in the section of interest is already formatted as desired, but the rest of the file is not.

jeremiah-c-leary, 2018-11-27:
I just want to document what we discussed on this issue.

  1)  We decided adding a pragma to the code is not desirable as we would be polluting the code with unnecessary comments.

  2)  We could add a per line exclusion, but that is a brittle solution.

  3)  I think we decided to leave this open until you can find a good example where this would be required.

I will mark it as deferred and an enhancement.

Too bad there is not an up vote on issues where people can decide which features/issues get resolved.

jeremiah-c-leary, 2018-12-08:
So have we decided we do not want this feature?

csomers82, 2019-03-05:
My vote is yes: a pragma feature being the option I support. 

In the same way vsg unites designers under a style of writing vhdl, pragmas/exceptions in the code draw attention to rules that designers do not agree on (grounds for configuration) and also to rules that do not capture the desired presentation (grounds for review of vsg rules themselves).

jeremiah-c-leary, 2019-03-05:
capturing example:

[image]

jeremiah-c-leary, 2019-03-10:
This functionality has been added.  Review commit logs for details.  There is also updated documentation until the heading "Code Tags".

closed 2019-03-10 by jeremiah-c-leary