jeremiah-c-leary/vhdl-style-guide #183: Processes lacking a sensitivity list clash with process rules

csomers82, 2018-11-29:
Although testbench code may be creeping out of scope, there is a compile conflict with the process rules. If no signals are present in a vhdl sensitivity list, then there must not be any parenthesis or IS keyword, else it will not compile.
...
MAIN : process
  variable var_a;
  variable var_b;
 begin
...
process_012 trips the violation and can be disabled; in doing process_002 trips, and in this case, variable_001 since the spacing in no longer correct.

jeremiah-c-leary, 2018-11-30:
they key to this is how to detect a process without a sensitivity list.

jeremiah-c-leary, 2018-12-01:
Release 0.26 has been pushed.  Please update and let me know if the issue has been resolved.

csomers82, 2018-12-03:
Process issue is resolved.

closed 2018-12-03 by csomers82