jeremiah-c-leary/vhdl-style-guide #186: 'end' in process identifiers miss-triggers rule architecture_010

csomers82, 2018-12-03:
----------------------------------------------------------------------------------------------
-- end and start pointer control
-- in M-code it is possible to have a start flag in 2 consecutive packets at startup
-- the sequencing must be controlled to enable correct decoding
-- create a 2 word fifo for the start pointer.
start_ptr_fifo:process(
    FIFO_START_PTR_WR, FIFO_START_PTR, q_fifo_start_ptr, q_fifo_wptr,
    RESTART, q_fifo_start_ptr_res, q_fifo_wptr_res, q_restart_mode
  )
  begin
    n_fifo_start_ptr        <= q_fifo_start_ptr;
    n_fifo_wptr             <= q_fifo_wptr;
    
    if (RESTART = '1') then                             -- restore the pointer
      n_fifo_start_ptr  <= q_fifo_start_ptr_res;
      n_fifo_wptr       <= q_fifo_wptr_res;

    elsif (q_restart_mode = '0') and (FIFO_START_PTR_WR = '1') then
      if (q_fifo_wptr = '0') then
        n_fifo_start_ptr(0) <= FIFO_START_PTR;
      else
        n_fifo_start_ptr(1) <= FIFO_START_PTR;
      end if;
      n_fifo_wptr           <= not q_fifo_wptr;  -- toggle between 2 locations
    end if;
  end process start_ptr_fifo;

  -- capture the end pointer
end_ptr_fifo:process(
    FIFO_END_PTR_WR, FIFO_END_PTR, q_fifo_end_ptr,
    RESTART, q_fifo_end_ptr_res, q_restart_mode
  )
  begin
    n_fifo_end_ptr          <= q_fifo_end_ptr;
    
    if (RESTART = '1') then                             -- restore the pointer
      n_fifo_end_ptr  <= q_fifo_end_ptr_res;

    elsif (q_restart_mode = '0') and (FIFO_END_PTR_WR = '1') then
      n_fifo_end_ptr <= FIFO_END_PTR;
    end if;
  end process end_ptr_fifo;

jeremiah-c-leary, 2018-12-04:
Issue has been resolved, but I have not pushed a new version.

jeremiah-c-leary, 2018-12-08:
Closing issue.  If this is still a problem we can re-open it.

closed 2018-12-08 by jeremiah-c-leary