jeremiah-c-leary/vhdl-style-guide #187: variable_009: Intentional to architecture instead of process?

stroebjo, 2018-12-11:
I was wondering if it was intentional that rule `variable_009` checks for the colon alignment along the whole architecture and not each process individual?

I have an architecture with two processes and one has a very long variable name, and the other quite short, so that would be a lot of whitespace in the one with short names.

jeremiah-c-leary, 2018-12-12:
Greetings stroebjo,

The intent was to align the colons for groups of variables.  Variables defined in one process should not affect the alignment in another process.  However, looking at the code it looks like it is aligning everything in the architecture.

I will update the rule to perform on a per process basis.

Thanks for finding this.

jeremiah-c-leary, 2018-12-12:
I restricted variable_009 to only operate in the architecture declarative region.
I then added variable_011 to operate within process declarative regions.
Each process declarative region is considered independent from the others, so alignments will be localized to each process.

I have one more issue to resolve and will release version 0.28 with this fix.

I will close this for now, but if this does not fully address your issue, please re-open it.

closed 2018-12-12 by jeremiah-c-leary

jeremiah-c-leary, 2018-12-12:
Hey stroebjo,

I released version 0.28.  Please let me know if the update I made is acceptable.

stroebjo, 2018-12-12:
Hey, thanks for the quick fix! This is perfect! :)