jeremiah-c-leary/vhdl-style-guide #191: Indentation rules when case labels are applied.

csomers82, 2018-12-12:
TESTSWITCH : process (      q_primary, q_secondary, 
                  SWSEL, TESTSEL, q_count) is
  begin
    case (SWSEL) is
      when B"00" =>
        n_muxout <= q_primary;    
      when B"01" =>
        n_muxout <= q_secondary;    
      when others =>
        TESTMUX : case (TESTSEL) is
          when B"00" =>
            n_muxout <= std_logic_vector(q_count);    
          when B"01" =>
            n_muxout <= TESTA;    
          when B"10" =>
            n_muxout <= TESTB;    
          when B"11" =>
            n_muxout <= TESTC;    
          when others =>
            null;
        end case TESTMUX;
    end case;
  end process TESTSWITCH;

csomers82, 2018-12-12:
Or, an equally effective solution: 
have a rule to enforce removal of case labels;

however, this is internally inconsistent. Every other rule in vsg is additive by nature.
Would the rule to remove case labels be the corollary to a rule?

jeremiah-c-leary, 2018-12-12:
VSG probably does not recognize it as a case statement.
I will look into it

jeremiah-c-leary, 2018-12-12:
I normally do not use 

> Or, an equally effective solution:
> have a rule to enforce removal of case labels;
> 
> however, this is internally inconsistent. Every other rule in vsg is additive by nature.
> Would the rule to remove case labels be the corollary to a rule?

so I normally do not use labels in front of case statements as I find a comment to be a better solution.  However, you bring up a good question.

Maybe there should be rule to remove case labels, not sure if this should be the default.
There could be another rule to enforce the use of case labels.
If case lables are enforced then the ending case label should be enforced.

...or...

have a rule that just enforces the end case label if the case label is given in the declaration.

Although the purpose of VSG is to make things consistent, so I'm not too sure where to go with this at the moment.

jeremiah-c-leary, 2018-12-13:
I added case_019 and case_020 to remove labels on case statements.
If you want to keep the labels, then disable those rules.
The indenting should work.
I will be releasing version 0.29 shortly.

closed 2018-12-13 by jeremiah-c-leary