jeremiah-c-leary/vhdl-style-guide #192: Test case for process alignment

csomers82, 2018-12-13:
```
 CLK_SYNC : process (CLK) is

    variable init_flag    : boolean := false;
    variable l            : line;
    variable good         : boolean;
    file init_data    : text;
    variable file_status  : file_open_status;
    variable bin_addr     : bit_vector(log2(G_DEPTH) - 1 downto 0);
    variable junk         : string(3 downto 1);
    variable bin_data     : bit_vector(G_WIDTH - 1 downto 0);
    variable cnt          : integer := 0;

  begin

end process CLK_SYNC;
```

csomers82, 2018-12-13:
There is an oddity about the indentation of the fourth parameter line. 
keyword "file" has one space and has the correct number of indents if "init_data" was aligned with the other variables names, but the combination of them is incorrect.

Is correct option 1?:
```
 CLK_SYNC : process (CLK) is

    variable init_flag    : boolean := false;
    variable l            : line;
    variable good         : boolean;
    file     init_data    : text;
    variable file_status  : file_open_status;
    variable bin_addr     : bit_vector(log2(G_DEPTH) - 1 downto 0);
    variable junk         : string(3 downto 1);
    variable bin_data     : bit_vector(G_WIDTH - 1 downto 0);
    variable cnt          : integer := 0;

  begin

end process CLK_SYNC;
```
or actually two?
```
 CLK_SYNC : process (CLK) is

    variable init_flag    : boolean := false;
    variable l            : line;
    variable good         : boolean;
    file     init_data        : text;
    variable file_status  : file_open_status;
    variable bin_addr     : bit_vector(log2(G_DEPTH) - 1 downto 0);
    variable junk         : string(3 downto 1);
    variable bin_data     : bit_vector(G_WIDTH - 1 downto 0);
    variable cnt          : integer := 0;

  begin

end process CLK_SYNC;
```

csomers82, 2018-12-13:
Revised, the root issue seems to be that there is no file keyword indentation rule

jeremiah-c-leary, 2018-12-15:
I had an idea to allow the user to define the number of spaces after **signal**, **variable**, **constant**, **file**, etc...  This would allow the user to choose the alignment.  Although this would not align the colons across multiple keywords.

jeremiah-c-leary, 2018-12-18:
I will be releasing version 0.30 with this fix.

closed 2018-12-18 by jeremiah-c-leary