jeremiah-c-leary/vhdl-style-guide #194: How might one add a rule to limit the number of consecutive empty lines?

csomers82, 2019-01-10:
Is there a classification for empty lines that could be used to generate whitespace rules? 
I have noticed whitespace seem to creep into my vsg affected files and I am uncertain how it is ending up there. I would like to create a rule to limit the number of empty lines to an arbitrary number.

jeremiah-c-leary, 2019-01-11:
I would think that is possible.  The rule could be "no more than x blank lines".  Should we default it to 1 or 2.

csomers82, 2019-01-11:
My thoughts are that whitespace is usually intentional, so two is a safer default since it would preserve what was already placed into the file. I would imagine 1 be a cleaner look though.

jeremiah-c-leary, 2019-01-11:
I think I would prefer a single line.  Multiple lines seems excessive.  I have several rules that removes multiple blank lines above and below some keywords.

jeremiah-c-leary, 2019-02-06:
I added whitespace_012 to enforce the number of consecutive blank lines.  I set the default to 1, but it can be overridden by a configuration.  The documentation for whitespace_012 shows how to set a different value.

closed 2019-02-06 by jeremiah-c-leary