jeremiah-c-leary/vhdl-style-guide #199: Generate Rules example

csomers82, 2019-01-30:
`
  SLICE :for i in 0 to 7 generate

    U1 : EXAMPLE1
      port map (
        RST_F          => RST_F,         

        I_CLK          => CLK,       
        I_AXI_AWSEL    => exm_awsel(i),  
        I_AXI_BUS      => exm_wbus(i),   

        O_CLK          => M175_CLK,      
        O_REQ          => w_exmw_req(i), 
        O_ACK          => w_mux_wack(i), 
        O_BUS          => w_exmw_bus(i)  
      );

    V1 : EXAMPLE2
     
      port map (
        RST_F           => RST_F,        
        CLK             => CLK,     

        WREQ            => w_exmw_req(i),
        WACK            => w_mux_wack(i),
        WBUS            => w_exmw_bus(i),

        RREQ            => w_exmr_req(i),
        RACK            => w_mux_rack(i),
        RBUS            => w_exmr_bus(i),

        REQ             => REQ(i),       
        ACK             => ACK(i),       
        BUS             => BUS_ARRAY(i)  
      );

  end generate slice;
  
  `

csomers82, 2019-01-30:
Suggested to capitalize finishing label, enforce minimum spacing before for keyword.

jeremiah-c-leary, 2019-02-04:
finishing label should be capitalized - agree
single space around : - agree
The generate rules are a little sparse.  In fact, VSG will not check for an ending label at all.
I wonder in this case if the for_loop rules would apply to the generate?  Probably not as it is a for ... generate versus a for ... loop.