jeremiah-c-leary/vhdl-style-guide #214: classify "filename" generic as file statement

ozdilo, 2019-03-06:
First, thanks a lot for this great tool. I love it, it is very useful, easy to use, easy to modify, clearly organized and well documented.

But I have a problem with the code shown below

```
generic map (
  FILENAME => string'("out.txt")
)
```
This code block fails file and indentation rules.

I propose to change the 
`if re.match('^\s*file', oLine.line, re.IGNORECASE):` line in file_statement.py with

`if re.match('^\s*file\s+', oLine.line, re.IGNORECASE):`.

For only the given case it works, I did not test it extensively.

Regards,
Omer

jeremiah-c-leary, 2019-03-09:
Hey Omer,

Good catch.  I seem to have an issue with my classifications being very greedy.

I will put this in the next release.

Regards,

Jeremy

jeremiah-c-leary, 2019-03-09:
Issue has been resolved.  See commit comments for details.  This will be part of the 0.33 release.

closed 2019-03-09 by jeremiah-c-leary