jeremiah-c-leary/vhdl-style-guide #215: Unusual placement of "else" statement not moved

csomers82, 2019-03-06:
problem text

```
  i_bus_rdata <=                   (others => '0')            when  w_bus_sync_math_rd_en = '0'
                     else           obuf_data_out             when  w_bus_sync_math_addr(G_IOBUF_ADDR_BIT)   = '1'
                     else         x"0000" & q_instr           when  reg_sel(c_instr_reg) = '1'
                     else             mod_len_32              when  reg_sel(c_mod_len_reg) = '1'
                     else             exp_len_32              when  reg_sel(c_exp_len_reg) = '1'
                     else               status                when  reg_sel(c_status_reg) = '1'
                     else               control               when  reg_sel(c_ctl_reg) = '1'
                     else      x"00000" & "000" &  w_msq_ma   when  reg_sel(c_msq_ma_reg) = '1'
                     else             x"00070000"             when  reg_sel(c_version_reg) = '1'
                     else std_logic_vector(to_unsigned(G_MAX_MOD_LEN,i_bus_rdata'length))  when  reg_sel(c_max_mod_len_reg) = '1'
                     else std_logic_vector(to_unsigned(G_MAX_EXP_LEN,i_bus_rdata'length))  when  reg_sel(c_max_exp_len_reg) = '1'
                     else std_logic_vector(to_unsigned(G_NUM_PE,i_bus_rdata'length))       when  reg_sel(c_num_pe_reg) = '1'
                     else           (others => '0');
```

csomers82, 2019-03-06:
ideal outputs
 ```
 i_bus_rdata <= (others => '0')            when  w_bus_sync_math_rd_en = '0' else
                 obuf_data_out             when  w_bus_sync_math_addr(G_IOBUF_ADDR_BIT)   = '1' else
                 x"0000" & q_instr           when  reg_sel(c_instr_reg) = '1' else
                 mod_len_32              when  reg_sel(c_mod_len_reg) = '1' else
                 exp_len_32              when  reg_sel(c_exp_len_reg) = '1' else
                 status                when  reg_sel(c_status_reg) = '1' else
                 control               when  reg_sel(c_ctl_reg) = '1' else
                 x"00000" & "000" &  w_msq_ma   when  reg_sel(c_msq_ma_reg) = '1' else
                 x"00070000"             when  reg_sel(c_version_reg) = '1' else
                 std_logic_vector(to_unsigned(G_MAX_MOD_LEN, i_bus_rdata'length))  when  reg_sel(c_max_mod_len_reg) = '1' else
                 std_logic_vector(to_unsigned(G_MAX_EXP_LEN, i_bus_rdata'length))  when  reg_sel(c_max_exp_len_reg) = '1' else
                 std_logic_vector(to_unsigned(G_NUM_PE, i_bus_rdata'length))       when  reg_sel(c_num_pe_reg) = '1' else 
                 (others => '0');
```
or
```
  i_bus_rdata <= (others => '0')                                                   when  w_bus_sync_math_rd_en = '0' else
                 obuf_data_out                                                     when  w_bus_sync_math_addr(G_IOBUF_ADDR_BIT)   = '1' else
                 x"0000" & q_instr                                                 when  reg_sel(c_instr_reg) = '1' else
                 mod_len_32                                                        when  reg_sel(c_mod_len_reg) = '1' else
                 exp_len_32                                                        when  reg_sel(c_exp_len_reg) = '1' else
                 status                                                            when  reg_sel(c_status_reg) = '1' else
                 control                                                           when  reg_sel(c_ctl_reg) = '1' else
                 x"00000" & "000" &  w_msq_ma                                      when  reg_sel(c_msq_ma_reg) = '1' else
                 x"00070000"                                                       when  reg_sel(c_version_reg) = '1' else
                 std_logic_vector(to_unsigned(G_MAX_MOD_LEN, i_bus_rdata'length))  when  reg_sel(c_max_mod_len_reg) = '1' else
                 std_logic_vector(to_unsigned(G_MAX_EXP_LEN, i_bus_rdata'length))  when  reg_sel(c_max_exp_len_reg) = '1' else
                 std_logic_vector(to_unsigned(G_NUM_PE, i_bus_rdata'length))       when  reg_sel(c_num_pe_reg) = '1' else 
                 (others => '0');
```

jeremiah-c-leary, 2019-03-09:
I ran into the else on a separate line a little while ago in issue #208.  I was thinking of adding a rule to move the else up a line, but I was hoping that was an isolated case.

jeremiah-c-leary, 2019-03-09:
What do you think of adding a rule to align the whens.  Then use the implementation of issue #181 to turn it off and on.  The user can then decide when the rule applies, because sometimes is looks better one way than the other.

jeremiah-c-leary, 2019-03-09:
I added a rule to move the else statement to the end of the previous line.
Let me know what you would like to do with aligning the when statements.

jeremiah-c-leary, 2019-03-09:
If you would like a rule to align the when statements, please create another ticket.  I am going to close this one as the else keyword is now moved.

closed 2019-03-09 by jeremiah-c-leary