jeremiah-c-leary/vhdl-style-guide #220: Valid code not handled correctly

nickvanderstel, 2019-03-22:
Running vsg against this code:

```
entity TEST_E is
end entity TEST_E;

architecture TEST_A of TEST_E is

begin

  TEST_PROCESS : process
    procedure test_procedure (
      constant test1_c    : in boolean := true
    ) is
    begin
    end procedure test_procedure;

  begin
  end process TEST_PROCESS;

end architecture TEST_A;
```

Throws these violations:

```
Phase 1... Reporting
  process_010               |         13 | Place "begin" keyword on seperate line.
  process_012               |         13 | Add "is" keyword after the closing parenthesis.
```

And mangles the code - notice the extra `is`:

```
entity TEST_E is
end entity TEST_E;

architecture TEST_A of TEST_E is

begin

  TEST_PROCESS : process
    procedure test_procedure (
                              constant test1_c    : in boolean := true
                                                                  ) is
                                                                  is     begin

                                                                  end procedure test_procedure;

    begin

    end process TEST_PROCESS;

end architecture TEST_A;
```

nickvanderstel, 2019-03-22:
I'm guessing the default parameter value is the issue. None of the procedure tests include a default value for any parameters.

jeremiah-c-leary, 2019-03-22:
Greetings NIck,
I am guessing the VSG is thinking the first begin is part of the process declaration instead of the procedure declaration.  This should be easy to fix.

jeremiah-c-leary, 2019-03-23:
Well...that was a little more involved than I thought it would be.  My first hunch was correct, but there was also another issue.  The procedure parameter was being classified as a variable assignment, which was then throwing the indenting off.

This fix will be in the next release.

closed 2019-03-23 by jeremiah-c-leary