jeremiah-c-leary/vhdl-style-guide #224: sourcefile wide capitalization check?

TripRichert, 2019-03-24:
Could a rule be added to check a source file for inconsistent capitalization?

Maybe by going line-by-line, put each token in a dictionary, where all lower case (or all upper case) is the key and the actual text is the value? Any time the token is already in the dictionary, make sure it matches the value?

Currently, the code below passes VSG

    library ieee;
        use ieee.std_logic_1164.all;

    entity TEST is
        generic (
            EDGE : std_ulogic
        );
        port (
            CLK : in        std_ulogic;
            DAT : out     std_ulogic_vector(3 downto 0)
        );
    end entity TEST;

    architecture A of TEST is

        function getconstdata (x : integer) return std_ulogic_vector is
        begin
            return x"4";
        end function getconstdata;
    
    begin

        SYNC : process (CLK) is
        begin

            if (CLK'event and CLK = EDGE and CLK'last_value = not EDGE) then
                DAT <= getconstData(1);
            end if;

        end process SYNC;

    end architecture AB;

But, the getconstdata function is inconsistently capitalized.

jeremiah-c-leary, 2019-03-24:
Building on your idea, I could do the following:

  1)  take each line, replace :'s, commas, periods, ticks, parenthesis, double quotes, semicolons, etc.. with spaces
  2)  split the line on spaces
  3)  remove VHDL keywords from the list
  4)  remove numbers
  5)  run through everything in list and compare it to a master list
        a)  if a caseless match is found, then check if the word in the list matches the case in the master list.  Throw a violation if the case does not match.
        b)  if a caseless match is not found, then put the word into the master list.

I think something like that would work.

TripRichert, 2019-03-24:
Sounds great!

I didn't think about double quotes. Content in and adjacent to double quotes should be removed, I think. The three cases i can think of for double quotes are: a constant number expression (e.g. x"4"), a constant string expression, or a report statement (which I guess is a technically a constant string expression). None of those will contain symbol names requiring consistent capitalization.

jeremiah-c-leary, 2019-03-28:
I have the detection of inconsistent non vhdl keywords working.  I just need to add in fixing the inconsistencies.

jeremiah-c-leary, 2019-03-29:
I ran into a case where a file wide check does not work:

`
177       DOB                => BRAM_Din_B(31 downto 0),
178       ADDRA(14 downto 5) => addra,
179       ADDRA(4 downto 0)  => "00000",
`
This is in an instantiation and the port ADDRA is uppercase while the signal addra is lowercase.  The port is being flagged as the wrong case, because the signal is defined as lowercase:

`
 61   signal addra             : std_logic_vector(9 downto 0);
`

I will have to think about how to tackle this case.

TripRichert, 2019-03-30:
I didn't think of that. Named association can be used in subroutines as well, with the same problem.

jeremiah-c-leary, 2019-03-31:
I will have to split a generic rule into a series of specific rules.  For example, a rule to verify signal names are consistent.  Another for constants, and another that checks function names are used consistently.

jeremiah-c-leary, 2019-03-31:
I think I got everything covered for this.  I am going to release what I currently have.

closed 2019-03-31 by jeremiah-c-leary