jeremiah-c-leary/vhdl-style-guide #225: component keyword capitalization ignored

csomers82, 2019-04-02:
```
U_TB_CTRL : COMPONENT TB_CTRL
  generic map (
   ...
  )
  port map (
    ...
  );
```

to 

```
U_TB_CTRL : component TB_CTRL
  generic map (
   ...
  )
  port map (
    ...
  );
```

jeremiah-c-leary, 2019-04-03:
That is interesting.  I do not know the advantages of using component.

jeremiah-c-leary, 2019-04-03:
Should the rule be to remove the component keyword?  It seems useless.

jeremiah-c-leary, 2019-04-03:
So I added three rules:  instantiation_031, instantiation_032, and instantiation_033.  The first two will lowercase the component keyword and ensure a single space exists after the component keyword.  The last will remove the component keyword.  The first two are disabled by default.

closed 2019-04-03 by jeremiah-c-leary