jeremiah-c-leary/vhdl-style-guide #230: vsg_on / vsg off not disables local rule checking

measpear, 2019-05-16:
Hi,
i just updated to the latest version to use the vsg_on/off functionality. Whereas i use vsg_on/off, on an entire file, local rules are not excluded and messages are reported at the vsg output.

jeremiah-c-leary, 2019-05-17:
Hey measpear,

I changed the default rule class to add in the code tags.  There are a couple of options:  you can add the following code under the main for loop in the analyze method:

''' python
      if self._is_vsg_off(oLine):
           continue
'''
You can see this in the following rule:  https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/master/vsg/rules/port/rule_018.py

The second option is to update to the new rule class.  This could be as simple as changing your analyze to _analyze and removing the for loop.  It could also involve using the _pre_analyze method.

If you could post/send me your local rule I can show you what needs to be done.

measpear, 2019-05-17:
Thank you for the quick response. By changing the analyse method and deleting the for loop i fixed the local rules. Now vsg_on/off is working for my local rules also. 

As a suggestion, it will be helpful for new users to update the local rules section of the documentation for the new users

closed 2019-05-17 by measpear

jeremiah-c-leary, 2019-05-17:
I agree on updating the documentation.  I was also thinking of adding a template for a local rule.