jeremiah-c-leary/vhdl-style-guide #232: Typo in process_030 rule and objection to it's implementation

jredders, 2019-06-03:
First the typo:
Compact sensivity list to reduce the number of lines is uses.
Should be:
Compact >sensitivity< list to reduce the number of lines >it< uses.

Second, the objection:
If a sensitivity list ever puts a single signal on a line it gets flagged.  I like to group common sets of signals, say those for the same feature on multiple blocks, but some signals don't fit this, so they get their own line.

I think this rule should only fire once for a sensitivity list if every single item is on its own line, which is wasteful.  If there are multiple signals on some lines and single signals on others, there's probably a clarity reason they were ordered that way.

jeremiah-c-leary, 2019-06-04:
hmmm....I can see your point.  Plus a single signal should not trigger the check.

I wonder if this rule is really enforceable.  You could have two signals that are not grouped, and have each on it's own line.  This would trigger the check also.

I put it in for a specific reason, but it may have been a short sighted rule.  Maybe I should just remove it.

What do you think?

jredders, 2019-06-04:
Are there any rules that exist, but are disabled by default?  I can see the use for this for quickly cleaning up if someone put a single signal per line for a process with a huge number of inputs, but I don't think that's a common thing to run into.

jeremiah-c-leary, 2019-06-04:
There are some rules that are disabled by default.  I could default to disabling.  Although this rule just flags the error, it doesn't fix it.

jeremiah-c-leary, 2019-06-08:
Maybe there could be a consecutive threshold value.  If the threshold is exceeded then the rule will fire.  This would require configuring the rule, as I would probably default it to 0.  Although that isn't much difference than disabling the rule for the file or using a code tag.

jeremiah-c-leary, 2019-06-08:
I updated the rule to only fire if the violation documented is seen.  This should allow you to have a mix of single and multiples.  As long as they are not all singles.

closed 2019-06-08 by jeremiah-c-leary