jeremiah-c-leary/vhdl-style-guide #233: comment_007 should not flag comments before if/elsif keywords

jredders, 2019-06-03:
For the following example, comment_007 fires for the comment before the elsif when I don't think it should:

-- If first case is true
if (foo = '1') then
  do_stuff;
-- If second case is true
elsif (bar = '1') then
  do_other_stuff;
end if;

The comment goes with the elsif, not the do_stuff code within the first part of the if block, so it should be aligned with the elsif.

It looks like adding a blank line after do_stuff; prevents it from firing.

jeremiah-c-leary, 2019-06-04:
That makes sense.  So any consecutive comments above an elsif and else should align with the keywords.  This is similar to comments above when keywords in case statements.

jredders, 2019-06-04:
Yes, that sounds right.

closed 2019-06-08 by jeremiah-c-leary