jeremiah-c-leary/vhdl-style-guide #234: type_003 and type_006

jredders, 2019-06-04:
I personally think the following looks good, but VSG complains about multiple things:

```
constant c_my_constant  : positive := 7;
subtype  t_my_subtype  is std_logic_vector(c_my_constant downto 0);
type     t_my_type     is array(c_my_constant downto 0) of t_my_subtype;
```

```
Phase 2... Reporting
  type_003                  |        3 | Remove all but one space after the "type" keyword.
  type_006                  |        3 | Ensure a single space before the "is" keyword.
```

Notice it doesn't complain about the subtype at all though.

jeremiah-c-leary, 2019-06-05:
VSG is a little week on subtype enforcement.  There are only two rules.  I can create a local rule to align the **is** keyword in consecutive type/subtype statements.  Or I can add a rule and leave it off.  There are plenty of options.

I am going to split the type_006 and type_003 issues apart.  If I can figure out how to do it.

jeremiah-c-leary, 2019-06-09:
I created issue #241 to cover the type_003 problem.

jeremiah-c-leary, 2020-02-25:
closing after discussing this with @jredders .

closed 2020-02-25 by jeremiah-c-leary