jeremiah-c-leary/vhdl-style-guide #235: concurrent_003 is overzealous

jredders, 2019-06-04:
I think whens need a special case for this rule, since the (I_CRUFT...) stuff isn't the same as the '0' that should align with I_FOO.

I think assignments that have a '(' in them should align the text after the '(' so the 1 aligns with the 0 as shown in the second example.

```
  w_foo <= I_FOO when ((I_BAR   = '1') and
                       (I_CRUFT = '1')) else
           '0';

  O_FOO <= (1 => q_foo(63 downto 32),
            0 => q_foo(31 downto  0));
```

```
Phase 6... Reporting
  concurrent_003            |         17 | Align first character in row to the column of text one space after the <=.
  concurrent_003            |         21 | Align first character in row to the column of text one space after the <=.
```

jredders, 2019-06-04:
Here's another example the current rule doesn't like, that I think is fine as is:

```
n_foo <= resize(unsigned(I_FOO) +
                unsigned(I_BAR), q_foo'length);
```

jredders, 2019-06-04:
I think all of these actually boil down to the second example if you exclude what's in front of the '('.

jeremiah-c-leary, 2019-06-05:
Should there be a different rule that pulls everything onto a single line?

``` vhdl
n_foo <= resize(unsigned(I_FOO) + unsigned(I_BAR), q_foo'length);
```

You could run into line length problems, but maybe that is an indication the concurrent assignment should be refactored.  Maybe it should be shortened.

closed 2019-12-04 by jeremiah-c-leary