jeremiah-c-leary/vhdl-style-guide #266: Problem when printing solution for case rules.

m-kru, 2019-10-14:
Generally `case` can be set for rules in json, and it works. However when the rule is violated the printed solution always recommends changing the case to default. For example:
In config I set:
```
constant_004:
    case: 'upper'
```
With the following line, the rule is violated:
 `constant c         : integer := 10;`
However the fix suggestion is:
`  constant_004              |         67 | Change constant identifiers name to lowercase`
while it should be:
`  constant_004              |         67 | Change constant identifiers name to uppercase`

I do not know what might be wrong. This is the code of the rule:
```
from vsg.rules import case_rule
from vsg import utils

class rule_004(case_rule):
    '''
    Constant rule 004 checks the constant names have proper case.
    '''

    def __init__(self):
        case_rule.__init__(self, 'constant', '004', 'isConstant')
        self.solution = 'Change constant identifiers name to ' + self.case + 'case'

    def _extract(self, oLine):
        return utils.extract_class_identifier_list(oLine)
```

@jeremiah-c-leary do you have any idea?

jeremiah-c-leary, 2019-10-14:
Funny, I think I just ran into something very similar when I create the rule architecture_025.
I believe the issue is self.solution is set when the rule object is created.
Then the configuration comes through and sets the self.case, but self.solution has already been set.

I think there are two ways to address this:
  1)  follow the example in architecture_025:  https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/master/vsg/rules/architecture/rule_025.py
  2)  There is a _get_solution method on the base rule class.

```python
    def _get_solution(self, iLineNumber):
        '''
        By default this method return self.solution.
        This method can be overloaded by a rule if a more complex solution output is required.
        '''
        return self.solution
```
  I forgot about it when I made architecture_025, but you should be able to override it in the case rule class.

```python
    def _get_solution(self, iLineNumber):
        '''
        By default this method return self.solution.
        This method can be overloaded by a rule if a more complex solution output is required.
        '''
        return 'Change constant identifiers name to ' + self.case + 'case'
```
The _get_solution is called by the report_violations method on the base rule class.

The second option seems cleaner than what I did for architecture_025.

m-kru, 2019-10-15:
I wanted to do some more debugging, as I think the problem lies deeper. I want to enable only single rule for debug, however following configuration checks 0 rules:
```
rule:
    global:
        disable: true
        indentSize: 2
    # Lowercase port name
    port_010:
        disable: false
        case: 'lower'
```

Output:
```
File:  sts_bit_sequence_detector.vhd
====================================
Phase 1... Not executed
Phase 2... Not executed
Phase 3... Not executed
Phase 4... Not executed
Phase 5... Not executed
Phase 6... Not executed
Phase 7... Not executed
====================================
Total Rules Checked: 0
Total Violations:    0
```

Any idea how to fix this first :)?

jeremiah-c-leary, 2019-10-15:
well...that is interesting.  I will investigate.

m-kru, 2019-10-15:
Ok, you were right. I like the `_get_solution` solution, I will implement a fix.

m-kru, 2019-10-15:
I have created PR for this.

closed 2019-10-15 by m-kru