jeremiah-c-leary/vhdl-style-guide #268: --configuration error reporting

mdennis070, 2019-10-14:
**More Descriptive Error Messages**
When the path pointing to the .json file is wrong the error “error in configuration file <filename>” is printed. This feature request would add a more descriptive error of the problem. “error configuration file <filename> does not exist” 

The same error message is printed when there is a comma missing in the .json file. A more descriptive error message would be "expecting , at line 12"

**Rule in Config not Used by VSG**
Currently if a rule is not recognized it will be ignored. This feature request would add a warning that is printed to show that VSG is not using all of the settings given in the config file. An example would be "Warning: rule at line 12 not in ruleset. VSG will ignore this rule"

jeremiah-c-leary, 2019-10-17:
The first part would be easy...the second part not so much.  One problem I have is I support both json and yaml files and I do not know how to tell the difference.  So I attempt to read it with json and if it fails I assume it must be a yaml file.  If that fails then I say there is an error.
I will have to think about how to handle the second part of this enhancement.

jeremiah-c-leary, 2019-10-18:
So I was able to address the **More Descriptive Error Messages** issue.
Still working on the **Rule in Config not Used by VSG** issue.

closed 2019-10-19 by jeremiah-c-leary