jeremiah-c-leary/vhdl-style-guide #269: singal_008 for other keywords

mdennis070, 2019-10-14:
signal_008 allows for a prefix to be enforced for signal names. This feature request would add this ability for generic, constant, type, subtype, variable, and component names

jeremiah-c-leary, 2019-10-15:
I agree, this would be a good idea.

m-kru, 2019-10-15:
I am working on this. Will create PR soon.

m-kru, 2019-10-15:
@jeremiah-c-leary I have encountered one issue, that we have already faced today :).

With this configuration:
```
rule:
    global:
        disable: true
    generic_020:
       prefixes: ['G__']
    signal_008:
        prefixes: ['s_']
```
 No checks are executed, what is not compliant with documentation that says: 

> Note: Prefixes are disabled by default. To enable prefixes, set the prefixes attribute on the rule.

To enable these checks I need to enable them explicitly:
```
rule:
    global:
        disable: true
    generic_020:
        disable: false
        prefixes: ['G__']
    signal_008:
        disable: false
        prefixes: ['s_']
```

It can be fixed in 2 ways.

1. The first one is to make them behave like is said in the documentation.
2. The second one is to treat all prefix rules like all other rules, they need to be enabled explicitly.

I prefer the second option, plus we provide some default prefixes like `G_` for generic, `s_` for signals, `v_` for variables and `C_` for constants.

jeremiah-c-leary, 2019-10-16:
I agree with the second option.  As you said, it makes it consistent with all the other rules and allows us to define a default value.

m-kru, 2019-10-17:
@jeremiah-c-leary I think it can be closed.

jeremiah-c-leary, 2019-10-18:
Agree.

Thank you again for implementing this feature.  I learned something new also.  I never knew there was an else option on a for loop.

closed 2019-10-18 by jeremiah-c-leary