jeremiah-c-leary/vhdl-style-guide #270: Running a phase that is specified by command line argument

mdennis070, 2019-10-14:
This feature request would add the ability to run a single phase. Currently there is only the option to run all phases and up to a phase. 

This feature request would also add the ability to have VSG report on all phases instead of stopping at the first phase where violations are found.

jeremiah-c-leary, 2019-10-16:
A lot of rules in later phases are making assumptions about the code that previous phases have addressed.  So while there should not be anything to prevent someone from running the phases out of order, the rules may not work the same way.

Can you elaborate on why you would want to run a phase without running the previous phases?

mdennis070, 2019-10-21:
I thought this would be a good feature to have when using --fix but go back and make adjustments to the formatting of the code. If you run a specific phase then --fix wont go back and change that rule. A way to not have this problem would be to disable the rule after the adjustment is made. Thinking more about it I am not sure it is worth it to add this feature.

jeremiah-c-leary, 2020-02-25:
@mdennis070 Is this a feature you would still want?

mdennis070, 2020-03-07:
no

closed 2020-03-08 by jeremiah-c-leary