jeremiah-c-leary/vhdl-style-guide #273: module instantiation has incorrect indentation when inside generate block

mdennis070, 2019-10-14:
**Original Code**
```vhdl
GEN_LABEL :
   for I in 0 to c_const1-1 generate
   begin
   
      MODULE1_1 : MODULE1
            generic map (G_GENERIC1 => c_const2,
                         G_GENERIC2 => c_const3)
            port map (PORT1 => SIGNAL1,
                      PORT2 => SIGNAL2
                     );
      
      MODULE2_1 : MODULE2
            port map(PORT1 => SIGNAL3,
                     PORT2 => SIGNAL4
                    );      
   
   end generate; 
```

**"Fixed"**
```vhdl
GEN_LABEL : for I in 0 to c_const1 - 1 generate
begin

   MODULE1_1 : MODULE1
      generic map (
         G_GENERIC1 => c_const2,
         G_GENERIC2 => c_const3
      )
         port map (
            PORT1 => SIGNAL1,
            PORT2 => SIGNAL2
         );

      MODULE2_1 : MODULE2
         port map (
            PORT1 => SIGNAL1,
            PORT2 => SIGNAL2
         );

   end generate GEN_LABEL; 
```

**Correct Formatting**
```vhdl
GEN_LABEL : for I in 0 to c_const1 - 1 generate
begin

   MODULE1_1 : MODULE1
      generic map (
         G_GENERIC1 => c_const2,
         G_GENERIC2 => c_const3
      )
      port map (
         PORT1 => SIGNAL1,
         PORT2 => SIGNAL2
      );

   MODULE2_1 : MODULE2
      port map (
         PORT1 => SIGNAL1,
         PORT2 => SIGNAL2
      );

end generate GEN_LABEL; 
```

mdennis070, 2019-10-14:
After more testing trying to reproduce the error it appears that the formatting issue is caused by issue #272. If the open parenthesis is closed or removed VSG preforms correct formatting. In the above comment issue  #272 would be in place of `c_const3`. 

Incorrect Indentation
```vhdl
G_GENERIC2 => (2 * c_const2
```

Correct Indentation 
```vhdl
G_GENERIC2 => (2 * c_const2)
G_GENERIC2 => c_const2
```

jeremiah-c-leary, 2019-10-15:
k, I will focus on issue #272.

closed 2019-10-16 by jeremiah-c-leary