jeremiah-c-leary/vhdl-style-guide #282: Having "signal a" on one line and ": STD_LOGIC" on the next causes vsg to break down.

ChrisDBrown5, 2019-10-17:
Example code shown below. 

signal a,
         b, c,
         d,
         e,
         f,
         g,
         h, i,
          j, k, l, m, n, o, 
         p
          : STD_LOGIC;

signal xyz
          : STD_LOGIC_VECTOR( 0 downto 0);

Most optimal(?) fix might be...

**signal a     : std_logic;
signal b     : std_logic;
....
signal p     : std_logic;

signal xyz  : std_logic_vector( 0 downto 0);**

But that would change all code with instances like...

**signal n_data, q_data : std_logic_vector**

Although, that's not really that bad of a change.

Other fix as this might make the line length way too long, but might be acceptable as well...
**signal a, b, c, d, e, f, g, h, i, j, k, l, m, n, o, p : std_logic**

Actual error in code where "_path_" is your installed location:

Traceback (most recent call last):
  File "_path_/python/bin/vsg", line 10, in <module>
    sys.exit(main())
  File "_path_/python/lib/python2.7/site-packages/vsg/__main__.py", line 174, in main
    oRules.fix(commandLineArguments.fix_phase)
  File "_path_/python/lib/python2.7/site-packages/vsg/rule_list.py", line 135, in fix
    oRule.fix(self.oVhdlFile)
  File "_path_/python/lib/python2.7/site-packages/vsg/rule.py", line 69, in fix
    self.analyze(oFile)
  File "_path_/python/lib/python2.7/site-packages/vsg/rule.py", line 89, in analyze
    self._analyze(oFile, oLine, iLineNumber)
  File "_path_/python/lib/python2.7/site-packages/vsg/rules/case_rule.py", line 48, in _analyze
    words = self._extract(oLine)
  File "_path_/python/lib/python2.7/site-packages/vsg/rules/signal/rule_011.py", line 16, in _extract
    return utils.extract_type_name(oLine)
  File "_path_/python/lib/python2.7/site-packages/vsg/utils.py", line 596, in extract_type_name
    sLine = sLine.split(':')[1]
IndexError: list index out of range

jeremiah-c-leary, 2019-10-18:
we could put a limit of two signal definitions in a signal declaration.

jeremiah-c-leary, 2019-11-10:
I added two rules to cover this issue:
signal_016 ensures all signal declarations are on a single line.
signal_015 allows the user to define how many signals can be declared in a signal declaration.  The default is 2.

closed 2019-11-10 by jeremiah-c-leary