jeremiah-c-leary/vhdl-style-guide #285: -oc optione does not work

njm295, 2019-10-21:
It looks like the -oc option has some issues:

vsg -f test.vhd  -oc Rules
File:  test.vhd
===============
Phase 1... Reporting
  signal_007                |         14 | Remove default assignment.
Phase 2... Not executed
Phase 3... Not executed
Phase 4... Not executed
Phase 5... Not executed
Phase 6... Not executed
Phase 7... Not executed
===============
Total Rules Checked: 59
Total Violations:    1
Traceback (most recent call last):
  File "/usr/bin/vsg", line 9, in <module>
    load_entry_point('vsg==0.42', 'console_scripts', 'vsg')()
  File "/usr/lib/python2.7/site-packages/vsg-0.42-py2.7.egg/vsg/__main__.py", line 196, in main
    dOutputConfiguration['rule'] = oRules.get_configuration()
  File "/usr/lib/python2.7/site-packages/vsg-0.42-py2.7.egg/vsg/rule_list.py", line 233, in get_configuration
    dConfiguration[sId] = oRule.get_configuration()
  File "/usr/lib/python2.7/site-packages/vsg-0.42-py2.7.egg/vsg/rule.py", line 128, in get_configuration
    dConfig[sParameter] = getattr(self, sParameter)
AttributeError: rule_012 instance has no attribute 'spaces'

jeremiah-c-leary, 2019-10-24:
yikes...I will take a look at it.

jeremiah-c-leary, 2019-10-24:
There were a couple of attributes where were missing or defined incorrectly.

closed 2019-10-24 by jeremiah-c-leary