jeremiah-c-leary/vhdl-style-guide #286: Excpetion on wrong amount of spaces after if-keyword

stroebjo, 2019-10-22:
**Environment**
VHDL Style Guide (VSG) version 0.42
Python 3.7.3

**Describe the bug**
If there is no space or more than one spaces after the `if` keyword VSG crashes during linting with the following traceback: 

`$ vsg -f test.vhd`  
```
Traceback (most recent call last):
  File "//anaconda3/bin/vsg", line 10, in <module>
    sys.exit(main())
  File "//anaconda3/lib/python3.7/site-packages/vsg/__main__.py", line 178, in main
    oRules.report_violations(commandLineArguments.output_format)
  File "//anaconda3/lib/python3.7/site-packages/vsg/rule_list.py", line 178, in report_violations
    iFailures += oRule.report_violations(iLineNumber, sOutputFormat, self.oVhdlFile.filename)
  File "//anaconda3/lib/python3.7/site-packages/vsg/rule.py", line 50, in report_violations
    sOutputString += self._get_solution(iLineNumber)
  File "//anaconda3/lib/python3.7/site-packages/vsg/rules/single_space_after_rule.py", line 54, in _get_solution
    return 'Ensure there are only ' + str(self.spaces) + ' space(s) after the "' + self.name + '" keyword.'
AttributeError: 'rule_003' object has no attribute 'spaces'
```

**To Reproduce**

Run the following file with `$ vsg -f test.vhd`:

``` 

architecture BEH of TEST is

begin

  COMPUTE : process (x) is
  begin

    if(r_v = 1) then
      -- do stuff
    end if;

  end process COMPUTE;

end architecture BEH;

```

**Expected behavior**
Missing space after `if` should be shown. No crash. ;)

**Additional context**

Telling VSG to fix the file with `$ vsg -f test.vhd --fix` works as expected.

jeremiah-c-leary, 2019-10-24:
hmm...my testing must be a little lax on that rule.

jeremiah-c-leary, 2019-10-24:
Good news, this is a duplicate of issue #285.  I have pushed a fix already.

closed 2019-10-24 by jeremiah-c-leary