jeremiah-c-leary/vhdl-style-guide #290: Multiple signal definitions with the same class and type.

m-kru, 2019-11-21:
I would like to be able to define multiple signals with the same class and type as follows:
```
signal a,
       b,
       c : std_logic;
```

Is there any rules configuration that enables this and checks for the proper alignment?

jeremiah-c-leary, 2019-11-21:
hmm...I just added signal_015 and signal_016, so I am not surprised those fired.  However, instantiation_024 should not have flagged anything.  This will have to be investigated.

As for allowing the above formatting, are you wanting to enforce the formatting or just not flag it if it occurs?

For example:

Violation
``` vhdl
  signal a, b, c : std_logic;

  signal x : std_logic;
  signal y : std_logic;
  signal z : std_logic;

```
Fix
```vhdl
  signal a,
            b,
            c : std_logic;

  signal x,
            y,
            z : std_logic;
```

We could take signal_016 as a starting point to enforce that type of formatting.

As a side note, that is in interesting way to define signals.  Each signal is on it's own line, which I like, but there is only one type for all the signals.  I would have to see a file fully defined that way, but I suspect it would be easier to read.

m-kru, 2019-11-22:
I want to enforce this style for multiple signal definitions. I want vsg to fire an error when it encounters something like this:
```vhdl
signal a, b, c : std_logic_vector(7 downto 0);
```

jeremiah-c-leary, 2019-11-27:
so...signal_015 will fire if a configurable number of signals are defined over any number of lines.

You would like a rule that fires if a configurable number of signals are defined over a single line.

Violation
```vhdl
  signal a, b, c : std_logic_vector(7 downto 0);
```
Fix
```vhdl
  signal a,
            b,
            c : std_logic_vector(7 downto 0);
```

m-kru, 2019-12-03:
I took a look at these rules and I am a bit overwhelmed by how rules that work on multiple lines are implemented. I would need considerable amount of time to implement what you have written. I also do not like the fact, that if 2 rules contradicts, than user still needs to manually disable one of them. 

One more remark. I have recently been trying to 'apply' vsg to some mature project, and it was not easy (actually I still face with some of the bugs/blockers). I was wondering if it would not be better if all rules were disabled by default? I am sure that if some new potential user tries vsg on mature project, he will feel lost. When I first tried vsg I was rather thinking what I would like to force in the formatting than rather what I would like to disable. Theoretically user can disable all rules in global (he can, can't he?), and then enable them one by one, but with this approach one might get the feeling that he does something wrong.

In my humble opinion vsg is a classical example of Pareto principle. You can achieve 80% of your goal with 20% effort, but when applying it to bigger project I got the feeling that 80% of my effort was to disable rules that I do not want or that are broken.

jeremiah-c-leary, 2019-12-04:
Hey m-kru,

I hear your frustration.

Maybe I need a "Getting Started" section that could describe some strategies.

One of my colleagues requested the -fp option so he could run phase1 and check the output, and then phase2 and check the output, etc...  He also requested the -b option so he could compare the fixed output to the original.

Another strategy could be one you suggested.  Disable everything using the global in a configuration file and then enable them one by one.  One issue is there are currently 380 rules.  It would be fairly tedious to enable each one by one.  That speaks to your assertion that the user could feel they are doing something wrong.

Maybe part of the problem you are having, besides rules that are broken, are rules that do not match the style for the mature code you applied it to.  VSG is currently setup to follow my style of coding and if you want to deviate then you have to disable rules, configure rules and/or write your own rules.  VSG allows you to do this, but it will take time to work through everything.

At my work we use a linting tool called spyglass.  The first time I ran the tool on our code we had thousands of violations.  We had to take time to weed out the ones we did not want to create a rule set everyone uses.

I use Codacy to check for linting and style checks.  I had style checks down to under 10, then I checked a week ago and now I have over 4000.  So something changed over at Codacy.  I suspect they enabled a new set of checkers.  Now i get violations on capitalizing class names (e.g. rule_011) and not capitalizing function parameters (e.g. oLine).

If you can think of a method or strategy that would make this easier for you please let me know.  I'm sure you are not the only person having trouble.  Others have probably tried out VSG and if it does not work they abandon it.  If it does then they will use it.  I would assume it comes down to the cost of making the tool work versus keeping the status quo.

jeremiah-c-leary, 2019-12-05:
So I was pondering this some more and I came to this conclusion:  Using VSG is easy, but configuring it can be difficult.  I started thinking of how to make configuring VSG easier and came up with an idea and wanted your feedback.

What if there was a GUI that would help you configure which rules to use.  I am envisioning the GUI would be split into three vertical frames.

The left most frame would display all the rules in a tree format.
The middle frame would be the file you want to transform.
The right frame would be the transformed file.

The user could then manipulate the rules on the left and see the results on the right frame.  The user would also have access to the configurable attributes (e.g. case) on the left side.  The right frame would update when those are changed.

When the user is done, there would be a drop down menu item that would save the configuration file.  Then the user can apply this configuration file to the rest of the VHDL files.

Conceptually it seems like it would be fairly easy to implement.

Do you think that would improve the user experience?

m-kru, 2019-12-05:
I was also pondering, and here are my thoughts.
I really appreciate that you are aware that: 

> VSG is currently setup to follow my style of coding and if you want to deviate then you have to disable rules, configure rules and/or write your own rules.

This is exactly, how I see it.

I think vsg as a project has following strengths from user perspective:

1. Easy to use and it utilizes user friendly formats for configuration (both json and yaml).
2. Very clear, auto generated documentation.

And here is its one drawback (from my point of view), which consists of several factors.

1. Initial configuration takes a lot of time.

-  A lot of rules are enabled by default. In my humble opinion too many. I get a feeling like vsg tries me to use some specific, very rigid style by default (in this case your style).
-  There are some formatting aspects that user would like to force, but it is impossible or it requires changing more than one rule, often because rules would contradict.
- There are still some bugs, but it is natural and there will be less and less with time.

About the GUI. I do not like this idea. In my opinion it does not solve any problems. You want to compare files before and after? Simply copy one of them and use meld. Will it ease the configuration process? I doubt, yaml is super easy, there is no verbosity, and you can have comments. Implementing GUI would also be a waste of human resources, there is still much to be done in vsg.

What do I propose? Let's reduce the number of rules enabled be default. Prepare some set of rules, that have value, no matter what style user prefers (I assume the user is sane). Then we should consider how to solve problem with contradicting rules and maybe rethink how to implement rules that handle multiple lines. Multi line rules are the one that I have the most problems with right now (both as a user and when I try to analyze the code). 

What do you think?

Btw. I will fix all remaining issues, for which the case is still fixed (not configurable).

jeremiah-c-leary, 2019-12-12:
I think you are suggesting that VSG should be lenient with applying rules out of the box and then allow the user to add rules as they see fit.

  1)  User runs VSG with minimal set of rules (e.g. indenting) and reviews the output
  2)  User reviews available rules
  3)  User enables desired rules
  4)  User runs VSG with new configuration and reviews the output.

Versus the current methodology:

  1)  User runs VSG with base (most) rules enabled and reviews the output
  2)  User reviews available rules
  3)  User disables rules which they do not want
  4)  User runs VSG with new configuration and reviews the output.

I am struggling with how to express my next thoughts, so if it seems rambling I apologize:

There are some people who care about formatting more than others.  I remember as I was searching for a VHDL style tool I ran across one post in which the OP just wanted something to format the code consistently.  I work with some people who have very strong opinions about how code should be formatted (usually senior developers) and others (usually junior developers) who do not.

Sometimes you don't know what you want until you see it.  This issue is a perfect example.  I had never thought to format signals in such a way.  I might adopt it, but I would want to see how it looks.

There does not appear to be a "standard" for VHDL code style similar to PEP8 for python.  When I started writing Python I found PEP8 to be helpful.  Being new to the language I assumed PEP8 is the "standard" for how Python code should be formatted.

So taking all that rambling into consideration...

Are there two "users" of VSG?  One user would be someone who configures it and the other is someone who runs it.  In my case I am the user who configures it and the other people in my group are the users who run it.

In your case you are the user who configures it and I am assuming other people in your group would be the users who run it.

I believe you are asserting VSG is currently focused on the user who runs it versus the user who needs to configure it.

I would agree with that assertion and I believe that issue should be addressed.

I also agree your approach makes sense, enabling as few rules as possible and making the person who is going to configure VSG enable the rules they want.

jeremiah-c-leary, 2020-07-01:
@m-kru,
I am going to start on this.  I think I will combine `signal_015` and `signal_016` into a single rule.

jeremiah-c-leary, 2022-10-17:
Hey @m-kru ,

Are you still interested in having this feature implemented?

If I do not hear back from you by December 31st, I will assume I can close this issue.

Thanks,

--Jeremy

closed 2022-12-30 by jeremiah-c-leary