jeremiah-c-leary/vhdl-style-guide #310: Remove extract_first_keyword function.

m-kru, 2019-12-28:
Why?

Its functionality is somehow redundant. It is used to fetch words, that are known in advance. For this purpose it is better to call `extract_words` function. I will work on this.

m-kru, 2019-12-28:
Ok, there is one problem. It would be nice to have less extract functions. However, there is a problem when the same keyword occurs also in the VHDL comment section after `--`. I have realized, that oLine passed to the analyze functions is always a line with comments.
I am not sure if it is worth to refactor this part of vsg. @jeremiah-c-leary what do you think?

m-kru, 2019-12-29:
Ok, I have one idea. `extract_keywords` function instead of `extract_words`. `extract_keywords` would ignore everything after `--`. I will check if this idea fits.

jeremiah-c-leary, 2019-12-30:
I would think all the extract functions should ignore anything in comments.  There is an existing line attribute `lineNoComment` that strips everything after the `--`.  You could change:

```vhdl
def extract_class_identifier_list(oLine):
    '''
    Returns a class identifiers list.
    Parameters:
       oLine: (line object)
    Returns: (list of strings)
    '''
    sLine = oLine.line.replace(';', '')
    sLine = sLine.split(':')[0]
    sLine = sLine.replace(',', ' ').split()
    if sLine[0].lower() in ['constant', 'variable', 'signal', 'file']:
        sLine = sLine[1:]

    return sLine
```

to 

```vhdl
def extract_class_identifier_list(oLine):
    '''
    Returns a class identifiers list.
    Parameters:
       oLine: (line object)
    Returns: (list of strings)
    '''
    sLine = oLine.lineNoComment.replace(';', '')
    sLine = sLine.split(':')[0]
    sLine = sLine.replace(',', ' ').split()
    if sLine[0].lower() in ['constant', 'variable', 'signal', 'file']:
        sLine = sLine[1:]

    return sLine
```

Would `extract_keywords` return a list of vhdl keywords?  Then if you want the first one you would just use the first in the list.  i

This might be a slight tangent, but I was wondering if we added a couple of attributes to the line object:  `tokens` and `keywords` would it help with the extraction functions.  These would be set when the line is initially read in and when the `update_line` method is called.  I am going to create an issue to explore this idea.

closed 2020-01-25 by m-kru