jeremiah-c-leary/vhdl-style-guide #314: Unify how labels are extracted.

m-kru, 2019-12-31:
Right now labels are extracted in different ways.
Probably there should be only one function `extract_label` capable of distinguishing if label is at the beginning or at the end.
Function with `extract_label` name already exists, however it only extracts labels that appear at the beginning. This function is also not used in every place it should be used.

jeremiah-c-leary, 2020-01-01:
Agree.

The following rules deal with labels:

- [ ] case_019
- [ ] case_020
- [ ] concurrent_005
- [ ] for_loop_003
- [ ] for_loop_004
- [ ] for_loop_005
- [ ] generate_002
- [ ] generate_005
- [ ] generate_011
- [ ] generate_012
- [ ] generate_013
- [ ] generate_015
- [ ] process_016
- [ ] process_017
- [ ] process_018
- [ ] process_019
- [ ] process_024
- [ ] process_032

This issue would cover the following tasks:

- [ ] modify `extract_label` to return the label from the beginning or ending of a line.
- [ ] modify rules above to use the `extract_label` function

Optional:

- [ ] Create a rule at the vsg/rules level to reduce duplication

jeremiah-c-leary, 2020-02-23:
@m-kru I am going to start working on this issue so I can fix a bug I found in issue #333.

m-kru, 2020-02-23:
Great!

jeremiah-c-leary, 2020-02-23:
@m-kru I pushed the final updates for this issue.  I will wait a bit until I close to ticket to give you a chance to see what I did.

m-kru, 2020-02-24:
I think it looks fine. By the way, if you work on some issue and you want me to take a look after the work is done, then it is better to work on separate branch and create pull request. It is much more convenient and easier to comment specific parts of the change.

jeremiah-c-leary, 2020-02-24:
Understand.  I struggled with whether to make a branch on this one.

closed 2020-02-24 by jeremiah-c-leary