jeremiah-c-leary/vhdl-style-guide #316: Make all case rules inherit from case_rule.

m-kru, 2020-01-03:
There are still some rules that check words case, that do not inherit from `case_rule`.
Examples: `architecture/rule_011`, `component/rule_012`.

ToDo: Find all such rules and make them inherit from `case_rule`.

m-kru, 2020-01-04:
@jeremiah-c-leary ranges rules have no trigger?

jeremiah-c-leary, 2020-01-04:
Yeah...they deal with the `downto` and `to` keywords and can show up in multiple places:

- signal declarations
- constant declarations
- port definitions
- port instantiations
- etc...

so there isn't a single trigger.  Unless we added an attribute to indicate a range is detected.

m-kru, 2020-01-04:
Do I understand correctly, that all lines are checked against them?

jeremiah-c-leary, 2020-01-07:
I just reviewed the code.  It checks every line.

closed 2020-01-25 by m-kru