jeremiah-c-leary/vhdl-style-guide #318: Automatically disable signal_007 (default assignment) for testbenches

nfrancque, 2020-01-04:
The rule to remove default assignments is good for catching design bugs, but for those of us who also use VHDL for verification default assignments are good practice.  Is it possible to just apply a regex to the filename and by default disable this rule?

i.e. 
`signal test_signal : integer := 0`;
should not be an error in file some_tb.vhd.

jeremiah-c-leary, 2020-01-04:
I would think this would be hard to keep consistent across multiple files.  There are package files that testbenches use.  They would also have to be named with _tb.

You could use a configuration for testbench files versus design files.

nfrancque, 2020-01-04:
It seems like that'd still be okay since the option remains available.  You would simply be swapping the default.  That style is how [Vunit](https://github.com/VUnit/vunit) picks up on testbenches, defaulting to tb_.* or .*_tb with an option to disable if necessary.

You may be right that it's unnecessarily greedy, though.  

The main reason I felt it might be good as a default is the goal is for VSG in its default state to recommend good practices, and any modifications more than likely are exceptions to your known good rules.  So, if by the tool's best guess this is a testbench, preassignment *is* best practice and it might even be desirable to recommend adding it to avoid annoying issues later on.  
That guess might be wrong, but it doesn't leave a user scratching their head as to why the tool doesn't want them to do something they know to be a good idea.

jeremiah-c-leary, 2020-01-07:
would it be file name or architecture name?

nfrancque, 2020-01-07:
That's an interesting question, I've never considered very closely what to
name architectures of testbenches.  Do you think that should be in the
style guide?

Looked at existing testbench, evidently I have called one entity
vunit_tb_master and architecture just called test.

Could be interesting to mandate something here, what do you think?

jeremiah-c-leary, 2020-01-07:
I have used `DV` before.  People have also used `BEH`, or '`TB`.

I am just not sure if VSG is the right place to enforce it.  It could be done there, but philosophically should it.

nfrancque, 2020-01-07:
Sure, that's a good point.  IMO checking the vunit-style .*_tb or tb_.* *entity* name isn't too restrictive.  

You're right that it opens up a whole other can of worms though.  For instance, you could evolve into something like pylint that makes sanity checks on the actual code beyond the style.  i.e. in design code could warn the user about a wait statement existing, but not in testbenches.  But, then you're basically spyglass.

Those types of things probably bring us back to the discussion on #312 with full-on language parsing.

In the end you're the owner here so up to you.  I can definitely see not wanting to go outside that scope, so I am okay either way!

kraigher, 2020-01-09:
In most modern FPGA it is recommended to avoid having reset and instead use initial values. FPGA reset all flip flops when reconfigured and adding a reset net to the design itself is just a waste of routing resources.

kraigher, 2020-01-09:
Also if the user is using a package manager like tsfpga or edalize it would know what files are simulation or synthesis resources and could set different options for them.

nfrancque, 2020-01-09:
> In most modern FPGA it is recommended to avoid having reset and instead use initial values. 

I haven't heard that recommendation before.  Can you provide a source?

Have read the Xilinx [whitepaper](https://www.xilinx.com/support/documentation/white_papers/wp272.pdf)  on minimizing what needs reset and leave the rest a don't care, but I have a hard time accepting no resets at all.  Especially for vendor neutral/future proof code, even if what you said is the case.

kraigher, 2020-01-09:
https://www.eetimes.com/how-do-i-reset-my-fpga/

Basically if you do not need to reset an individual subsystem independently from the rest of the system you can avoid reset and do global reset instead. You can still write your IP with a synchronous reset signal but if you use init values you can choose to connect the reset line to a constant inactive value to optimize it away in synthetis.

nfrancque, 2020-01-09:
Interesting, thanks for the information.

It looks like if you're exclusively planning on targeting Xilinx devices
this would save you some routing resources.  But to be vendor agnostic IMO
still a dangerous assumption.

https://stackoverflow.com/questions/6363130/is-there-a-reason-to-initialize-not-reset-signals-in-vhdl-and-verilog

Glad to learn something new.

jeremiah-c-leary, 2020-02-25:
@nfrancque I am going to close this issue.  If a similar request comes in then I will reconsider.

closed 2020-02-25 by jeremiah-c-leary