jeremiah-c-leary/vhdl-style-guide #319: Discussion over rising_edge

nfrancque, 2020-01-04:
I'd like to bring up a discussion on the usage of `rising_edge`.

Most modern style guides recommend using the standard procedure.

https://insights.sigasi.com/tech/clock-edge-detection/
https://forums.xilinx.com/t5/Welcome-Join/Difference-between-rising-edge-clk-and-if-clk-1-then/td-p/298387

Of particular benefit is that `rising_edge` explicitly checks for 0->1 transitions, whereas ``clk`event and clk='1'`` will trigger on any event that causes it to transition to 1.  Is that of some benefit I'm not aware of?

At the least I think it would be beneficial to add an option to invert the rule and convert all to `rising_edge`.  Then users can define which is their preferred implementation.

jeremiah-c-leary, 2020-01-04:
Hey Nick,

Like you said, the first article shows `rising_edge` explicitly checks for 0->1 transitions.  I suppose you could say all clocks will be valid.

> This condition will also trigger when the clk signal changes from 'U' to '1', or even from 'H' to '1'. These events have no meaning in physical hardware, but they are perfectly legal in simulated VHDL code. Hence, this coding style introduces a discrepancy between simulation and synthesized hardware.

At the moment I can't think of a case in which the clock would be floating or an 'X'.  Although if something is not setup correctly in the testbench and the clock is incorrectly stimulated, then perhaps the `rising_edge` could mask the problem.

I would also wonder if `rising_edge` is more work for the simulator.  It has two extra functions (To_X01) and an `AND`.  That seems more complex than the traditional style.  Maybe a performance test would show if there is any appreciable difference.

We can always add a parameter to switch back and forth.

nfrancque, 2020-01-04:
That's an interesting point that it could be a performance hit, hadn't considered that.  You would think the simulator vendors would have found a way to perform some optimization on it, but maybe not possible?  Who knows.

> Although if something is not setup correctly in the testbench and the clock is incorrectly stimulated, then perhaps the rising_edge could mask the problem.

I think it's the opposite, is it not?  In real life your flops will trigger on a 0 to 1 transition, so ``clk`event`` increases the chances of simulation/synthesis mismatches by not also checking for the same condition in sim.  

I agree though, probably shouldn't ever matter and would be boneheaded mistake to have a clock not be 0 or 1.

kraigher, 2020-01-09:
A lot of synchronous code for FPGA use `wait until rising_edge(clk)` at the start of the process.

closed 2020-02-22 by jeremiah-c-leary

birdybro, 2022-07-06:
I know this is closed, but modifying the behavior of the synthesized design should not be something a formatter does. I ran into this when testing out your formatter and was highly surprised that a formatter would alter the synthesizable logic. Should I open a separate issue to request that the default configuration avoid altering a potential deliberate design decision?

jeremiah-c-leary, 2022-07-07:
Hey @birdybro,

A separate issue would be appreciated.

--Jeremy