jeremiah-c-leary/vhdl-style-guide #322: Create gitter

nfrancque, 2020-01-09:
Converstion here is starting to pick up in issues.  Would be nice to have a place to talk.  Also add readme badge.

@jeremiah-c-leary I believe only the repo owner can create a linked gitter community.

jeremiah-c-leary, 2020-01-09:
k, I will figure it out.

m-kru, 2020-01-20:
@jeremiah-c-leary I do not think we need gitter. This is small project with clear goal in mind. In my humble opinion it would be better to keep everything in one place as issues on github.
Please close if you agree with me.

nfrancque, 2020-01-22:
Probably fair, guess that brief uptick was temporary.  

Don't see much harm in creating but if you guys think it's a waste I can close.

GlenNicholls, 2020-01-22:
I was pointed here by someone from the sublime VHDL extension. My vote is definitely yes to a Gitter channel as it avoids cluttering issues with questions about using the extension and how to integrate with other FOSS tools (as you will see with an issue I'm about to open). Gitter can also help with synchronizing this project with other popular FOSS VHDL projects that might find benefit from this project.

nfrancque, 2020-01-22:
@GlenNicholls side note - can you link to that discussion?

Would be interesting to see if we could swap out the VHDL mode beautify with VSG.

GlenNicholls, 2020-01-22:
@nfrancque The issue I was referring to that I opened is https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/328

jeremiah-c-leary, 2020-02-26:
so just for fun I created a gitter community:  https://gitter.im/vhdl-style-guide.  If it doesn't see any/much activity in the next month I will delete it.

I also attempted to add a badge, but i can't figure out how to get the badge image.  There is a link to it in the README.rst file though.

closed 2020-02-26 by jeremiah-c-leary