jeremiah-c-leary/vhdl-style-guide #330: signal rule 013 behavior

m-kru, 2020-01-25:
In my humble opinion the behavior of signal rule 013 is broken.
Let examine following entity using vsg:
```vhdl
library ieee;
  use ieee.std_logic_1164.all;

entity ent is
  port (
    i : in    std_logic;
    o : out   std_logic
  );
end entity ent;

architecture rtl of ent is

  signal a : std_logic;
  signal b : std_logic;

  signal a_long_name : std_logic;
  signal b_long_name : std_logic;

begin

  a <= i;
  b <= a;

  a_long_name <= a;
  b_long_name <= b;

  o <= a_long_name xor b_long_name;

end architecture rtl;
```
The output reports:`   signal_013                |      12-20 | Align colon with right most colon.`
I do not want to align declarations of signals `a` and `b` to signals `a_long_name` and `b_long_name`. This is why I have separated them with blank line. I think rule `signal 013` should work similar to the rule `concurrent 006`, namely it should operate on group of lines.

jeremiah-c-leary, 2020-01-31:
How about an option that allows you to align them based on groups?

m-kru, 2020-01-31:
We can, but I think we should then edit the base class as both `signal 013` and `concurrent 006` inherit from `keyword_alignment_rule`. It is actually funny that both rules inherit from the same class but from user perspective their behavior is different. I know this is because of the attributes values:
```python
        ''' signal rule 013 '''
        self.sStartGroupTrigger = 'isArchitectureKeyword'
        self.sEndGroupTrigger = 'isArchitectureBegin'
        self.sLineTrigger = 'isSignal'
```
```python
        ''' concurrent rule 006 '''
        self.sStartGroupTrigger = 'isConcurrentBegin'
        self.sEndGroupTrigger = 'insideConcurrent'
```

I expected these attributes for `concurrent rule 006` look more like:
```python
        ''' concurrent rule 006 '''
        self.sStartGroupTrigger = 'isArchitectureBegin'
        self.sEndGroupTrigger = 'isArchitectureEnd'
        self.sLineTrigger = 'isConcurrent'
```

jeremiah-c-leary, 2020-01-31:
Okay....

I was very confused on why ```concurrent_006``` was missing the ```self.sLineTrigger``` option.  It turns out there are two ```keyword_alignment_rule``` classes:
* ```vsg/rules/keyword_alignment_rule```
* ```vsg/rules/concurrent/keyword_alignment_rule```

Thinking back... in second keyword_alignment_rule I was looking for a negative.  So a line that is not part of a concurrent statement.  While the first keyword_alignment_rule is looking for a positive.

You can compare line 55 of the first base rule against line 49 of the second base rule.

Maybe we elevate the second keyword_alignment to a new base class that focuses on groups.  Then you can call it for the alignment of :'s in groups of lines.  However, this would result in a new rule to align :'s and conflict with the old rule.  So that doesn't seem desirable.

We could merge the two ```keyword_alignment_rule``` classes and remove the second one.

Your thoughts?

Interestingly, the following rules do not use either ```keyword_alignment_rule```:
* entity_017
* instantiation_010

m-kru, 2020-02-08:
I like the implementation with:
```
        self.sStartGroupTrigger = 'a'
        self.sEndGroupTrigger = 'b'
        self.sLineTrigger = 'c'
```
It feels more natural to me. I would make it a base class for keyword alignment rules.

I would also add one more option to this rule, configurable via config file, that would allow user to define behavior when blank line is encountered between for example signal declarations. When set or unset blink line would function as a separator for groups that should be aligned.

One more thing. I wold like this rule to work with following signal declarations: [Multiple signal definitions with the same class and type](https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/290) .

m-kru, 2020-02-12:
I have spent some time analyzing the code and I think I will be able to unify behavior of keyword alignment rules and use one base class for all of them. I have also discovered one weird thing.
Take a look at keyword_alignment function from `check`:
```python
def keyword_alignment(self, iLineNumber, sKeyword, lGroup):
    '''
    Checks keywords in a group of line objects are aligned in the same column.

    Parameters:

      self: (rule object)

      iLineNumber: (integer)

      sKeyword: (string)

      lGroup: (list of line objects)
    '''
    iKeywordAlignment = None
    iMaximumKeywordColumn = 0
    sViolationRange = str(iLineNumber) + '-' + str(iLineNumber + len(lGroup) - 1)

    self.dFix['violations'].setdefault(sViolationRange, {}).setdefault('line', {})

    for iIndex, oGroupLine in enumerate(lGroup):
        if sKeyword in oGroupLine.line:
            self.dFix['violations'][sViolationRange]['line'][iLineNumber + iIndex] = {}
            self.dFix['violations'][sViolationRange]['line'][iLineNumber + iIndex]['keywordColumn'] = oGroupLine.line.find(sKeyword)

            iMaximumKeywordColumn = get_maximum_keyword_column(oGroupLine, sKeyword, iMaximumKeywordColumn)

            iKeywordAlignment = update_keyword_alignment(oGroupLine, sKeyword, iKeywordAlignment)

            if not iKeywordAlignment == oGroupLine.line.find(sKeyword):
                add_range_violation(self, sViolationRange)

    self.dFix['violations'][sViolationRange]['maximumKeywordColumn'] = iMaximumKeywordColumn
```

You add all groups to the dFix dictionary even if the rule is not violated in given group.
Then in the fix function you iterate over `dFix` instead of `violations`:
```python
def keyword_alignment(self, oFile):
    '''
    Aligns keywords across multiple lines.

    Parameters:

      self: (rule object)

      oFile: (vhdlFile object)
    '''
    for sKey in self.dFix['violations']:
        iMaximumKeywordColumn = self.dFix['violations'][sKey]['maximumKeywordColumn']
        for iLineNumber in self.dFix['violations'][sKey]['line']:
            iKeywordColumn = self.dFix['violations'][sKey]['line'][iLineNumber]['keywordColumn']
            oLine = oFile.lines[iLineNumber]
            oLine.update_line(oLine.line[:iKeywordColumn] + ' '*(iMaximumKeywordColumn - iKeywordColumn) + oLine.line[iKeywordColumn:])
```

It of course works but is inefficient (fixing things that are already ok).
As a test I have modified `keyword_alignment` `fix` function to look like this:
```python
def keyword_alignment(self, oFile):
    '''
    Aligns keywords across multiple lines.

    Parameters:

      self: (rule object)

      oFile: (vhdlFile object)
    '''
    for sKey in self.dFix['violations']:
        if sKey not in self.violations:
            continue

        iMaximumKeywordColumn = self.dFix['violations'][sKey]['maximumKeywordColumn']
        for iLineNumber in self.dFix['violations'][sKey]['line']:
            iKeywordColumn = self.dFix['violations'][sKey]['line'][iLineNumber]['keywordColumn']
            oLine = oFile.lines[iLineNumber]
            oLine.update_line(oLine.line[:iKeywordColumn] + ' '*(iMaximumKeywordColumn - iKeywordColumn) + oLine.line[iKeywordColumn:])
```

All tests pass. I will also improve keyword alignment violations handling.

m-kru, 2020-02-12:
@jeremiah-c-leary can you explain me what was the idea behind the `dFix` and `violations` attributes in `rule` class. I can see that sometimes you iterate over `dFix` to fix issues and sometimes over `violations`. What is more `dFix` dictionary has `violations` key with dictionary as a value.
```python
class rule():

    def __init__(self, name=None, identifier=None):
        self.name = name
        self.identifier = identifier
        self.solution = None
        self.violations = []
        self.indentSize = 2
        self.phase = None
        self.subphase = 1
        self.disable = False
        self.fixable = True
        self.dFix = {}
        self.dFix['violations'] = {}
        self.configuration = ['indentSize', 'phase', 'disable', 'fixable'
```
I am confused.

jeremiah-c-leary, 2020-02-13:
Hey mkru,

I apologize for the late reply.  There are several questions here so I will respond in order.

> I would also add one more option to this rule, configurable via config file, that would allow user to define behavior when blank line is encountered between for example signal declarations. When set or unset blink line would function as a separator for groups that should be aligned.

I was wondering if a blank line is enough of a delimiter.  For example:
```vhdl
  -- Some signal declarations
  signal a : std_logic;
  signal b : std_logic;
  -- Some more signal declarations
  signal write_enable : std_logic;
  signal read_enable : std_logic;
```
Should the two groups of signals be considered in independent or would the :'s be aligned like this:
```vhdl
  -- Some signal declarations
  signal a            : std_logic;
  signal b            : std_logic;
  -- Some more signal declarations
  signal write_enable : std_logic;
  signal read_enable  : std_logic;
```
Maybe we could use the inverse of the trigger as an option to the base rule.  So we basically gather lines that only meet the trigger until it finds a line that does not meet the trigger.

>One more thing. I wold like this rule to work with following signal declarations: Multiple signal definitions with the same class and type .

Interesting idea.  If the base rule is already gathering the signals it could reformat it to that style.

I was just thinking of comment you had made about conflicting rules and was wondering if this is one of those cases.  If you consider the following rules:  signal_012, signal_015, signal_016 and the one you are proposing.  Should we combine them into a single rule with configuration options?  Then the user would have a single place to configure the formatting of signal declarations instead of having to know there are multiple rules.  When we implement your style, then you will have to disable signal_015 and signal_016.  You might have to disable signal_012 also depending on how it behaves.

Question:  Would you want to align colons for your preferred signal declaration method?

>It of course works but is inefficient (fixing things that are already ok).
As a test I have modified keyword_alignment fix function to look like this:

I can not remember why I chose to fix things that did not need to be fixed, but it makes sense to skip those that do not.  So I would commit the proposed change.

>@jeremiah-c-leary can you explain me what was the idea behind the dFix and violations attributes in rule class. I can see that sometimes you iterate over dFix to fix issues and sometimes over violations. What is more dFix dictionary has violations key with dictionary as a value.

The `dFix` dictionary came about when I ran into rules where I needed more information in order to fix the violation.  If you consider the rule `process_019` which checks the case of the end process label, that rule does not need to remember anything except the line number that violates the style.

However, if you consider the rule `process_018`, which checks for the existence of the end process label, that rule needs to know the line number and the previous process label that needs to be applied because there can be multiple processes in a single file.  `process_018` gathers process labels as it moves through the file and then checks the end process line to see if a label exists.  If it does not, then it stores the line number that should have a label and the previous process label it found.  It also stores the line number in the `violations` list.

We could move all the rules to use the `dFix` variable by changing the `add_violation` and `report_violations` methods on the base rule class.  We would also have to change lines in rules like the following:

```python
    def _fix_violations(self, oFile):
        for iLineNumber in self.violations:
```
It would probably be best to add a method to return the violations as a list instead of just referencing the list directly.

```python
    def _fix_violations(self, oFile):
        for iLineNumber in self.get_violations():
```

jeremiah-c-leary, 2020-02-13:
arg...hit the wrong button.

m-kru, 2020-02-13:
> I was just thinking of comment you had made about conflicting rules and was wondering if this is one of those cases. If you consider the following rules: signal_012, signal_015, signal_016 and the [...]

I propose to merge conflicting rules into a single rule and provide configuration options.

Well, there is quite a lot of things to be done. I do not know where to start. Maybe from unifying how violations are fixed (`dFix`).

jeremiah-c-leary, 2020-02-13:
I think unifying the violations should be done first.  I create issue #335 to cover that aspect of this thread.

m-kru, 2020-02-26:
@jeremiah-c-leary There is one issue right now.Take as a example following port declarative part:
```vhdl
    Port (  
        clk_i : in std_logic := 'X';                                            -- system clock
        data_i  : in std_logic_vector (N-1 downto 0) := (others => 'X');         -- noisy input data
        data_o    : out std_logic_vector (N-1 downto 0);                           -- registered stable output data
        strb_o    : out std_logic                                                  -- strobe for new data available
    );
```

Now imagine the user has enabled alignment for: `:`, `:=`, `--`. The result is as follows:
```vhdl
    Port (
        clk_i     : in std_logic                       := 'X';                                            -- system clock
        data_i    : in std_logic_vector (N-1 downto 0) := (others => 'X');                                -- noisy input data
        data_o    : out std_logic_vector (N-1 downto 0);                                                  -- registered stable output data
        strb_o    : out std_logic                                                                         -- strobe for new data available
    );
```
Notice how far on the right the comment is located right now. We can change the logic of keyword alignment fix to remove any extra spaces before the keyword. What do you think? The result would be as follows:
```vhdl
    Port (
        clk_i  : in std_logic                       := 'X';             -- system clock
        data_i : in std_logic_vector (N-1 downto 0) := (others => 'X'); -- noisy input data
        data_o : out std_logic_vector (N-1 downto 0);                   -- registered stable output data
        strb_o : out std_logic                                          -- strobe for new data available
    );
```

If we agree to do this there are 2 questions to answer.
1. What to do when the keyword is the first non space character in the line? I propose to do nothing in such case.
2. What to do when the alignment is correct, but there are extra spaces. For example should this:
```vhdl
    constant my_const      : natural;
    signal s               : std_logic;
    signal enable          : std_logic;
```
be fixed to this:
```vhdl
    constant my_const : natural;
    signal s          : std_logic;
    signal enable     : std_logic;
```
?

jeremiah-c-leary, 2020-02-27:
@m-kru My original implementation was to always use the right most keyword as the alignment column.  I figured it was the "safest" fix.  However, it was really an arbitrary decision.

>1. What to do when the keyword is the first non space character in the line? I propose to do nothing in such case.

Do you mean something like this if the keyword is `:`?

```vhdl
   constant my_const
   : natural;
   constant my_other_constant : natural;
   constant my_third_constant      : natural;
```
Fixed to:
```vhdl
   constant my_const
   : natural;
   constant my_other_constant : natural;
   constant my_third_constant : natural;
```
>2. What to do when the alignment is correct, but there are extra spaces. For example should this:

I was thinking we could add a `left_align` option and default it to `True`.  That would pull everything to the left, which I believe is the preferred style choice.  Allowing it to be configured allows the user to decide.

m-kru, 2020-03-01:
@jeremiah-c-leary your example:
```vhdl
   constant my_const
   : natural;
   constant my_other_constant : natural;
   constant my_third_constant      : natural;
```

Would be fixed to this:
```vhdl
   constant my_const
                              : natural;
   constant my_other_constant : natural;
   constant my_third_constant : natural;
```

The question is what should be done with this:
```vhdl
       a <= b --
     after 1 ns;
       b <= c
      after
            1
            ns;
       c <= d
          after
            1 ns;
```
In this case the 'after' keyword is always the first word in the line. How to decide what is the reference for the alignment in such case?

jeremiah-c-leary, 2020-03-05:
I am not sure if VSG would current handle this example well:

```vhdl
       a <= b --
     after 1 ns;
       b <= c
      after
            1
            ns;
       c <= d
          after
            1 ns;
```
..but I believe in this case we should just line up the `afters`:
```vhdl
       a <= b --
     after 1 ns;
       b <= c
     after
            1
            ns;
       c <= d
     after
            1 ns;
```

There should be another rule that states the `after` should be on the same line:
```vhdl
       abcd <= b after 1 ns; --
       bc <= cffef after
            1
            ns;
       cxyzef <= fd after
            1 ns;
```
Then the alignment of the `=>` would take affect next:
```vhdl
       abcd   <= b after 1 ns; --
       bc     <= cffef after
            1
            ns;
       cxyzef <= fd after
            1 ns;
```
Finally the alignment of the `after` would take affect:
```vhdl
       abcd   <= b     after 1 ns; --
       bc     <= cffef after
            1
            ns;
       cxyzef <= fd    after
            1 ns;
```
We could use the `subphase` to enforce the order.

m-kru, 2020-03-11:
@jeremiah-c-leary ping

jeremiah-c-leary, 2020-03-11:
@m-kru Just saw this.  I will look into it tonight.

jeremiah-c-leary, 2020-03-12:
@m-kru I think we have decided that keyword alignments should occur where at the proper "level" of alignment.  Given that, I think the following makes sense:

comment_001 could move to library rules because it is specific to the use statement
comment_003 could move to process rules because it deals with comments in the process region.
comment_004 would stay as it applies to all comments regardless of location
comment_005 could move to case rules because it is specific to comments above the when statements
comment_006 could move to the process rules because it is specific to the process declarative region
comment_008 could move to the if rules because it is specific to the elsif keyword
comment_009 could move to the if rules because it is specific to the else keyword
comment_010 would stay as it applies to comments regardless of where they occur.

Would you agree?

m-kru, 2020-03-12:
This is one approach and there will be only 2 rules left in the comment group. Another approach is to put all comment rules in the comment group. Then it would be obvious for the user to look in the comment group if he looks for something related to the comment formatting. On the other hand if you want to format comments inside, lets say process, it is quite obvious to look for the rule in the process group. I struggle to find any strong argument for any of these approaches.

jeremiah-c-leary, 2020-03-12:
> I struggle to find any strong argument for any of these approaches.

I agree.  Either option makes sense.  I think it is interesting that this appears to only apply to the `keyword_alignment` rules.  Probably because it applies over multiple lines and depends on the context (inside if statements, inside case statements, inside process declarative regions, etc...).

So I would propose if a rule depends on it's context, then it should be grouped with that context.  I believe it would make more sense to someone who was browsing the documentation.

For example, if someone is browsing the `if` rules they would see `comment_008` and `comment_009` in relation to all the other `if` rules.  Even though those two rules deal with comments, they only make sense in relation to the `if` rules.

closed 2020-03-21 by m-kru