jeremiah-c-leary/vhdl-style-guide #332: concurrent_007 should allow item following else on the same line if there is only a single else

jredders, 2020-01-30:
**Is your feature request related to a problem? Please describe.**
The following is a very simple and concise way to generate a tri-state output. but is not allowed by the concurrent_007 rule.

`O_FOO <= w_foo when (w_foo_en = '1') else 'Z';`

**Describe the solution you'd like**
I'd like to see vsg allow single line concurrent when-else statements as long as there is only a single when-else like the example above. I don't want to disable the rule, because I'd still like it to flag when-else constructs with multiple elses on a single line like the following example.

`O_FOO <= w_foo when (w_foo_en = '1') else w_bar when (w_bar_en = '1') else 'Z';`

jredders, 2020-01-30:
So maybe two different concurrent rules?  One for when-single-else cases and one for when-multiple-else cases?

jeremiah-c-leary, 2020-01-31:
I can explore a configuration option so the same rule can handle both cases.  I can add an option called ```allow_single_else``` and default it to **false** to keep the existing behavior.  Then if the option is set to **true** then it will allow single line concurrent statements.

This would also force the following code:
```vhdl
O_FOO <= w_foo when (w_foo_en = '1') else
         'Z';
```
to be fixed to this:
```vhdl
O_FOO <= w_foo when (w_foo_en = '1') else 'Z';
```
unless we add another option to allow either option.

jeremiah-c-leary, 2020-02-11:
I think I like your idea of having a rule to handle multiple whens declared on a single line and the existing one would allow for single line whens.

jeremiah-c-leary, 2020-02-23:
I ended up keeping the single rule and having it handle multiple whens/elses.  Check the documentation for how to configure the option to allow for single line concurrent statements with a single else.

closed 2020-02-23 by jeremiah-c-leary