jeremiah-c-leary/vhdl-style-guide #335: Unify how violations are collected and reported

jeremiah-c-leary, 2020-02-13:
### History
Within the base rule class there is a `violations` list attribute which collects any line number with a violation for that particular rule.  This list is then used during the **fix phase** to indicate where changes can be made.

At some point during development, the `violations` list attribute was not enough to perform a suitable fix.  A dictionary `dFix` was added to keep track of the more complex fixes which needed to be performed.

### Issue
The `violations` list attribute is now redundant to the `dFix` dictionary and should be removed to improve code clarity.

### Tasks

- [ ] Add a method to the base rule class to replace references to the `violations` attribute
- [ ] Update all references to the `violations` attribute to use the new method

m-kru, 2020-02-13:
@jeremiah-c-leary I have an irresistable impression that `dFix` should be a list not a dictionary. Within a given rule if there is any violation you store data needed to fix it in a specific data type. Sometimes you need a dictionary sometimes only a line number, but this is always the same type within a given rule. It indicates that data type for storing violations/fixes for given rule should be a list not a dictionary.

There are 2 options:

1.  We remove `dFix` attribute and add all data for fixing to the `violations` list. It would be a list of integers in case of line number and a list of dictionaries in case of more complex rules.
2. We remove `violations` attribute and change type of `dFix` from dictionary to list.

With a list we do not need any extra function to get the list of violations. We can still iterate over a list, like now  we do.

jeremiah-c-leary, 2020-02-13:
@m-kru ... that is an interesting concept.  So we keep the violations as a list and the rule determines the type to store in the list.  That would keep the complexity of the violation isolated to the rule that needs the complexity.

I like it.

I will experiment with taking an existing rule that uses dFix and push that information in the violations list.  Then update fix violations method to pull the information out of the list.

jeremiah-c-leary, 2020-02-14:
I am starting to look at this, and was thinking about standardizing the `dFix` dictionary so data could be pulled from it by upper level functions.  Something like:

```python
dFix[lineNumber] = integer
dFix[solution] = string
```

The dictionary could then be extended by the rule itself, but would have at least the `lineNumber` key.

Then I though maybe a violation could be a class.  The `violations` attribute would then be a list of violation objects.  Then pulling information from the violation is performed with methods.  Or is that too much abstraction?
 
I am going to update process_018 with a list of dictionaries first and see how it goes.

jeremiah-c-leary, 2020-02-14:
@m-kru I just committed a change to the process_018 rule on the issue_335 branch.  Take a look.  I think your suggestion works out really well.

m-kru, 2020-02-14:
This is very similar to what I had in mind, but in my opinion it is more complex than what is needed.

Here is my idea. Remove completely `dFix` attribute (it can't be done right now as other rules still use it, but at the end there should be no `dFix` attribute in the rules class.
Take a look at this:
```python
from vsg import rule
from vsg import utils

import re

class rule_018(rule.rule):
    '''
    Process rule 018 checks the "end process" has a label.
    '''

    def __init__(self):
        rule.rule.__init__(self, 'process', '018')
        self.solution = 'Add a label for the "end process".'
        self.phase = 1
        self.subphase = 2

    def _pre_analyze(self):
        self.labelStack = ''
        self.iProcStartLine = 0
        self.iProcLabelLine = 0

    def _analyze(self, oFile, oLine, iLineNumber):
        if oLine.isProcessKeyword:
            self.iProcStartLine = iLineNumber
            oMatch = re.match('^\s*\S+\s*:\s*process', oLine.lineLower)
            if oMatch:
                self.iProcLabelLine = iLineNumber
                self.labelStack = utils.extract_label(oLine)[0]
        if oLine.isEndProcess:
            dViolation = {}
            dViolation['lineNumber'] = iLineNumber
            if self.labelStack and self.iProcStartLine == self.iProcLabelLine:
                dViolation['processLabel'] = self.labelStack
            if not re.match('^\s*\S+\s+\S+\s+\S+', oLine.line):
                self.add_violation(dViolation)
            self.labelStack = ''

    def _fix_violations(self, oFile):
         for dViolation in self.violations:
             try:
               oLine = oFile.lines[dViolation['lineNumber']]
               sLine = oLine.line
               iIndex = oLine.lineLower.find('process') + len('process')
               oLine.update_line(sLine[:iIndex] + ' ' + dViolation['processLabel'] + sLine[iIndex:])
             except KeyError:
               pass
```

> Then I though maybe a violation could be a class. The violations attribute would then be a list of violation objects. Then pulling information from the violation is performed with methods. Or is that too much abstraction?

About creating the `Violation` class. The only attribute that is common for all violations for all rules is the `line_number` and even that is not true as some rules need a range of lines

About pulling information from the violation with methods. I do not see any gain right now. It looks to me like only the given rule wants to access its own `violations` list.  Is there any case where some object wants to access the `violations` of other object?

jeremiah-c-leary, 2020-02-14:
>Here is my idea. Remove completely dFix attribute (it can't be done right now as other rules still use it, but at the end there should be no dFix attribute in the rules class.

I agree with removing the `dFix` attribute and switching to the `dViolation` dictionary.

> One more question. Why have you pushed it on top of issue-319 branch, not on master?

Could you double check on this?  I was intending to commit to the issue_335 branch until I converted everything over.  Although maybe in this case using a branch will be too much of merge problem.  I typically like to commit straight to master, but git seems to be geared toward branching with pull requests.  I have read some people use pull requests for everything they do on their projects.

m-kru, 2020-02-15:
I have checked.
```
* 4da0 (origin/issue_335) First pass at adding violation dictionaries to the violations list attribute.
* 6569 (origin/issue-319) Updates for issue #331
| * c430 (HEAD -> master, mkru/master) Improve keyword_alignment fix function.
| *   3220 (origin/master, origin/HEAD) Merge pull request #329 from willkamp/quickfixes
| |\
| | * 3856 Catch Permission error on trying to write fixes back to a read only file. Also removed superfluous file.close().
| | * ab97 Update signal_015 to only count commas between the signal keyword and the colon - not in any following initialization statement.
| | * 9e70 Update remove comment function. Now handles quoted double hyphens. Strips trailing whitespace.
| | * 41c3b Use setdefault function to initialise dictionaries.
| | * 36b1 Update comment detection regex.
| | * 3fc7 Various changes so that hyphens inside double quotes are not treated as a comment marker.
| | * 822ea Fix for stripping comments that understands that double quoted stings with hyphens in it doesn't count.
| | * be00 Fix for rule whitespace_011, it was confused by hyphens quoted strings (don't cares).
| | * 7475 Fix rule comment_004 to use the comment location in oLine, such that it picks up the fix for comment detection.
| | * cf426 Fix inline comment detection so that it ignores double hyphens in double quoted strings.
| | * dbce Fix type keyword regex that miss-matches on record fields that start with "type".
| * |   ad07e Merge pull request #327 from m-kru/first_keyword
| |\ \
| | * | e380 Replace extract_first_keyword calls with extract_words function.
| * | |   c0fb Merge pull request #326 from m-kru/master
| |\ \ \
| | |/ /
| |/| |
| | * | 985e Make architecture rules inherit from case rule.
| |/ /
| * |   a069 Merge pull request #324 from m-kru/master
| |\ \
| | * | b27d Make range rules inherit from case rule.
| |/ /
| * |   39b6 Merge pull request #323 from m-kru/master
| |\ \
| | * | 3dc7d Make extract functions operate on lineNoComment.
| |/ /
| * |   4d09 Merge pull request #321 from jeremiah-c-leary/issue-319
| |\ \
| |/ /
|/| /
| |/
* |   a5ef Merge branch 'issue-319' of https://github.com/jeremiah-c-leary/vhdl-style-guide into issue-319
```

When you push first correct example for transition from `dFix` to `violations` to master I will be able to start porting other rules to do the same. If we do it in two it will be much quicker as there are quite a lot of rules.

A lot of people use pull requests for everything because they want at least one person to review the changes.

m-kru, 2020-02-15:
@jeremiah-c-leary One more thing I have just realized. Functions like `is_uppercase`, `is_lowercase` (there is probably more such functions) call `add_violation()`. The problem I see is that this functions do more than is indicated by the name. `add_violation()` should be called in analyze function of given class. In case of `is_uppercase` and `is_lowercase` I think we can just remove them as I see no gain in `if is_uppercase(...)` over `if sString == sString.upper()`.

jeremiah-c-leary, 2020-02-15:
> I have checked.

I have no idea how that happened, and at the moment no idea how to fix it.

I will work on cleaning up the code and then merging it to master.

m-kru, 2020-02-15:
@jeremiah-c-leary  I have prepared patch for `case_rule` https://github.com/jeremiah-c-leary/vhdl-style-guide/pull/336 . I really like the result. Less application code, more tests code, plus the solution is simple and clear.

jeremiah-c-leary, 2020-02-15:
@m-kru   I agree.  Using the `violations` attribute as a list and adding a dictionary for each violation simplifies the code.  In the case_rule update with the addition of the `words_to_fix` key, you know exactly what needs to be changed.

jeremiah-c-leary, 2020-02-15:
I just pushed my changes for process_018.

Question, how do we divide and conquer this without working on the same code?

m-kru, 2020-02-15:
> Maybe one of us works alphabetically down the rules in vsg/rules while the other works alphabetically up?

It does not mean that there will be no collision as some rules in different `vsg/rules` have the same base classes.

Well, I am not hiding that I want `vsg` to support these features: https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/330 and https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/290 as soon as possible. That's why I propose that I will prepare patch for `keyword_alignment` rule to use `violations` attribute for fixes and then I will work on the above issues. When I finish I can help you with this issue, if you do not finish by then.

jeremiah-c-leary, 2020-02-15:
> Well, I am not hiding that I want vsg to support these features: #330 and #290 as soon as possible. That's why I propose that I will prepare patch for keyword_alignment rule to use violations [...]

Sounds like a plan.  I work to align the rest of the rules to use the `violations` attribute for fixes.

jeremiah-c-leary, 2020-02-15:
I got through quite a few rules, but there are still a lot to go.

I added two functions to the vsg.tests.utils file.  They transform a number or a list into the new format for the violations list.  It works if you only need to store the line number.  Anything more complicated, like the case rules, will require formatting by hand.

m-kru, 2020-02-16:
@jeremiah-c-leary If you use PyCharm you can place break point and right click to copy value of `violations` attribute. I did that for `case_rule`. I am not a big fan of having functions for formatting the expected data in test functions. When I look at tests I want to immediately know what is expected, I do not want to analyze third party functions. This is of course only my opinion.

jeremiah-c-leary, 2020-02-17:
@m-kru I think I changed all the rules except the keyword alignment rule you are working on.  I need to investigate why componet_015 is not being used.

m-kru, 2020-02-17:
@jeremiah-c-leary I am working on the `keyword_alignment_rule` and there is one decision that needs to be made.
The `keyword_alignment_rule` now has following 2 configuration options:
```python
        self.blank_line_ends_group = # should be True or False by default?
        self.configuration.append('blank_line_ends_group')
        self.comment_line_ends_group = # should be True or False by default?
        self.configuration.append('comment_line_ends_group'
```
We need to decide if these options should be `True` or `False` by default. If we want to keep backward compatibility we should configure them differently for different keyword alignment rules, but I do not like this idea. I think it would be good to break backward compatibility a bit for unifying the default configuration for keyword alignment rules. It seems to be more user friendly (If you already know one keyword alignment rule you expect the other ones to have the same default configuration).  I opt for setting them to `True` by default.Of course, the final decision is yours.

jeremiah-c-leary, 2020-02-17:
If we survey the current configuration of tests that use `keyword_alignment_rule` against the two configuration items:

```console
rule               |  bleg  |  cleg  | keyword
--------------------------------------------------
constant_009         False    False       :
instantiation_015    False    False      =>
variable_009         False    False       :
after_002            False    False     after
comment_003          False    False      --
comment_006          False    False      --
entity_018           False    False      --
component_017        False    False      :
component_020        False    False      --
signal_013           False    False      :
concurrent_006       True     True       <=
concurrent_008       True     False?     --
```

I wonder how the `cleg` will work with the rules that use the `--` keyword?

I would say we default them to true, but override them in those rules that extend the `keyword_alignment_rule`.

m-kru, 2020-02-17:
> I wonder how the cleg will work with the rules that use the -- keyword?

I think there should be no problems as only the line starting with `--` will end the group. In case of any problems `cleg` will be permanently set to `False` for such rules.

> I would say we default them to true, but override them in those rules that extend the keyword_alignment_rule. 

Do I understand it correctly that you want to keep backward compatibility? If so it makes more sense to set the defaults to `False` as more keyword alignment rules behave in such way right now.

jeremiah-c-leary, 2020-02-17:
> Do I understand it correctly that you want to keep backward compatibility? If so it makes more sense to set the defaults to False as more keyword alignment rules behave in such way right now.

I was thinking about the base `keyword_alignment_rule` versus the rules like constant_009 and how future rules that would use `keyword_alignment_rule` should default.

It seems to me the base `keyword_alignment_rule` should start out as restrictive as possible.  Which would be both `bleg` and `cleg` being set to `True`.  That way when someone creates another rule based on `keyword_alignment_rule` the groups the alignments are performed over are smaller.  If you want to include more code then you disable the two options.

Although maybe in the long run it doesn't really matter.  The user will just configure the two options to fit their needs if the current configuration doesn't.

I think we need to ensure the current rules perform the same, but allow the user to change their behavior.

So yes to backwards compatibility, but I think we should set the defaults to `True` to encourage future rules to use smaller groups.

I hope that made sense.

m-kru, 2020-02-17:
I am think about this right now and to be honest setting the defaults to `True` for all keyword alignment rules won't break the backward compatibility in a way we usually think of when speaking of breaking backward compatibility. Current users will not have to adjust their .vhd files. vsg will just use smaller groups for alignment. I would opt for setting the defaults to `True` for all keyword alignment rules, it will be more consistent from user perspective. What do you think?

jeremiah-c-leary, 2020-02-19:
I was thinking hard about whether to change the default behavior of the `keyword_alignment_rule` and I realized I was facing the same problem other users face when using this tool.  Namely, changing the default means I now need a configuration so VSG formats the code to my style.

That is an aspect of using the tool I have not had to encounter.  You could think of it as a blind spot.   So I am now thinking there are pain points other users are experiencing that I am not, and if I want to really improve VSG then I should start experiencing it as other users do.

So...to answer your question:  let's change the default options to `True`.  It is better to be more conservative with changes and allow the user to configure it to be more liberal.  I also want to see how I would handle a change to VSG that differs from how I would configure it.

m-kru, 2020-02-20:
@jeremiah-c-leary Is the order of analysis and fixing deterministic within given rule class? For example lets assume that there are `rule_001` and `rule_002` in `process` directory. Will `rule_001` always be fixed before `rule_002`? What the order of between classes. Is it deterministic alphabetical order?

jeremiah-c-leary, 2020-02-20:
I believe it ends up being alphabetical order, but I do not specifically enforce that.  Sometimes I have had to renumber rules to get them to sequence correctly.

I did add a `subphase` attribute to the rule class so you can sequence rules within a phase.

m-kru, 2020-02-20:
Refactoring `keyword_alignment_rules` is much harder than I thought. Rewriting the rule was easy, however there are other rules that also insert blank characters to the line and tests fails because of wrong number of spaces. I have changed 1 thing and 5 new broke ...

m-kru, 2020-02-20:
Ok, `subphase` attribute has probably save me a lot of time.

m-kru, 2020-02-20:
@jeremiah-c-leary I  have created merge request so that you can look how it looks right now. Do not merge it as it is not finished yet.

jeremiah-c-leary, 2020-02-20:
understand

m-kru, 2020-02-21:
@jeremiah-c-leary I am adjusting tests right now and there is one thing that I do not know how to do.
In `test_code_examples.py` there are functions like:
```python
    def test_spi_slave(self):
        oRuleList = rule_list.rule_list(oSpiSlave)
        oRuleList.fix(7)
        lExpected = ['']
        utils.read_file(os.path.join(os.path.dirname(__file__),'spi_slave.fixed.vhd'), lExpected)
        for iLineNumber, sLine in enumerate(lExpected):
            self.assertEqual(oSpiSlave.lines[iLineNumber].line, sLine)

    def test_spi_master(self):
        oRuleList = rule_list.rule_list(oSpiMaster)
        oRuleList.fix(7)
        lExpected = ['']
        utils.read_file(os.path.join(os.path.dirname(__file__),'spi_master.fixed.vhd'), lExpected)
        for iLineNumber, sLine in enumerate(lExpected):
            self.assertEqual(oSpiMaster.lines[iLineNumber].line, sLine)

    def test_grp_debouncer(self):
        oRuleList = rule_list.rule_list(oGrpDebouncer)
        oRuleList.fix(7)
        lExpected = ['']
        utils.read_file(os.path.join(os.path.dirname(__file__),'grp_debouncer.fixed.vhd'), lExpected)
        for iLineNumber, sLine in enumerate(lExpected):
            self.assertEqual(oGrpDebouncer.lines[iLineNumber].line, sLine
```
Now in the `test_spi_master` function I would like to globally change the default value of configuration `comment_line_ends_group`. How can I achieve this?

jeremiah-c-leary, 2020-02-21:
I would create a dictionary and call the `rule_list.configure` method.

```python
oRuleList = rule_list.rule_list(oSpiSlave)

dConfiguration = {}
dConfiguration['rule'] = {}
dConfiguration['rule']['global'] = {}
dConfiguration['rule']['global']['comment_line_ends_group'] = False

oRuleList.configure(dConfiguration)
oRuleList.fix(7)
```
Essentially mimicking the user passing a configuration.

m-kru, 2020-02-22:
@jeremiah-c-leary I have spotted little inconsistency. For entity declaration there are 2 distinct rules that separately assure alignment for generic part and port part (generic/rule_012 and entity/rule_017). But for example for instantiation there is single rule that tries to assure alignment for generic part and port part simultaneously (instantiation/rule_029). My proposition, lets have single rule and provide configuration option (for example `separate_generic_port_alignment`). What do you think? If you agree, then what should be the default value for this parameter? I would opt for `False`.

jeremiah-c-leary, 2020-02-22:
I double checked the rules and I see the following:

**instantiation_10** aligns => over the port map
**instantiation_15** aligns => over the generic

**entity_17** aligns : over all the ports in an entity
**generic_12** aligns : over all generics in an entity

**entity_18** aligns all inline comments in an entity over both generics and ports
**instantiation_29** aligns all inline comments in an instantiation over both generics and ports

I think it makes sense to change **generic_012** to an entity rule to make it consistent with the instantiation rules.

To be consistent with **entity_18** and **instantiation_29** we would either combine **instantiation_10** and **instantiation_15** into a single rule.  Then combine **entity_17** and **generic_12** into a single rule.  

Do you think adding the `seperate_generic_port_alignment` would make the `keyword_alignment_rule` too specific?  Although, the `if_statement_control_statements_end_group` already does this so maybe it is not a big deal.

The other option would be to split **entity_18** and **instantiation_29** into two rules.  One to cover generics and the other to cover ports.

I am good with either direction and a default of `False` for `separate_generic_port_alignment` makes sense.

m-kru, 2020-02-22:
> Do you think adding the seperate_generic_port_alignment would make the keyword_alignment_rule too specific? Although, the if_statement_control_statements_end_group already does this so maybe it is not a big deal.

In the final pull request configurations like `seperate_generic_port_alignment` and `if_statement_control_statements_end_group` will be placed in appropriate classes. The base `keyword_alignment_rule` class will only have `blank_line_ends_group` and `comment_line_ends_group`.

m-kru, 2020-02-24:
@jeremiah-c-leary How to build the docs, so that I can see how output pdf looks like?

jeremiah-c-leary, 2020-02-24:
@m-kru in the docs directory issue a `make` command and it will show the options:

```bash
docs$ make
Sphinx v2.1.2
Please use `make target' where target is one of
  html        to make standalone HTML files
  dirhtml     to make HTML files named index.html in directories
  singlehtml  to make a single large HTML file
  pickle      to make pickle files
  json        to make JSON files
  htmlhelp    to make HTML files and an HTML help project
  qthelp      to make HTML files and a qthelp project
  devhelp     to make HTML files and a Devhelp project
  epub        to make an epub
  latex       to make LaTeX files, you can set PAPER=a4 or PAPER=letter
  latexpdf    to make LaTeX and PDF files (default pdflatex)
  latexpdfja  to make LaTeX files and run them through platex/dvipdfmx
  text        to make text files
  man         to make manual pages
  texinfo     to make Texinfo files
  info        to make Texinfo files and run them through makeinfo
  gettext     to make PO message catalogs
  changes     to make an overview of all changed/added/deprecated items
  xml         to make Docutils-native XML files
  pseudoxml   to make pseudoxml-XML files for display purposes
  linkcheck   to check all external links for integrity
  doctest     to run all doctests embedded in the documentation (if enabled)
  coverage    to run coverage check of the documentation (if enabled)
```

I normally use the `html` option.

m-kru, 2020-02-24:
@jeremiah-c-leary I think I am finishing the work and I think again about the default values for the configurations. There are 5 configurations right now with following default values:
```python
blank_line_ends_group = True # Should it be True by default?
comment_line_ends_group = True # Should it be True by default?
separate_generic_port_alignment = False # Should it be False by default?
if_control_statements_end_group = True # This one I am sure should be True by default.
case_control_statements_end_group = True # This one I am sure should be True by default.
```

What do you think? By the way, the pull request is up to date so you can take a look.

jeremiah-c-leary, 2020-02-24:
@m-kru I believe they should all be `True` by default.  You made me realize that blank lines and comments are natural separators (seems kind of obvious now), so setting `blank_line_ends_group` and `comment_line_ends_group` to `True` makes sense.

The same applies for `seperate_generic_port_alignment`, because generics and ports have a natural separator.

If statements and cases have natural separators also.

I have my own preferences for `blank_line_ends_group` and `comment_line_ends_group`, but to be consistent I think all the options should be set to `True`.

jeremiah-c-leary, 2020-02-25:
@m-kru Now that I have merged the pull request, can this issue be closed?

m-kru, 2020-02-25:
You should not have merged this. It was not finished :/. It was mentioned in the commit message `[do not merge yet]`.

m-kru, 2020-02-25:
Do not _unmerge_ now. I will prepare yet another pull request with polishing changes.

jeremiah-c-leary, 2020-02-25:
My apologies.  I assumed you were done.  Next time I will double check before I merge.

m-kru, 2020-03-17:
@jeremiah-c-leary Isn't this issues ready to be closed?

jeremiah-c-leary, 2020-03-17:
@m-kru you had mentioned there was going to be another pull request to polish up some changes.  So I was going to keep this open until then.

m-kru, 2020-03-17:
@jeremiah-c-leary the pull request is waiting for your action.

closed 2020-03-21 by jeremiah-c-leary