jeremiah-c-leary/vhdl-style-guide #341: Function rule_011 is not documented.

m-kru, 2020-02-24:
Function rule_011 is not documented in the `function_rules.rst` file.

jeremiah-c-leary, 2020-02-24:
This is a bit of mystery.  It was added as part of fixing issue #287.  Although it never had a test created for it, not was it included in the `vsg.rules.function.__init__.py` file.

Looking at the description of the rule, I do not believe it would detect or fix what is claims.

The only thing I can think of is I thought I needed a new rule to address the violation, but realized I did not.  Then when I committed my changes I included the rule by accident.

I will contemplate how to proceed on this rule.

jeremiah-c-leary, 2020-02-25:
I can not think of what this rule was supposed to do that is already not being done.  I will delete it.

closed 2020-02-25 by jeremiah-c-leary