jeremiah-c-leary/vhdl-style-guide #345: Unify usage of note text in the docs.

m-kru, 2020-02-26:
I am writing documentation for keyword alignment rules and I have realized, that there is no concept how and for what the note text is utilized. Examples:

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Note that:
1. Sometimes there is only single paragraph, while sometimes there are 2 paragraphs per note.
2. Sometimes the default configurations are in the note, while sometimes there are just in the normal text.

Questions to be answered:
1. Do we even need the note text or maybe everything can be written in the normal text mode?
2. If we need the note text, then what should be placed in the notes. (the default configurations, information about how to configure, reference to the section describing configuration or maybe all of these?)
3. If we need the note text, then how do we structure it? Single paragraph or maybe paragraph per thing that is being discussed in the note?

I think, that based on the answers the docs should be unified.

jeremiah-c-leary, 2020-02-26:
This was the original process I thought someone would go through when using VSG:
```gherkin
Scenario:  VSG generates a violation for a rule that is configurable

  Given:  VSG is installed
   When:  user runs VSG on a file
    And:  VSG indicates an error
   Then:  user looks violation up on read the docs
   When:  user reads NOTE about configurable parameters
   Then:  user reads Configuration section on read the docs on how to set configurable items
   Then:  user creates configuration file
    And:  user applies configuration file to VSG run
```

Reflecting on this, the `Then` statements require a level of knowledge that is not readily available.

There was also a time when most of the rules were not configurable, so the NOTEs were to indicate those rules that are configurable.  We seem to be a point where the user should just assume a rule can be configured.  That would eliminate all configuration notes from the rules documentation for those rules that are enabled.  There are several rules that are disabled by default (architecture_025) so maybe those NOTEs stay.

To address the process knowledge gap I was wondering if VSG could print something at the end of a 
run:
```bash
example$ python3 ../bin/vsg -f spi_slave.vhd
File:  spi_slave.vhd
====================
Phase 1... Reporting
  port_016                  |         49 | Move port definition to it's own line.
  entity_015                |         56 | The "end" keyword, "entity" keyword and entity name need to be on the same line.
  type_009                  |         60 | Move enumerated type to it's own line.
  type_008                  |         61 | Move the closing parenthesis to it's own line.
  signal_007                |         62 | Remove default assignment.
  signal_007                |         63 | Remove default assignment.
  signal_007                |         64 | Remove default assignment.
  signal_007                |         66 | Remove default assignment.
  signal_007                |         67 | Remove default assignment.
  signal_007                |         68 | Remove default assignment.
  process_012               |         72 | Add "is" keyword after the closing parenthesis.
  process_016               |         72 | Add a label for the process.
  process_029               |         76 | Use 'event for clocks.
  process_018               |        247 | Add a label for the "end process".
  architecture_010          |        249 | Add "architecture" keyword after "end" keyword.
Phase 2... Not executed
Phase 3... Not executed
Phase 4... Not executed
Phase 5... Not executed
Phase 6... Not executed
Phase 7... Not executed
====================
Total Rules Checked: 61
Total Violations:    15

Rule configuration options:
process_029 :
    "clock": [event|edge]
```

Maybe we could print out the documentation.  Maybe like a man page:

```bash
example$ vsg -man process_029

process_029
###########

This rule checks for the format of clock definitions in clock processes.
The rule can be set to enforce **event** definition:

    if (CLK'event and CLK = '1') then

..or **edge** definition:

    if (rising_edge(CLK)) then

event configuration
^^^^^^^^^^^^^^^^^^^

NOTE: This is the default configuration.

**Violation**

   if (rising_edge(CLK)) then

   if (falling_edge(CLK)) then

**Fix**

   if (CLK'event and CLK = '1') then

   if (CLK'event and CLK = '0') then

edge configuration
^^^^^^^^^^^^^^^^^^

NOTE:  Configuration this by setting the *'clock'* attribute to *'edge'*

      {
        "rule":{
          "process_029":{
             "clock":"edge"
          }
        }
      }

**Violation**

   if (CLK'event and CLK = '1') then

   if (CLK'event and CLK = '0') then

**Fix**

   if (rising_edge(CLK)) then

   if (falling_edge(CLK)) then
```

Maybe something to nudge the user in the right direction.

m-kru, 2020-02-26:
> To address the process knowledge gap I was wondering if VSG could print something at the end of a run:

At the end of a run user usually expects the results, not a list of available configurations. We could add an one line reminder about the available docs pointing to the url.

> Maybe we could print out the documentation. Maybe like a man page:

I am not sure if we need this. The generated documentation is really good. In my opinion https://vhdl-style-guide.readthedocs.io/en/latest/ and pdf is enough.

 One more thing. Some of the documentation may be in the Python sources, I am not sure if it is in sync with the docs.

jeremiah-c-leary, 2020-02-26:
It seems the `Configuring` section in the documentation is a good place to collect details on how to configure the rules.   The question I have is how would you know which configuration applies to which rule.  I think some are obvious, but others might not be.

Maybe I am over thinking this.

If we removed all the NOTEs that refer back to the Configuring section, it would clean up the rules documentation and force all configuration documentation into a single location.

If we print out something at the end of the run with a violation that refers the user to the configuring section of the documentation, it would nudge the user to look at configuration possibilities.  Since all the configuration possibilities are in one place then the user will be able to determine how to configure any rules to suit their needs.

m-kru, 2020-02-26:
> If we removed all the NOTEs that refer back to the Configuring section, it would clean up the rules documentation and force all configuration documentation into a single location.

I do not want to remove the content from the NOTEs that refers to the configuration section. I just want to put that text into the normal mode, not `.. NOTE::`.

jeremiah-c-leary, 2020-02-26:
After looking at the PDF documentation, the NOTE do tend to stand out a little too much.
The description of the rules are typically a single sentence, so the NOTE takes up a lot of room relative to it.

I'm fine with make the NOTEs normal text.

m-kru, 2020-04-03:
Turned out most of the notes disappeared after the case rule is set for lowercase for all rules.

closed 2020-04-03 by m-kru