jeremiah-c-leary/vhdl-style-guide #346: Rule generic_020 always detects error, even for properly formatted names

abyszuk, 2020-02-26:
**Environment**
Ubuntu 19.10, Python 3.7.5, VSG 1.2.0 from PIP

**Describe the bug**
When I enable rule generic_020 it will always complain with `Generic identifiers should start with one of ['G_, g_'] prefixes.` message

JSON config:
```
    "generic_020": {
      "disable": false,
      "fixable": false,
      "indentSize": 2,
      "phase": 7,
      "prefixes": [
        "G_, g_"
      ]
```

Example VHDL code:
```vhdl
  generic (
    G_PE_N                 : positive := 4;   --number of PEs
    G_PE_SEL_WIDTH         : positive := 16;  --1bit per PE
    G_ADDR_WIDTH           : positive := 16;  --log2(BankSize)
    G_MEMWORD_WIDTH        : positive := 32;  --bits in memword
    G_BYTE_SEL_WIDTH       : positive := 8;   --1bit per byte in memword
  );
```

jeremiah-c-leary, 2020-02-26:
Hey abyszuk,
I will take a look at it.

jeremiah-c-leary, 2020-02-27:
Hey abyszuk,

I create this test case vhdl file:
```vhdl
 
 entity BLAH is
   generic (
     G_PE_N                 : positive := 4;   --number of PEs
     G_PE_SEL_WIDTH         : positive := 16;  --1bit per PE
     G_ADDR_WIDTH           : positive := 16;  --log2(BankSize)
     G_MEMWORD_WIDTH        : positive := 32;  --bits in memword
     G_BYTE_SEL_WIDTH       : positive := 8;   --1bit per byte in memword
   );
 end entity BLAH;
```
and used this configuration:
```json
 {
   "rule": {
     "generic_020": {
       "disable": false,
       "fixable": false,
       "indentSize": 2,
       "phase": 7,
       "prefixes": [
         "G_", "g_"
       ]
     }
   }
 }
```
...but VSG is not flagging the generics:
```bash
example$ python3 ../bin/vsg -f example.vhd -c config.json
File:  example.vhd
==================
Phase 1... Reporting
Phase 2... Reporting
Phase 3... Reporting
Phase 4... Reporting
Phase 5... Reporting
Phase 6... Reporting
Phase 7... Reporting
==================
Total Rules Checked: 381
Total Violations:    0
```
I changed the prefix values to "A_", "a_" and I get the following:
```bash
example$ python3 ../bin/vsg -f example.vhd -c config.json
File:  example.vhd
==================
Phase 1... Reporting
Phase 2... Reporting
Phase 3... Reporting
Phase 4... Reporting
Phase 5... Reporting
Phase 6... Reporting
Phase 7... Reporting
  generic_020               |          4 | Generic identifiers should start with one of ['A_', 'a_'] prefixes.
  generic_020               |          5 | Generic identifiers should start with one of ['A_', 'a_'] prefixes.
  generic_020               |          6 | Generic identifiers should start with one of ['A_', 'a_'] prefixes.
  generic_020               |          7 | Generic identifiers should start with one of ['A_', 'a_'] prefixes.
  generic_020               |          8 | Generic identifiers should start with one of ['A_', 'a_'] prefixes.
==================
Total Rules Checked: 381
Total Violations:    5
```

I am running with Ubuntu 18.04, python 3.7.2 and vsg 1.2.0 from the release tag.

I noticed your config example was not a complete config.  Are there other rules you are configuring?
Maybe there is an odd interaction going on.

Could you replicate my files and see if the issue shows up for you?

Meanwhile, I will download python 3.7.5 and see if the issue pops up.

jeremiah-c-leary, 2020-02-27:
I tried python 3.7.5 and it behaves the same as 3.7.2.

abyszuk, 2020-02-27:
Turns out it was my fault all along.
The typo I fixed in the first message? It was present also in JSON config:
`"G_", "g_"` vs `"G_, g_"`
This is also visible in the output message, although missing apostrophes are hard to spot at first glance.
`Generic identifiers should start with one of ['G_, g_'] prefixes.`
vs
`Generic identifiers should start with one of ['G_', 'g_'] prefixes.`

I'm very sorry for wasting your time with this false alarm...

closed 2020-02-27 by abyszuk

jeremiah-c-leary, 2020-02-27:
That is a subtle difference.

I would not say it was a waste of time.  I gained several pieces of valuable knowledge:

1.  Where gitHub stores tags.
2.  How to create a branch from a tag.
3.  Determined how to organize my python installs to support multiple concurrent versions.
4.  I should probably upgrade my Ubuntu install to a newer version.

Regards,

Jeremy