jeremiah-c-leary/vhdl-style-guide #349: Next release schedule

m-kru, 2020-03-17:
@jeremiah-c-leary I am not sure if you use [semantic versioning](https://semver.org/), but if you do, the next release number must be `2.0.0`, not `1.3.0`. Keyword alignment rules have introduced some necessary backward incompatibility. I think we could take this opportunity to introduce some more backward incompatible changes and make a new official release.

Here is a list of changes I think should be implemented before release `2.0.0` (the list is in order in which I think we should take actions):

- [ ] Move comment rules to the blocks they concern. It was discussed here https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/330 but needs separate issue as https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/330 is ready to be closed.

- [ ] Set the default case to lowercase for all case rules. Discussed here https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/343.

- [ ] There was once a discussion about what rules should be enabled by default. I think it is good point to come back to this discussion and if we agree to make some changes they should be done before this release.

- [ ] Provide example configurations. It was described here https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/343 but needs separate issue. At least one legacy configuration is necessary for users to make the transition to new release smoother.

- [ ] Re-enable `test_code_examples` tests. I have disabled these tests as it was pain in the ass to update them. These tests should be run against example configurations, not for raw rules.

- [ ] Unify usage of note text in the docs and generally update the docs for all changes since last release - https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/345.

I am not sure I haven't forgotten about something.

What do you think? Do you agree? Do you want to add something more?

jeremiah-c-leary, 2020-03-17:
@m-kru I agree the next release should be `2.0.0` since there are a lot of user facing changes.

I would add the following issues:

#290 because we were talking about merging several signal rules into a single rule
#344 because it changes the format of the configuration file

We could also scrub the rules and see if there are any other organization changes we would like to make before the release.

m-kru, 2020-03-17:
Can you create the issues and update the Kanban board? It would be also beneficial if we explicitly assign all the issues for `2.0.0` either to me or you.

jeremiah-c-leary, 2020-03-17:
will do

jeremiah-c-leary, 2020-03-18:
@m-kru I have added the following issues:

#350 
#351 
#352 
#353 
#354 
#355 

I believe this covers everything, plus a little extra.

m-kru, 2020-03-18:
@jeremiah-c-leary great, I will assign myself to few of them.

jeremiah-c-leary, 2020-06-30:
@m-kru ,
I was looking at the list of tickets:
> #350
#351
#352
#353
#354
#355

and we have decided to defer #353 and I pushed #355 to the release after 2.0.0.
The only one open  is #352, which should hopefully be completed with pull request #433.

Just wondering if you feel good closing this issue.

jeremiah-c-leary, 2020-07-18:
Hey @m-kru ,
I am going to close this issue as I just released 2.0.0.

closed 2020-07-18 by jeremiah-c-leary