jeremiah-c-leary/vhdl-style-guide #350: Built In Configurations for backwards compatibility

jeremiah-c-leary, 2020-03-18:
The changes proposed for release `2.0.0` affect how rules are named and some of their default values.  In order to make `2.0.0` a more seemless transition, and provide a means for users to manage their own configurations, the concept of **build in configurations (BIC)** has been proposed (see issue #343 for some initial concepts). 

This issue includes the following tasks:

[ ] Decide on a name for this feature.  Is "Built In Configurations" good enough?
[ ] Determine how where the configurations will be stored in the directory hierarchy
[ ] Define format of the configurations
[ ] Determine how to let users know these configurations exist 
[ ] Define initial set of configurations to support
[ ] Determine how to test the configurations with unit tests
[ ] How can users submit their own configurations for inclusion with VSG
[ ] How can users include their local configurations as an option
[ ] Should users be able to disable existing built in configurations?

m-kru, 2020-04-05:
@jeremiah-c-leary I have an idea how we can provide it to the user. First I thought we should just prepare example `config.yaml` files and let users edit them. Now I have better idea. We should embed these styles into the vsg. For example user would call vsg in following way: `vsg --style style_name -f file.vhd`. This approach has 2 pros:

1. If you want to use one of the embedded styles you do not need to store `config.yaml` in your repository.
2. If you want to slightly modify the default style, you need to only define an 'overlay' for embedded style. This makes `config.yaml` which you need to store much shorter.

jeremiah-c-leary, 2020-04-05:
@m-kru That makes a lot of sense.  So VSG could provide a series of styles to choose from, and the user could pick which one they want.  If there are modifications, then they just make a configuration and pass it via the  `-c` option.

```bash
 $vsg --style minimal -c test_config.yaml
 $vsg --style legacy -c my_config.yaml
```
In those cases, the style would be the base configuration and the `-c` would modify it.

closed 2020-06-23 by jeremiah-c-leary