jeremiah-c-leary/vhdl-style-guide #352: Scrub rules for context

jeremiah-c-leary, 2020-03-18:
In the discussion thread of issue #330 relating to the comment rules, the following guideline was proposed:

> So I would propose if a rule depends on it's context, then it should be grouped with that context. I believe it would make more sense to someone who was browsing the documentation.

The comment rule changes will be handled by issue #351.  This issue will cover the effort to review the existing rules and determine if any others should be moved using the guideline above.

Tasks include:

[ ] Review existing rules
[ ] Propose changes given guideline above
[ ] Implement changes when approved

m-kru, 2020-04-06:
@jeremiah-c-leary as there are a lot of rules to go through, I will do it partially by editing this comment. I will let you know when I end.

**After Rules**
This group seems a little bit weird to be. Looks like all rules are related to the simulation stuff. Maybe we should create Simulation Rules? Just loudly thinking.

**Architecture Rules**
All rules in this group should stay in this group. Maybe some rules from other groups should be moved here.

jeremiah-c-leary, 2020-04-06:
> After Rules
This group seems a little bit weird to be. Looks like all rules are related to the simulation stuff. Maybe we should create Simulation Rules? Just loudly thinking.

That is an interesting point.  We should keep that in mind as you go through the rest of the rules.

jeremiah-c-leary, 2020-04-09:
>Attribute Rules
In my opinion attribute_001 rule should be removed, as such alignment should be ensured by rules that ensure proper alignment in declarative regions.

I had thought about having a single indent rule, but for some reason I thought maybe someone would want to indent differently based on the type.  `library_008` is a perfect example.  Some people where I work like it left justified, but I prefer it indented.

I see what you are saying though, and it makes sense.  You could argue that indenting is just another form of keyword alignment.  If you wanted to indent everything in the architecture declarative at a different value than after the begin statement, you would have to change a bunch of rules instead of just one.  Plus signals can be declared in multiple places.   Hmmm...will definately have to think about this.

> concurrent_002 and concurrent_004

So in looking up information to respond, I believe I have been making a distinction that doesn't exist.  I always assumed...
```vhdl
   architecture RTL of FIFO is
   begin
       full <= q_full;
   end architecture RTL;
```
that `full` was a concurrent statement, but it turns out that processes are concurrent statements also.

So now things get interesting.  I have a `sequential` section of rules that are very similar to `concurrent` rules.  The only difference being sequential rules are inside a process while concurrent rules are outside a process.  When they really should be combined into a `signal_assignment` category.

But to address your comment, we could move it to a `whitespace` rule.

> constant_003 and file_003

The configurable nature of the rule was a request from someone I work with.  This page shows you what I was attempting:  https://vhdl-style-guide.readthedocs.io/en/latest/faq.html#how-do-i-align-signal-003-constant-003-file-003-type-003-subtype-003-and-file-003

Maybe we need a "column" alignment rule that will align words into columns?

> constant_005 and constant_010

There was a time I was getting adamant about trying to move everything to the left as much as possible.  Now I can see how having it be at least one space makes sense.  I feel it should be paired with an alignment rule though.

m-kru, 2020-05-03:
@jeremiah-c-leary I have finished the review process. Here are my extra remarks to what I have already written above.

**Indentation**
I have checked how you have implemented this, and my final thought is that there is no point in changing how it works. There are no strong arguments which approach would be better, and from user perspective there would be no difference.

**Identifiers alignment**
This is a bit messy right now in my opinion. There are rules, that allow enforcing specific number of spaces after such keywords as `constant, signal, file, variable` and there are also some rules trying to align identifiers independently such as `process_31`. We can leave user the choice, but in some configurations the rules would contradict with each other.

> The configurable nature of the rule was a request from someone I work with. This page shows you what I was attempting: https://vhdl-style-guide.readthedocs.io/en/latest/faq.html#how-do-i-align-signal-003-constant-003-file-003-type-003-subtype-003-and-file-003

This is ok, but what if you have only `types` and `files`. It enforces you to have extra spaces:
```vhdl
type     state_machine is (IDLE, WRITE, READ, DONE);
type     my_type is (cat, dog);
file     defaultImage : load_file_type open read_mode is load_file_name;
file     golden : load_file_type open read_mode is load_file_name2;
```
On the other hand, if you add some line with `signal` or `constant` you do not have to realign the identifiers.

By the way, it is the same story with `in, out, inout` and type alignment in the entity port declaration.

Probably the best solution would be to support fixed and dynamic identifiers alignment. Of course user would have to choose which one he prefers. Probably it would be also nice to detect, that user has configured contradicting rules.

I am waiting for your feedback to start working!

m-kru, 2020-05-17:
@jeremiah-c-leary I need your feedback.

jeremiah-c-leary, 2020-06-02:
@m-kru I am going to start looking through your comments on this issue.

jeremiah-c-leary, 2020-06-02:
> Generic Rules
I am a bit confused about this group. Is the intent of this group to gather rules that are common for entity and component declarations? I can see 3 points in which generics are used: entity declaration, component declaration, instantiation. There are two options here. This group can stay or rules should be moved to entity, component, instantiation rules. @jeremiah-c-leary can you please define the goal of this group?

The goal of the generic group was to handle all rules related to generics.  That include indenting, case, alignments and whitespace.  I could see that `generic_001`, `generic_002` and `generic_008` could be moved out of the group.  We would have to then create nine other rules to handle the cases for the entity, component and instantiations.  I am not sure that is worth it.

jeremiah-c-leary, 2020-06-02:
>If Rules
My overall feeling is that most of the rules in this group should stay in this group. However, I wonder if rules if_032 and if_033 should not be moved to Comment Rules. In my humble opinion there should be only one rule in Comment Rules that ensures alignment of consecutive comment only lines.

I can see it either way, where the rule could be either in the `comment` or the `if` group. 

>Now I see what might be the drawback of associating rules with context. Instead of one global rule, you may end up with multiple rules doing the same, but with scope limited to the context.

Those two rules end up adjusting the indent of the comment lines.  VSG runs through the code once and assigns indent levels.  So if the `else` is at an indent level of 2, then the comments are at an indent level of 3.  So those two rules are an exception to the assigned indent.

I think if we had a global rule, it could get really complicated in handling exceptions to the general rule of checking the indent.  I guess the complexity has to live somewhere.  Maybe with a context rule it is a little easier to read?

jeremiah-c-leary, 2020-06-07:
> Source File Rules
In my humble opinion it should not be handled as a rule for vsg. I would opt for making it just Python exception FileNotFoundError: [Errno 2] No such file or directory: 'entity.vhd'.

I remember someone wanting a missing file reported in the JUnit file so if it failed during a CI flow it would be reported.

jeremiah-c-leary, 2020-06-07:
@m-kru Could you check the following list to see if it summarizes how we should complete this issue?

  1) leave the indent rules as they are.

  2) Keep the after rules section.

  3)  create an identifiers alignment rule to cover constant_003, file_003, etc...

  4)  single space after rules change to at least a single space
      constant_005
      constant_010
      etc...

  5)  Merge concurrent rules into sequential rules

  6)  Add a rule deprication feature to communicate upcoming changes.

I was going to create new issue for items 3 through 6 and then close this ticket.

m-kru, 2020-06-08:
1. Yes.
2. Yes/No? The idea about Simulation Rules group is still open?
3. Yes.
4. Yes.
5. Not sure, what is the goal of sequential rules? Maybe they should be moved to Signal Rules, as they are related to the signals assignment?
6. What would it look like?

One more:
What about merging `Variable Assignments` Rules to `Variables` rules? What was the idea for having separate group for assignments?

jeremiah-c-leary, 2020-06-09:
> What about merging Variable Assignments Rules to Variables rules? What was the idea for having separate group for assignments?

Forgot about that one.  I can not remember why I had them separate.   It may have been because I needed a way to differentiate between them because the rules are different.

`Variable` rules look for the `variable` keyword and use the `isVariable` line attribute.
`Variable Assignment` rules look for the `:=` keyword and use the `isVariableAssignment` line attribute.

Now whether that is justification enough to put them in different groups is debatable.  Is there a distinction made in the VHDL language reference manual?

> 5. Not sure, what is the goal of sequential rules? Maybe they should be moved to Signal Rules, as they are related to the signals assignment?

I think in the language reference manual they are called sequentials. which I believe is where I read that what I thought were concurrent statements are really sequential statements which happen to also be concurrent statements.  Processes are current statements also.

> 2. Yes/No? The idea about Simulation Rules group is still open?

Can you think of anything else that would fit in a simulation group?  I kind of like the `after` because that is the keyword the rules operate on.

> 6. What would it look like?

I am thinking an attribute on the `rule` class called `depricated`.  Maybe it holds a string to communicate to the user.  For example, if we plan to merge concurrent rules into sequential rules we would set the `depricated` attribute to something like:

```python
from vsg.rules import indent_rule
class rule_001(indent_rule):
    '''
    Concurrent rule 001 checks for the proper indentation at the beginning of the line.
    '''
    def __init__(self):
        indent_rule.__init__(self, 'concurrent', '001', 'isConcurrentBegin')
        self.depricated = 'This rule is scheduled to be removed in a future release.  It will be replaced by rule sequential_001'
```
Then when VSG is ran it would output something at the end like:
```mono
NOTE:  The following rules are being depricated:
   concurrent_001 : .....
   concurrent_002 : ....
   concurrent_003 : ...
```
Maybe there is a `--suppress_depricated_warnings`
Or we could just report:
```mono
NOTE: Rules have been scheduled for deprication.  Use --report_depricated_rules for more information.
```

jeremiah-c-leary, 2020-06-24:
> Not sure, what is the goal of sequential rules? Maybe they should be moved to Signal Rules, as they are related to the signals assignment?

So I looked at the LRM again and there is a section for sequential signal assignments:
[image]
and concurrent signal assignments:
[image]
and those would be different than the signal definition lines.

So I think I will leave the concurrent and sequential rules as they are.

jeremiah-c-leary, 2020-07-18:
Hey @m-kru ,
I was going to close this issue because I released version 2.0.0.

closed 2020-07-18 by jeremiah-c-leary