jeremiah-c-leary/vhdl-style-guide #353: Default configurations

jeremiah-c-leary, 2020-03-18:
A discussion on gitter with @nfrancque proposes having a layered configuration approach where the user can run with default configurations without having to use the `--configuration` option.  The proposal includes:

[ ] Having a user defined configuration stored in their home account stored as a `.vsgrc` file
[ ] Having a project based configuration possibly set using an environment variable `VSG_PROJECT_CONFIG`

The current proposal would be:

1. use the `.vsgrc` file if it exist and the environment variable `VSG_PROJECT_CONFIG` is not set.
2. Use the configuration defined by the environment variable `VSG_PROJECT_CONFIG` if it is set and ignore the `.vsgrc` file if it exists

We need to determine the interaction of the `--configuration` option when either the `.vsgrc` file or the `VSG_PROJECT_CONFIG` file exists.

I propose `--configuration` has highest priority and we ignore the `.vsgrc` and environment variable.

VSG should also print out which configurations it used.  This should make it clearer to the user when trying to debug why VSG made changes.

nfrancque, 2020-03-21:
Another thought on this, unsure how it would look like implemented.  

There are many cases where you need to pull in IP written against a different coding standard.  It would be nice if VSG could discover that the most local config that exists and use that.  
project
| vsg.yaml
| block1/
----| vsg.yaml
----| block1.vhd
----| block1_subblock1/
--------| block1_subblock1.vhd
| block2/
----| block2.vhd

Block2 should take on the project config, whereas block1 (and everything underneath it) should take on block1's  config.

I am unsure whether that should only happen where they conflict (i.e., block1 inherits the project config unless block1 overrides it), or if it should only use the lowest level of rules possible.

jeremiah-c-leary, 2020-03-28:
Currently VSG must be told which configuration to use, it doesn't do any discovery.  If it did do discovery, then VSG would pick the lowest level configuration and run it.  I think that makes the most sense.

If you merged the project and block level configurations, you could end up with something very interesting.

VSG will merge configurations passed by the `-c` option, but now I wonder if that was a good idea.

jeremiah-c-leary, 2020-03-28:
> It would be nice if VSG could discover that the most local config that exists and use that.

Is this a new feature request?

jeremiah-c-leary, 2020-03-28:
I am thinking of changing this to drop the `.vsgrc` option and just use an environment variable `VSG_CONFIG`.  This allows the most freedom to the user and minimizes confusion on which configuration is used.

The user can create a `.vsgrc` file if they want and point `VSG_CONFIG` to it in the startup script.  This variable can then be overwritten by a project to point to their configuration file.

here is a proposed vsg configuration file format:
```yaml
    command_line_arguments:
    
      configuration :
        - first.cfg
        - second.cfg
    
      local_rules :
        - LOCAL_DIR
    
      output_format :
        - syntastic
    
      backup : False
```

It would configure the operation of VSG versus any rules.  VSG would read this configuration first and then override them with anything that was passed to it.

>We need to determine the interaction of the --configuration option when either the .vsgrc file or the VSG_PROJECT_CONFIG file exists.

>I propose --configuration has highest priority and we ignore the .vsgrc and environment variable.

I think I answered my question here.  If the `--configuration` option is used then it would only override the `configuration` option in the vsg project configuration.

m-kru, 2020-04-03:
One remark from me. Do not trash my home directory. Instead of `.vsgrc`, you should use `/home/user/.config/vsg/config.yaml`.

Emilv2, 2021-01-01:
Any updates on this? I personally like the way [isort](https://pycqa.github.io/isort/docs/configuration/config_files/) deals with this.

VSG could do this for every file it encounters to allow for different coding standards, but is this really necessary since you can already specify per file configuration in the top configuration file?

I think merging configuration will only cause confusion.

jeremiah-c-leary, 2021-01-01:
Hey @Emilv2,

This has stalled for the moment.  I did implement something on the issue_353 branch, but it was shelved to focus on other things.  The odds of this being implemented are pretty low.

--Jeremy

Emilv2, 2021-01-02:
Do you accept contributions?

What I personally would require of (both automatic discovery and supplying with `--configuration`) configuration:

1. running vsg on a file with the same configuration file should always give the same output, no matter from which directory (in the same project) you run it. 
2. running vsg with files specified in the configuration file should always give the same output, no matter from which directory (in the same project) you run it. 
3. vsg should only run on files in the current directory and it's subdirectories (unless you explicitly do it with `--filename ../`)
4. if files are specified with the `--filename` option only process those files
5. if a configuration file is specified with the `--configuration` option only use that configuration file and do not search for configuration files

Currently those requirements are not satisfied. The most troublesome part is the files part in the configuration and having the possibility of different configuration per file, that makes things pretty complicated.

I thought of some possibilities I could try to implement, with their own advantages and disadvantages:

* a. The simplest option. Remove the option to have different settings for different files and remove the `file_list` in the configuration file. vsg will search for the closest configuration file and use that. Adding to that running vsg `--filename .` could find all vhdl files in subdirectories, and maybe also follow `.gitignore` (and similar files of other vcs) so that auto generated files can be ignored.  IP written against a different coding standard is simply converted to your coding standard, this is done automatically so not really an issue.
* b. Similar to a, remove the `file_list` but also find the closest configuration file for each file that is given with `--filename`. This would allow for different configurations for different files in one project, but only if they are in a different folder.
* c. b, but also take higher level configuration files in regard and merge the two (with the closest taken precedence). The problem with this is determining when to stop searching for configuration files.
* d. Only one configuration file per project with all the files in `file_list` and their different configurations. When `--filename` is used the relative location between the configuration file and the vhdl file is determined so the correct configuration is used.
* e. Somehow try to combine b, c and d.
* f. An environment variable. I don't really like this because it's less transparent, you can't put environment files in source control. It also doesn't really solve the issues with the `file_list` in configuration or different configuration per file. 

What do you think of this?

jeremiah-c-leary, 2021-01-06:
Hey @Emilv2 ,

> Do you accept contributions?

Of course.  I welcome all contributions.

A little history on configurations:

The configurations came about to allow users to apply exceptions to the predefined rules.  Eventually the per file exceptions in the configuration file was requested.  At the outset the configurations were intended to be mergeable to allow users the  ability to create a layered configuration approach.

One use case which drove this thinking was design versus verification code.  Where the design code and verification code would have rules in common, but also rules that would be different.  Thinking about it now it doesn't seem to make much sense.

Another use case was with different code bases used in the same project.  If you can not, or are unwilling to, change part of the formatting of a code base, you could still enforce a code style for that portion.  Although in this case you would not want to merge configurations as that would then change the rules used against the code.

I believe others, like @staerz, have used a separate configuration file to isolate rules which are failing for his code base to turn them off until they can be addressed.

At my work, we have individual configurations for blocks in our design.  When I added the rule `length_001`, it started failing a bunch of files.  Someone was able to create a configuration that just disabled that rule, and then added it to a script each vsg command in a script we use in our bamboo run.  Maybe in that case editing the individual configurations would have been a better idea.

I can not remember why I added the `file_list` to the configuration.  It was probably because I wanted to reduce the `--filename` command line option.  With the number of options, the command line can get quite long.  I think I wanted a history also, where the files and the rule configuration exceptions would be in one place.  One could argue, and probably successfully, that it should be handled above VSG.

I also added `indent` to allow for configuration of how VSG indents.

I think that covers the history and thinking behind how configurations were implemented.

Now, there are some issues with it.

  1)  The merging, while providing flexibility also adds complexity and can obfuscate which rules are actually applied.
  2)  The file names in `file_list` are defined relative to where VSG is executed.  (we use environment variables to address this)
  3)  There are probably other issues I can't think of right now

I apologize if this is a little long winded.  Sometimes I find it useful to provide some context so we can be on the same page when it comes to possibly changing the behavior of a feature.

If find your approach to the self discovery interesting and could provide the preferred method going forward.  It looks like it could live in parallel to the existing configuration methods.

So given all of that...

>What I personally would require of (both automatic discovery and supplying with --configuration) configuration:
>
> 1. running vsg on a file with the same configuration file should always give the same output, no matter from which directory (in the same project) you run it.

agree

> 2. running vsg with files specified in the configuration file should always give the same output, no matter from which directory (in the same project) you run it.

agree

> 3. vsg should only run on files in the current directory and it's subdirectories (unless you explicitly do it with --filename ../)
if files are specified with the --filename option only process those files

Would there be a way to exclude files?

> 4. if a configuration file is specified with the --configuration option only use that configuration file and do not search for configuration files

agree.  This maintains backwards compatibility

> The most troublesome part ... and having the possibility of different configuration per file, that makes things pretty complicated.

Are you referring the ability to configure rules per file in the configuration or that you could end up with multiple configuration files (worst case, one per file)?

> I thought of some possibilities I could try to implement, with their own advantages and disadvantages:

> a. The simplest option. Remove the option to have different settings for different files and remove the file_list in the configuration file. vsg will search for the closest configuration file and [...]

What would be the definition of closest configuration file?  Same directory or the first one in a parent directory? Is there a limit to how far you would go up the directory tree?

I like the `--filename .` idea.  Globbing is currently supported:  `--filename *.vhd`, but that is only at the current level.

>b. Similar to a, remove the file_list but also find the closest configuration file for each file that is given with --filename. This would allow for different configurations for different files in one project, but only if they are in a different folder.

Maybe combined with a `.gitignore` equivalent like in a?

>c. b, but also take higher level configuration files in regard and merge the two (with the closest taken precedence). The problem with this is determining when to stop searching for configuration files.

Agree it would be hard to know when to stop.  Maybe you only merge if multiple configurations are given at the command line.  Otherwise do not  merge.

[... cut ...]
Would the configuration for each file be placed in a configuration file?

> e. Somehow try to combine b, c and d.
> f. An environment variable. I don't really like this because it's less transparent, you can't put environment files in source control. >It also doesn't really solve the issues with the file_list in configuration or different configuration per file.

Agree on f.  An environment variable would not be a good choice.

I think your ideas could really improve the usability of VSG.

Regards,

--Jeremy

staerz, 2021-01-06:
Hi @jeremiah-c-leary, @Emilv2,

since I was mentioned, allow me to lay out how I use VSG and set up all the configuration thing around it.

First, let's start with a little context:

Our project is organised in a git repo with multiple git submodules.
Each submodule contains code that VSG is running on, and there is a fixed directory structure for VHDL source files - the same for any submodule.
This is actually not necessary, but simplifies many things.

The VSG configuration is managed centrally in the root git repo, with each git submodule adding their particular configuration.

In particular, the root repo has a `.vsg` directory that contains a subdir `rules` where all rules are stored in dedicated files.
In there, I have:
- `alignment.json`
- `casing.json`
- `config.json`
- `disabled.json`
- `indentation.json`

Now these files simply contain each a set of VSG configurations for rules that fit the purpose, i.e. in `casing.json` I have all the settings for rules that decide on lower or upper case; and `config.json` contains any "other" configuration settings that don't fit anywhere else.
For now, `indentation.json` holds a list of rules that are disabled simply because the indentation is messed up (for my preference) if they were active - my idea is to re-enable them all, once VSG rules are compliant with what I expect (there are numerous issues I opened with that regard, @jeremiah-c-leary knows very well).

When running VSG, I simply go to the root of my repository (or the root of any git subrepo) and call `vsg $(cat .vsg/params.txt)` and that's it.
Now what is key at this point is the content of this `params.txt`, which is simply the following:
```
-c .vsg/rules/config.json .vsg/rules/disabled.json .vsg/rules/alignment.json .vsg/rules/casing.json .vsg_local/configuration.json -lr .vsg/rules
```

You'll see 2 things that stick out of the crowd:
1) `-lr .vsg/rules`: As VSG allows, one can define user rules. We had some, but now they are part of the VSG default rules, so we could remove them. But it doesn't hurt to leave the option open.
1) `.vsg_local/configuration.json`: This requires a particular file per repository to exist that simply contains the `files_list`. The beauty about that is that it still allows to make further settings, i.e. disabling a certain rule, on particular files if needed. Of course, ideally this should be the pure list of files only, given that all rules should always be applied.

For this to work each git subrepo has:
- the `.vsg` directory being a soft link to the super-project's `.vsg` directory
- a true `.vsg_local` directory holding the `configuration.json` file

The super-project has true `.vsg` directory holding the files as listed above (plus the local `configuration.json`); and a soft link `.vsg_local` to it.

By this setup I have always been able to globally set rules or disable some (if they were leading to problems in our code), or do local adjustments for particular files in any git sub repo.
Needless to add that we have a CI to test all that for all git subrepos independently.

Although that works very well, and for my purposes I think I'm settled, allow me to comment on your proposal.

Yes, it would be a feature gain if hierarchical configuration would be added.
Usually the approach would be that local rules always take precedence over global rules; and the most global rule would be the default rule that comes with VSG, the second most global rule would be the user's VSG config file; and ultimately, the most local rule would be those provided via command line arguments.

If we consider a simple project (I copied from the original post):
```
project
├ vsg.yaml(p)
├ block1/
│  ├─ vsg.yaml(b1)
│  ├─ block1.vhd
│  └─ block1_subblock1/
│      ├─ vsg.yaml(sb1)
│      └─ block1_subblock1.vhd
└ block2/
   └─ block2.vhd
```
and assume that there is a default VSG configration `vsg_default.yaml` and a user configration `vsg_user.yaml` then the following should apply to the 3 `vhd` files in question:
- `block2.vhd` applies `vsg.yaml(p)`. Rules not set by `vsg.yaml(p)` are completed by rules set by `vsg_user.yaml`. Any other now still undefined rules are taken from `vsg_default.yaml`.
- `block1_subblock1.vhd` applies `vsg.yaml(sb1)`. Rules not set by `vsg.yaml(sb1)` are completed by rules set by `vsg.yaml(b1)`. Rules not set by these are completed by `vsg.yaml(p)`. ... then `vsg_user.yaml`, then `vsg_default.yaml`.
- `block1.vhd` applies `vsg.yaml(b1)`, ... you get the idea.

VSG should hence always look up each level and check if it finds more rules. The end for such a (local) search is the root directory (`/`). Only user rules and finally VSG default rules then complete missing configuration.

One can though argue about "user" rules: I'm actually strictly against user rules and hence one could drop "user" rules from the described chain of precedence as user rules are likely not portable. But if we assume the above example (and no user rules), then the result would be the same no matter where VSG runs, as long as it's the same version of VSG (hence identical "default" rules). You can even check out that project on a different machine, simply make sure to use VSG in that same version and voilà, get the exact same result (assuming that there was no `vsg.yaml` in the tree above `project`).

[... cut ...] ncountered in the directory tree that VSG is supposed to work on.

And @jeremiah-c-leary, one note on command line arguments: Please consider changing your long options to true long options, i.e. `-lr` is a long option and should actually be `--lr`; the reason is that `-lr` is the combination of short options `-l` and `-r` (example: `git commit -am "Fancy commit message"` which is identical to `git commit -a -m "Fancy commit message"`). I think VSG 3.0.0 would be a good chance to fix that :wink:

Emilv2, 2021-01-06:
> -lr .vsg/rules: As VSG allows, one can define user rules.

About that, it's not very clear to me yet what the difference between a `local_rules` file and a configuration file is

> We had some, but now they are part of the VSG default rules, so we could remove them. But it doesn't hurt to leave the option open.

I think `--debug` (or maybe a `--verbose` with different levels) could warn about users setting default rules. I'd think warning on unknown rules and deprecated options would be more user friendly than exiting like what happens now. And also at least warn on invalid settings (e.g. `case: edge`) instead of silently ignoring them. But that's maybe for another discussion.

> * block2.vhd applies vsg.yaml(p). Rules not set by vsg.yaml(p) are completed by rules set by vsg_user.yaml. Any other now still undefined rules are taken from vsg_default.yaml.
> * [...]

>VSG should hence always look up each level and check if it finds more rules. The end for such a (local) search is the root directory (/). Only user rules and finally VSG default rules then complete missing configuration.

Would that not make things very complicated quickly? That means there's a whole bunch of possible locations of files that can influence your files. I think only looking at one configuration file and not going beyond the git/hg/.. directory would be clearer. It would also mean that projects with a configuration file will not be influenced by user configuration. user configuration is only applied when no configuration is found.

> And, notably, this procedure of configuration would be the alternative to the current way. I think trying a mixture of both would lead to unnecessary complications. I.e. I'd say VSG should complain [...]

My pull request #507 tries to work with both procedures (what I proposed as option e), and I think it could work out better than I initially expected. I agree that VSG should complain if you're trying to combine the two workflows.

Emilv2, 2021-01-06:
> I can not remember why I added the file_list to the configuration. It was probably because I wanted to reduce the --filename command line option. With the number of options, the command line can get [...]

If no one actually uses the `file_list` it would probably easier to let it go, but I think it's actually not that bad.

> I apologize if this is a little long winded. Sometimes I find it useful to provide some context so we can be on the same page when it comes to possibly changing the behavior of a feature.

Context is good. Personally I'm used to python tools and thus that's what I compare VSG with. ([black](https://github.com/psf/black) is kind of the equivalent of VSG for Python but it took completely the opposite approach: Nearly no configuration possibilities. VHDL context is of course different, but for Python it's actually amazing not needing to think about coding style configuration)

> Would there be a way to exclude files?

I would at least ignore files in `.gitignore` and `.hgignore` (and an option to disable that), and maybe add another way of ignoring files if that's necessary. Or do you mean something else?

> Are you referring the ability to configure rules per file in the configuration or that you could end up with multiple configuration files (worst case, one per file)?

I mean that with the globbing rules it is possible to have one file configured in multiple configuration files.

> What would be the definition of closest configuration file? Same directory or the first one in a parent directory? Is there a limit to how far you would go up the directory tree?

Same directory, then parent, etc. Go up to the git/hg root directory or a reasonably high number (25 right now).

staerz, 2021-01-06:
> > -lr .vsg/rules: As VSG allows, one can define user rules.
> 
> About that, it's not very clear to me yet what the difference between a `local_rules` file and a configuration file is

Well, a rule is a formatting / style rule, e.g. `port_map_009` - just like VSG has about 500 by now.
But you still may have a construct for which there is no rule - or, you have a VSG rule that you don't like and define your own instead of it. This would go into these rules.

A configuration though is anything that simply sets some parameters to defined rules, e.g. [`length_001` can be configured to check for lines not exceeding the value that you specify](https://vhdl-style-guide.readthedocs.io/en/latest/configuring.html#configuring-length-rules).

Also, part of this configuration is if you'd want to disable a set of default rules - or enable a few rules which are disabled by default.
That, of course, only applies in the current situation where VSG will apply all rules by default.

 
> I think `--debug` (or maybe a `--verbose` with different levels) could warn about users setting default rules. I'd think warning on unknown rules and deprecated options would be more user friendly [...]

I'd say there is no need to implement sth. new since this feature already exists: Simply use the `-oc` option to dump the configuration.
OK, agreed, maybe to spot self-defined user rules (so non-VSG-standards) could be extracted somehow in a dedicated file with a new option.

> Would that not make things very complicated quickly? That means there's a whole bunch of possible locations of files that can influence your files. I think only looking at one configuration file and [...]

"user configuration is only applied when no configuration is found." - yes, I'd say that should be it, if even user configuration is (should be) allowed.

"not going beyond the git/hg/..": Well, I find that very difficult to implement: First, you would have to support any versioning options (git, hg, svn, cvs, ...), second you'd have to detect the repo boundary and last, this boundary may not be a real boundary (see my case where I use git submodules).

"Would that not make things very complicated quickly?" - I'd argue that it makes it very straight forward and very clear actually. And I'm not talking about the implementation, but the fact that the rule is very simple to understand for everybody (user perspective).

> My pull request #507 tries to work with both procedures (what I proposed as option e), and I think it could work out better than I initially expected. I agree that VSG should complain if you're [...]

I still see a problem with dropping the feature of `file_list`: Files in the same directory may still need a different configuration - how do you incorporate that in your approach (maybe I missed it)?

staerz, 2021-01-06:
> If no one actually uses the `file_list` it would probably easier to let it go, but I think it's actually not that bad.

I use it :rofl:

> > Would there be a way to exclude files?
> 
> I would at least ignore files in `.gitignore` and `.hgignore` (and an option to disable that), and maybe add another way of ignoring files if that's necessary. Or do you mean something else?

Good point - but ignoring should be done explicitly, but I doubt that `.gitignore` would be helpful here. The list of files to ignore should be part of the configuration of VSG.

> > Are you referring the ability to configure rules per file in the configuration or that you could end up with multiple configuration files (worst case, one per file)?
> 
> I mean that with the globbing rules it is possible to have one file configured in multiple configuration files.

That is correct and the behaviour is such that you must watch out for your globbing. I can only tell you that it currently works like a charm, if you do your globbing correctly - and especially when you want the "anti"globbing.

> > What would be the definition of closest configuration file? Same directory or the first one in a parent directory? Is there a limit to how far you would go up the directory tree?
> 
> Same directory, then parent, etc. Go up to the git/hg root directory or a reasonably high number (25 right now).

That is what I was suggesting as well, modulo the remark on the repo boundary.

Emilv2, 2021-01-06:
>I'd say there is no need to implement sth. new since this feature already exists: Simply use the -oc option to dump the configuration.

I would think warning about setting defaults or setting wrong values is a bit more user friendly than just dumping configuration and comparing by hand, especially if you have set many rules over different files.

> "not going beyond the git/hg/..": Well, I find that very difficult to implement: First, you would have to support any versioning options (git, hg, svn, cvs, ...), second you'd have to detect the [...]

Only versioning options that are actually used by a significant userbase would need to be added. Checking for git, hg and svn repositories is trivial, and even if you use something else it could still work if you put an empty `vsg_conf.yml` in you root directory (That is, if you don't merge config files, otherwise you would just need to make sure the rest of the path is clear to achieve the same result). I think that even with submodules it makes sense, I would think a submodule should be self contained and not take configuration from the "supermodule". As I understand that is also how you use it?

> "Would that not make things very complicated quickly?" - I'd argue that it makes it very straight forward and very clear actually. And I'm not talking about the implementation, but the fact that the rule is very simple to understand for everybody (user perspective).

But that would mean if you have a vsg_conf.yml somewhere in the path between / and your project, your project will be affected. Imagine you have a project that has it's own `vsg_conf.yml` and you have a git submodule with differently formatted code: running `vsg -f .` will apply your formatting to the code of the submodule if you don't check for repository boundries and/or merge configuration files. Also the configuration for one file could be scattered over several configuration files, I think having only one configuration file that configures a file is simpler overall.

>I still see a problem with dropping the feature of file_list: Files in the same directory may still need a different configuration - how do you incorporate that in your approach (maybe I missed it)?

That would indeed be a problem with dropping the file_list, in my current pull request I have not dropped file_list.

>And a note on #507: Maybe it's a good idea to separate fixes for trailing whitespace into a dedicated MR from the actual feature implementation. I'm sure that @jeremiah-c-leary would simply apply the whitespace fixes (at his convenience).

True, that was a bit sloppy of me.

> Good point - but ignoring should be done explicitly, but I doubt that .gitignore would be helpful here. The list of files to ignore should be part of the configuration of VSG.

I was thinking that maybe auto generated files for example would already be in .gitignore and probably should not be formatted by VSG. I'm fine with either explicitly or default, my experience is that several tools (ag/rg/black that I know of) already do ignore .gitignore by default and people seem to be happy about that.

>That is correct and the behaviour is such that you must watch out for your globbing. I can only tell you that it currently works like a charm, if you do your globbing correctly - and especially when you want the "anti"globbing.

I think that at the least vsg should behave consistently if you happen to do weird things like having files matched by different globs in different configuration files and probably also warn about that.

All that said, I'm happy to change my pull request to what the consensus and @jeremiah-c-leary decide is the best approach.

staerz, 2021-01-06:
> I would think warning about setting defaults or setting wrong values is a bit more user friendly than just dumping configuration and comparing by hand, especially if you have set many rules over [...]

OK, so in that perspective it would be sth. like a `--debug` output. Then agreed, it should be most user friendly.
But I estimate this to be a task of its own.

> Only versioning options that are actually used by a significant userbase would need to be added. Checking for git, hg and svn repositories is trivial, and even if you use something else it could [...]

No, that is not how I use it: I have a distributed approach: The general base of rules and their configuration is given in the super-project (and only there) and all git sub-repos "include" it (by the soft link to the configuration files). Each sub-repo though has a `configuration.jston` where the `file_list` is defined (that may then disable a certain subset of rules for particular files, which is to prevent any false VSG complaints right now, the long-term aim is that this should only be a flat list of files - or, not even needed as VSG would find the files itself as proposed by this issue here).

> > "Would that not make things very complicated quickly?" - I'd argue that it makes it very straight forward and very clear actually. And I'm not talking about the implementation, but the fact that the rule is very simple to understand for everybody (user perspective).
> 
> But that would mean if you have a vsg_conf.yml somewhere in the path between / and your project, your project will be affected. Imagine you have a project that has it's own `vsg_conf.yml` and you have a git submodule with differently formatted code: running `vsg -f .` will apply your formatting to the code of the submodule if you don't check for repository boundries and/or merge configuration files. Also the configuration for one file could be scattered over several configuration files, I think having only one configuration file that configures a file that can only be in the project's repository is simpler overall.

Agreed. So that should drive us to the conclusion that a "lose" spread of VSG configuration files is actually a very bad idea. The only way to make it static is to
a) do as now and pass all configuration files as parameters,
b) allow "lose" configuration with the side node that a particular file (best identified by a unique file name) would mark the "top end" of the search path (let's say that your project must contain a `.vsg_config.json` as root config file, and any subdirectory could have any additional `vsg_config.json` ... then crawling up would eventually find `.vsg_config.json` and know that this is the boundary)
c) go via an include-only based approach where the "top end" is simply defined by not finding any more `include`s

> I was thinking that maybe auto generated files for example would already be in .gitignore and probably should not be formatted by VSG. I'm fine with either explicitly or default, my experience is [...]

I'm not a friend of implicit rules. But I fully agree, you would want a way to be able to exclude any auto-generated files - or files that you pull in from a foreign project or any other source where you are not the author.

> I think that at the least vsg should behave consistently if you happen to do weird things like having files matched by different globs in different configuration files and probably also warn about [...]

That would in deed be a good feature: If VSG would warn about multiple (contradicting) configurations for a file, that would at least allow the project owner to easily spot an intended misconfiguration! That warning could even be part of the regular run of VSG, I'd say (but should be a warning, not altering the exit code from 0, and not an error, I'd say; although that this could even be debatable).

jeremiah-c-leary, 2021-01-08:
Hey @Emilv2 and @staerz ,

Work has been a bear this week and I am now getting back to this.  It looks like some great discussions.

I comment on some easier stuff first:

> And @jeremiah-c-leary, one note on command line arguments: Please consider changing your long options to true long options, i.e. -lr is a long option and should actually be --lr; the reason is that [...]

I agree the command line options should change and 3.0.0 would be the time to do it.  If you have time, could you create an issue with some proposals.

Also, did we address all the stderr issues?

>> If no one actually uses the file_list it would probably easier to let it go, but I think it's actually not that bad.
>
> I use it 🤣

At some point we have to draw a boundary over what VSG is responsible for and what the user is responsible for.  The `file_list` is pretty handy, but I wonder if I crossed a boundary with it.  It could be handled outside of VSG with another script.  Then the user would be completely responsible for it.  That might have the side effect of making VSG harder to use though.

>  (black is kind of the equivalent of VSG for Python 

I should look into using black, as you can tell from my trailing whitespace issue. 😃 

> One can though argue about "user" rules: I'm actually strictly against user rules and hence one could drop "user" rules from the described chain of precedence as user rules are likely not portable.

I agree with not having "user" rules.  It would be best if all rule configurations are contained in the project and under configuration control.

> And also at least warn on invalid settings (e.g. case: edge) instead of silently ignoring them. But that's maybe for another discussion.

That should be fairly easy to implement.  All the possible configurable parameters are stored in a list in the `rule` class.  When the `configure`  method is called, it could check if the value given is in the list.  If not it could throw and error.  We would have to have an exception for the prefix and suffix rules, but that should not be difficult to implement.

jeremiah-c-leary, 2021-01-08:
So after reading through the discussions several times, it occurred to me that we are talking about the difference between implicit and explicit operation.  Currently VSG operates explicitly.  You must tell it exactly what to do:  which files to operate on, which configurations to use, etc...

@Emilv2 is proposing to add some implicit behavior:  searching for files and configurations.  I think these are the basic rules for these behaviors:

 1)  search for configurations up the directory tree
 2)  search for files to operate on and configurations going down directory tree.

The merging of configurations is a non-trivial task, as the order can matter since the last configuration has highest priority.  It would seem this is not well suited for an implicit behavior.  It would require the user to set it up.  Similar to @staerz's layered approach.  So I would advocate not merging configurations when operating implicitly.

Either explicit or implicit behavior could be inferred from the `--filename` and `--configuration` command line arguments.

```mono
    --filename      |    --configuration   |    Action
--------------------+----------------------+-------------
       not given    |       explicit       |   run configuration given by --configuration
       not given    | implicit             | search for configurations at current directory and below
       explicit     |  not given           |  run on files given by --filename
       implicit     |  not given           | run on files at current directory and below using default VSG rules
       implicit     | implicit             |  search for nearest configuration in current directory and up, then search for files in current directory and below.  If a configuration is found which searching for files, then use that configuration.
       explicit     | explicit             | operate on files given by --filename using configurations given by --configuration
       implicit     | explicit             | search for files in current directory and below using configurations given by --configuration
       explicit     | implicit             |  search for nearest configuration in current directory and up on files given by --filename
```
We would need to define what `implicit` and `explicit` look like.
For the file search, I think you would look for files with an extension of `.vhd` or `.vhdl`.  I have seen `.vho` files.
For the configuration search, we would define a specific file name for the configuration.  Several have been proposed:  `vsg_conf.yml` or `vsg_config.json`.

Maybe to address the auto-generated files we could add a `file_exclude` section to the configuration file.  Since configurations are normally close to their code, it keeps everything together.  You do not need to know to look in separate file to find the files to exclude.

That reminds me of an issue I had recently with the indent updates.  I had everything working and committed.  However, when you tried to install it using `setup.py` a yaml file was not being copied to the site-packages directory.  It took me awhile to figure it out.  I had to add an inclusion to the `MANIFEST.in` file.  It was frustrating trying to figure that out because I rarely have to interact with that file.

Does this resolved the "how do we know when to stop"  question?  Without merging you just stop at the first configuration you find on your way up and then replace as you go down.

Emilv2, 2021-01-08:
> Either explicit or implicit behavior could be inferred from the `--filename` and `--configuration` command line arguments.
> 
> ```
>     --filename      |    --configuration   |    Action
> --------------------+----------------------+-------------
>        not given    |       explicit       |   run configuration given by --configuration
>        not given    | implicit             | search for configurations at current directory and below
>        explicit     |  not given           |  run on files given by --filename
>        implicit     |  not given           | run on files at current directory and below using default VSG rules
>        implicit     | implicit             |  search for nearest configuration in current directory and up, then search for files in current directory and below.  If a configuration is found which searching for files, then use that configuration.
>        explicit     | explicit             | operate on files given by --filename using configurations given by --configuration
>        implicit     | explicit             | search for files in current directory and below using configurations given by --configuration
>        explicit     | implicit             |  search for nearest configuration in current directory and up on files given by --filename
> ```
> 
> We would need to define what `implicit` and `explicit` look like.
 
`--filename`/`--filename` with no arguments as implicit flags? (I don't know if argparse supports this). But I would give  `run on files at current directory and below using default VSG rules` an explicit flag.

>For the configuration search, we would define a specific file name for the configuration. Several have been proposed: vsg_conf.yml or vsg_config.json

I prosose using `vsg_conf.yml`, it's easier to for humans to read and write, and also supports comments. People with existing json config can simply rename their `vsg_conf.json` to `vsg_conf.yml` (json is valid yaml, explains why I couldn't find any refererence to json config in the code) or automatically convert it to yaml. Only using yaml also solves the question about what to do when both files are present. I would also remove the `vsg_conf.yaml` from my pull request, in retrospect it can only lead to confusion. At most we should warn when `vsg_conf.yaml` or `vsg_conf.json` is found.

>Maybe to address the auto-generated files we could add a file_exclude section to the configuration file. Since configurations are normally close to their code, it keeps everything together. You do not need to know to look in separate file to find the files to exclude.

That seems good.

>Does this resolved the "how do we know when to stop" question? Without merging you just stop at the first configuration you find on your way up and then replace as you go down.

Yes, for the most part. I still think that also stopping at the repository root is a good feature and relying on configuration beyond your repository boundary (be it a git submodule or the top module) is simply a sign of bad design, but you decide what goes in :)
(~I would say that in the case of @staerz it makes more sense to either not use submodules or change the super-project and submodule around, the vsg configuration is a softlink to the configuration submodule in each project. Rules that don't work yet can be commented out in yaml, if there are only a select number of files that don't work yet they can get their own configuration file.~ Doesn't even matter that much, if you softlink `.vsg_conf.yml` to `../vsg_conf.yml` you could go beyond your git repository if you really want to do that)

jeremiah-c-leary, 2021-01-08:
> --filename/--filename with no arguments as implicit flags? (I don't know if argparse supports this). But I would give run on files at current directory and below using default VSG rules an explicit flag.

The "not given" was supposed to infer this:
```mono
vsg --configuration vsg_conf.yml
vsg --filename fifo.vhd
```
Where the first call is missing the `--filename` and the second is missing the `--configuration`.

> I prosose using vsg_conf.yml,

I am fine with that.

> Yes, for the most part. I still think that also stopping at the repository root is a good feature and relying on configuration beyond your repository boundary (be it a git submodule or the top module) is simply a sign of bad design, but you decide what goes in :)

What are the details for finding the working copy root?  I think on SVN you can get it with `svn status .`

Also, what do you do if we can not determine we are in a working copy?

staerz, 2021-01-08:
I would really not opt for any implementation basing the configuration on any repository boundary - may it be git, hg, svn or what ever.

The simple reason is that I don't think you can make this bullet proof: There will always be some edge cases that are very hard to consider properly. Think about a git repo with submodules: Somebody might want to have a configuration per submodule and another project (mine) might have a distributed configuration and another project might have the configuration entirely managed in the main repo.
Not to mention any case where a git repo includes an svn repo (you could imagine any combination).

How do you deal with all that - if not via an independent, self-consistent setup via dedicated VSG files?

Emilv2, 2021-01-08:
>What are the details for finding the working copy root? I think on SVN you can get it with svn status .

Checking for `.git`, `.hg` or `.svn` folders should work.

>Also, what do you do if we can not determine we are in a working copy?

Search until the max depth (or `/`? ) and if we find nothing apply default config.

>I would really not opt for any implementation basing the configuration on any repository boundary - may it be git, hg, svn or what ever.

> The simple reason is that I don't think you can make this bullet proof: There will always be some edge cases that are very hard to consider properly. Think about a git repo with submodules: Somebody [...]
Not to mention any case where a git repo includes an svn repo (you could imagine any combination).

>How do you deal with all that - if not via an independent, self-consistent setup via dedicated VSG files?

I just don't see why you ever would want to do that.
What's the use of a submodule if it depends on the top module? If you're using it somewhere else your config is gone. If you don't, why go through the hassle of using submodules. If you want to reuse a config in multiple different projects just do it the other way around and make the config repository the submodule. And if you really want to do it that way you can still simlink the config file to the top repository config file and get what you want. That's how I see it anyway.

staerz, 2021-01-08:
> I just don't see why you ever would want to do that.
> What's the use of a submodule if it depends on the top module? If you're using it somewhere else your config is gone. If you don't, why go [...]

Well, first, please unlink the idea of VSG imposing any structure of repositories. It is up to the developer(s) to find a repository structure that best suits their needs. VSG should be transparent to that. So I'd say the question "What's the use of a submodule if it depends on the top module?" is irrelevant (in that context).
And just to recap (I already explained it), I have a distributed VSG configuration setup and that works pretty well.
And to underline that I found that setup as an adaptation of what VSG can do to what I need.

And "just don't see why you ever would want to do that.": simply because you cannot imagine that someone would want (or is forced to do) that doesn't mean that somebody actually would.

So again, I would not tie the VSG configuration to any repository structure. In my eyes you have a big potential of opening a lot of room for many issues that may arise in various constellations. Rather keep it simple, clean and straight forward.

jeremiah-c-leary, 2022-10-10:
Hey everyone,

So it has been 2 years since we last discussed this and I am getting to the point where I am contemplating the next major version.

With respect to this issue, I believe the management of configurations and files should be done outside of VSG.  That would give the user the greatest flexibility in their use of VSG.

VSG should only do what the user tells it to do.  If VSG was searching for specific filenames up directories and pulling in configurations without the users knowledge, I could see it leading to confusing the user.

Given what I just stated, default configurations would not be implemented.

Any feedback on my comments above would be appreciated.

I will plan to close this issue after November 5th if I do not hear anything.

Thanks,

--Jeremy

staerz, 2022-10-11:
Hi @jeremiah-c-leary,

after 2 years also I have learned a few things.

Yes, "the management of configurations and files should be done outside of VSG", I agree.

Depending on a particular setup of a (VHDL) project, it might though be useful to store its VSG configuration with it in some (set of) configuration files. That's what I do for my project, and then I "load in" this configuration via the `--configuration` command line parameter.

In the meantime, I have learned a lot about code quality and VSG certainly is a tool to indicate code quality.
Allow me to put some links here that I came across my journey on that subject:
- [GitLab Code Quality analysis plugins (and configuration)](https://docs.gitlab.com/ee/ci/testing/code_quality.html#using-analysis-plugins)
- [Available default analysis plugins by CodeClimate](https://docs.codeclimate.com/docs/list-of-engines)

In the spirit of consistency (or at least similarity), I encourage you to have a look at what the folks from CodeClimate are doing. You might end up promoting VSG to be one of the tools they support by default.
This might though mean that you should look into how other tools already on that list handle configuration. Some tools don't have any configuration, some have a little, but VSG certainly has a lot.

Having understood some basics and knowing how I set it up for my project, I would envision to go with a local configuration file in a (VHDL source code) repository. That configuration file, or maybe even configuration directory, could be tested by VSG and be taken as its source for configuration. (And the assumption here would be that VSG should only look in the `pwd` to find that configuration ... the CI tools basically do that: Navigate to the repo's root directory and execute code analysis tools from there.)

Now if a provided command line argument overrides this configuration or not is up to discussion (follow popular examples, I'd say).

A recommendation though would be to not worry about more complicated repository structures: That would be handled by the owners of the projects: As long as it's very precise (, straight forward, consistent) and clear how VSG gets its configuration, people will manage to make that happen for their particular projects.

In summary: Yes, reading in a default configuration from the present working directory would be a useful feature for VSG. This should make it most consistent with other analysis tools.

jeremiah-c-leary, 2024-03-10:
Evening,

Well...it has been about 4 years and I am going to retire this PR and the associated issue.  In that time I have come to the conclusion that VSG should not be responsible for managing configurations.  It will accept them, but the decision on which to use should be determined by the user.

There has not been any traction on this issue or the linked PR,.  Therefore I will assume others agree.
 
Regards,

--Jeremy

closed 2024-03-10 by jeremiah-c-leary