jeremiah-c-leary/vhdl-style-guide #354: Updates to output

jeremiah-c-leary, 2020-03-18:
In a gitter discussion with @nfrancque , it was suggested to add column headers to the output.  This should make it clearer what each column represents.  After reviewing the output, I was wondering if there are other changes that should be made:

- Report configurations
- Point user to documentation on read the docs
- It would be awesome to have the rules "link" to the read the docs website
- Remove the Total Rules Checked and Total Failures?
- Remove the phase reporting?

How can we make the output more user friendly?

jeremiah-c-leary, 2020-03-21:
I found this and wonder if it would work for links:

echo -e '\e]8;;http://example.com\aThis is a link\e]8;;\a'

If I can get this to work I can link the failure to the readthedocs website

although it would always link to the latest ones

nfrancque, 2020-03-21:
https://docs.pylint.org/en/1.6.0/output.html describes a little bit of how pylint does their output configuration.  Evidently you can write whatever string with the variables provided and messages come out in that format.

Would it be possible to have a command line option for getting error descriptions?

jeremiah-c-leary, 2020-03-21:
> Would it be possible to have a command line option for getting error descriptions?

Something like man page?

jeremiah-c-leary, 2020-03-21:
What do you think of something like this:
```mono
   $ vsg -f PIC.vhd 
   =======================================================================================
   File:  PIC.vhd
   =======================================================================================
   Phase 5 of 7... Reporting
   Total Rules Checked: 208
   Total Violations:    523
   ----------------------------+------------+---------------------------------------------
     Violation                 |  line(s)   | Solution
   ----------------------------+------------+---------------------------------------------
     comment_002               |         51 | Ensure proper alignment of comment with previous line.
     comment_002               |         52 | Ensure proper alignment of comment with previous line.
     comment_002               |         54 | Ensure proper alignment of comment with previous line.
     comment_002               |         55 | Ensure proper alignment of comment with previous line.
     comment_003               |     76-256 | Inconsistent alignment of comments within process.
     sequential_005            |      87-93 | Inconsistent alignment of "<=" in group of lines.
     sequential_005            |    102-103 | Inconsistent alignment of "<=" in group of lines.
     sequential_005            |    105-108 | Inconsistent alignment of "<=" in group of lines.
     sequential_005            |    110-113 | Inconsistent alignment of "<=" in group of lines.
     sequential_005            |    115-118 | Inconsistent alignment of "<=" in group of lines.
     sequential_005            |    120-124 | Inconsistent alignment of "<=" in group of lines.
     sequential_005            |    129-133 | Inconsistent alignment of "<=" in group of lines.
     sequential_005            |    160-161 | Inconsistent alignment of "<=" in group of lines.
     sequential_005            |    173-174 | Inconsistent alignment of "<=" in group of lines.
     comment_002               |        183 | Ensure proper alignment of comment with previous line.
     sequential_005            |    225-226 | Inconsistent alignment of "<=" in group of lines.
     sequential_005            |    238-239 | Inconsistent alignment of "<=" in group of lines.
   ----------------------------+------------+---------------------------------------------
   NOTE: Refer to online documentation at http://... for more information.

```

jeremiah-c-leary, 2020-03-21:
> https://docs.pylint.org/en/1.6.0/output.html describes a little bit of how pylint does their output configuration. Evidently you can write whatever string with the variables provided and messages come out in that format.

That is interesting.  There is currently a `syntastic` option, but this would be generic and you could define it anyway you wanted.

nfrancque, 2020-03-21:
I like that format a lot.  Maybe change solution to issue though?  Or change the descriptions so that they all sound like suggested solutions.

jeremiah-c-leary, 2020-03-21:
I think we would need to scrub the solutions so they sound like solutions:
```mono   $ vsg -f PIC.vhd 
   =======================================================================================
   File:  PIC.vhd
   =======================================================================================
   Phase 5 of 7... Reporting
   Total Rules Checked: 208
   Total Violations:    523
   ----------------------------+------------+---------------------------------------------
     Violation                 |  line(s)   | Solution
   ----------------------------+------------+---------------------------------------------
     comment_002               |         51 | Align comment with previous line. 
     comment_002               |         52 | Align comment with previous line. 
     comment_002               |         54 | Align comment with previous line. 
     comment_002               |         55 | Align comment with previous line. 
     comment_003               |     76-256 | Align comments within process.
     sequential_005            |      87-93 | Align "<=" within the group of lines.
     sequential_005            |    102-103 | Align "<=" within the group of lines. 
     sequential_005            |    105-108 | Align "<=" within the group of lines. 
     sequential_005            |    110-113 | Align "<=" within the group of lines. 
     sequential_005            |    115-118 | Align "<=" within the group of lines. 
     sequential_005            |    120-124 | Align "<=" within the group of lines. 
     sequential_005            |    129-133 | Align "<=" within the group of lines. 
     sequential_005            |    160-161 | Align "<=" within the group of lines. 
     sequential_005            |    173-174 | Align "<=" within the group of lines. 
     comment_002               |        183 | Align comment with previous line.
     sequential_005            |    225-226 | Align "<=" within the group of lines. 
     sequential_005            |    238-239 | Align "<=" within the group of lines. 
   ----------------------------+------------+---------------------------------------------
   NOTE: Refer to online documentation at http://... for more information.

```

jeremiah-c-leary, 2020-03-21:
I am hoping I can make the violations link to the online documentation, although it would always link to the latest version of the documentation.

nfrancque, 2020-03-21:
Can you have it point to a local copy of the documentation that ships with the package?

jeremiah-c-leary, 2020-03-21:
hmm....It is all in restructured text and it would be installed in the `site-packages` directory.  Would you want someone looking in there for the documentation?

nfrancque, 2020-03-21:
No, but I'm sure there's a way to generate the html/pdf/whatever at build time and provide a copy of that with the package.  So you would be linking to the same "webpage", it just happens to be stored on the user's PC.

nfrancque, 2020-03-21:
Alternatively, it does look like readthedocs provides some option for versioning your generated documentation.

closed 2020-04-01 by jeremiah-c-leary