jeremiah-c-leary/vhdl-style-guide #355: Allow user to augment existing command line arguments

jeremiah-c-leary, 2020-03-18:
In a gitter discussion with @nfrancque it was proposed to add the ability for end users to augment command line arguments to support scripting.

This would involve being able to request the command line arguments from VSG similar to how VUNIT works.

This issue will discuss how this should work, as it is still a little unclear to me, and whether to implement it.

nfrancque, 2020-04-03:
So at the time of suggesting this I was unaware the config file could also contain command line arguments. That is an interesting approach I have not seen before, and means this is definitely less valuable than with vunit. 
If that is the intended approach I am not sure most users are aware of that ability, is it featured in the examples? 

The only thing this feature would then add is if a user wanted to also do some extra work in front of vsg. You can imagine any number of use cases - gathering up what files to give vsg, pre or post processing vsg results, adding optional simplified vsg arguments - in general building domain specific tool variants on top of vsg.

jeremiah-c-leary, 2022-10-09:
Hey @nfrancque ,

I am going through old issues and was wondering if this is still something you would like to have.

--Jeremy

jeremiah-c-leary, 2022-10-23:
Hey @nfrancque ,

I am going to set to defer and schedule to close it after Dec 31st.

--Jeremy

closed 2022-12-30 by jeremiah-c-leary