jeremiah-c-leary/vhdl-style-guide #365: Add commit sha to version output.

m-kru, 2020-03-29:
It would be nice if `vsg -v` / `vsg --version` could return sha commit. Sometimes I Install current master, and there is no easy way to make sure I use the version from the commit I wanted.

jeremiah-c-leary, 2020-03-30:
I have not heard of a sha commit.  What would the output look like and how do we get the sha value?

m-kru, 2020-03-31:
I mean commit SHA-1 checksum. Each commit in git has its distinct SHA1. The output would be like:
`VHDL Style Guide (VSG) version 1.2.0 (fd767e)`

Not all characters are needed as, 6 would be enough.

jeremiah-c-leary, 2020-06-30:
@m-kru,

This seems like a chicken and egg problem.  I can't set the SHA for the release(tag) until I create the tag.  At which point I can't update the SHA for the version command.

I could do this as part of the deploy to pypi, but it would not work if you pulled down master.

I found this command:
```mono
git show-ref -s 1.11.1
31d146a61f4bcdd00e9743f376379eebd69fb0ad
```
which I could parse for the pypi release.
There is also
```mono
git show-ref --heads
02e99db82f8fd7245a31e8d7be9b83fc56120b39 refs/heads/issue-350
9839f4a164414c05432ea9b7c1afe624235abd06 refs/heads/issue-432
c778fc8a756103ea83f4b20f9ea46c21a6978f30 refs/heads/master
a340c76170147a1d1cef9d613d3aefe7544da6f4 refs/heads/release_1.0_support
```

To implement this would VSG need to issue git commands?

staerz, 2020-08-03:
Hi @jeremiah-c-leary,

so the command that you should imply for your version is `git log --pretty=%h -n 1`.

This will simply give you the the (short version of the) commit SHA.

This will work as long as you have VSG as a repo.

I don't know how the pypi release works. But at least for your local install script, you could simply run this git command and make the output part of the `--version` output at the time when running the script.

And I'm certain that you'd find an answer in the web about it.

jeremiah-c-leary, 2020-08-03:
maybe I could make it part of a `try:`, `except:`.  Then it could work for either case.

jeremiah-c-leary, 2020-08-04:
Well...there are times when I tend to over think things.  The `try: except:` was the solution I was looking for, but never thought of until now.

I just pushed a commit with this feature added.

closed 2020-08-04 by jeremiah-c-leary

staerz, 2020-08-04:
> Well...there are times when I tend to over think things. The `try: except:` was the solution I was looking for, but never thought of until now.
> 
> I just pushed a commit with this feature added.

So you "try" the git command, and "except" prints the version that you hard-code?

jeremiah-c-leary, 2020-08-04:
I `try:` the git command using `subprocess`.  If you are in the VSG Git working copy, then the command will work and I print out the SHA.  If you are outside the Git working copy, then the command fails and I suppress printing anything about the SHA out.

I either case I print the hard coded version.

staerz, 2020-08-04:
Well, this won't do the job, I'd say, as I'll usually call VSG from where it is installed.
I do though install it from a local clone using the python install script (rather than pip install...).

So coming back to the checken-egg problem (where my opinion is that the egg was there first without a doubt), I'd propose the following procedure that works in my head (but I'm not a python developer):
- your install script will have to run `git describe` or `git log --pretty=%h -n 1` before actually copying anything to the system. It could then write a file that contains the version info. You then "pythong compile" VSG with that info and then install it to the system.
- the delivery to pip should contain the similar process (and here is where I'm not sure how it works ...) that you first extract the SHA, then incorporate it into your "--version" command and then only provide it to pip.