jeremiah-c-leary/vhdl-style-guide #373: New rule proposal.

m-kru, 2020-04-06:
In case of block comments developers often use special lines for the first and last lines. There are 2 popular versions.
```vhdl
--=================================================================
-- Some
-- block
-- comment.
--=================================================================
``` 

```vhdl
-------------------------------------------------------------------
-- Some
-- block
-- comment.
-------------------------------------------------------------------
``` 

Add rule detecting such lines and forcing particular style. The allowed characters should be a list, so that both or even more styles are possible to be defined.

jeremiah-c-leary, 2020-05-03:
Agree on this rule.  I have also seen something like:
```vhdl
--+-------------------------
--|  Some
--|  block
--|  comment.
--+-------------------------
```

Pushing to release 2.1.0

staerz, 2020-05-12:
I have prepared a feature request in our project actually.

Note that this is based on us using doxygen, but one might want to support other documentation tools like sphinx, too:

## Feature Request Summary

I request to add a rule to vsg which evaluates the correct format of the `------------------------------<-    80 chars    ->------------------------------` comment, **when NOT used inside the doxygen `code` environment**

The format checked should be
```vhdl
------------------------------<-    80 chars    ->------------------------------
--! <any text>
--! <any number of further commented lines
--------------------------------------------------------------------------------
```

where:
1) the first line must be an exact match
1) the last line must be an exact match
1) there is at least one line between the first and the last
1) there may be any number of lines between the first and the last
1) no indentation is allowed other than given in the above example
1) any line in between the first and the last must not exceed 80 characters in total, except for lines containing the doxygen keyword `image`
1) any line in between the first and the last must start with `--! `

- [ ] Add a rule to LASP vsg which evaluates the correct format of the `--!------------------------------<-    80 chars    ->------------------------------` comment, **when used inside the doxygen `code` environment**:

```vhdl
--!------------------------------<-    80 chars    ->------------------------------
--!-- <any text>
--!-- <any number of further commented lines
--!--------------------------------------------------------------------------------
```

where:
- the same rules 1) to 5) apply with the following modifications:
8) any line in between the first and the last must not exceed 83 characters in total
1) any line in between the first and the last must start with `--!-- `

## Feature Details

If the comment is used but it doesn't have the right format, show a rule message to fix it.
It could be added an automate fix. (optional)

## Purpose, Expected Benefit

The proper use of guidelines on code

jeremiah-c-leary, 2020-05-13:
@staerz is the `--<- 80 chars ->--` actually in the comment or is it just to indicate the line is 80 chars long?

@m-kru Here are my initial thoughts on handling this:

```mono
--AB
--C
--DE
```
Where A is a string parameter which can accept None
Where B is a string parameter
Where C is a string parameter which can accept None
Where D is a string parameter which can accept None
Where E is a string parameter

So if you wanted to recreate your first example:

```mono
--=================================================================
-- Some
-- block
-- comment.
--=================================================================
```
```mono
A = None
B = '===========================...'
C = None
D = None
E = '===========================...'
```
To recreate @staerz second example:
```mono
A = !
B = '---------------------------<- 80 chars ->-----------------------------'
C = !--
D = !
E = '-------------------------------------------------------------------------'
```
there should also be a line limit attribute.

staerz, 2020-05-14:
Dear @jeremiah-c-leary,

yes, it is grown habit, so our code really has this `...-<-    80 chars    ->-...` literally.

The idea is clear: I wanted to mark the 80 characters line limit - and the last character of this line is at position 80.

Now, as you say, some people might think that 80 is to few and rather define 100 or 120 or anything else - leading to the statement that the length should be configurable.
That opens another issue: If you make it configurable, then you should check that the given configuration is consistent - i.e. that "--AB" matches the configured maximum length.
So VSG should foresee to complain on invalid configuration.

And yes, your proposal with A, B, C, D and E seem good to me!
We do understand that there should be any text possible after C, but again checking against the line limit (now taking into account the length of C itself).

Note though that one could want multiple flavours of this - i.e. I need 2 flavours, one for pure VHDL highlighting, one for doxygen highlighting in VHDL. And note that there might be different documenation tools that use some different character - the "!" here is what doxygen uses (and I don't know others).

@jeremiah-c-leary, in your very last example, I think you have like 3 or so "-" too much in E: The length of E and B should match (given that A and D do). So self-consistency checks to implement would be that
- length of (A+B) + 2 = configured max length, 
- length of (D+E) + 2 = configured max length,

jeremiah-c-leary, 2020-05-14:
I was thinking about the `<- 80  chars ->` and how to implement it.  It is really just a string embedded in a series of dashes.  The `80` has meaning, but it can be handled with a max length attribute.

To make this more generic we could do the following:

```mono
--ABCB
--D
--EFGF
```

Where A is a string parameter which can accept None, or empty string
Where B is a character which will be repeated
Where C is a string parameter which can accept None, or empty string
Where D is a string parameter which can accept None, or empty string
Where E is a string parameter which can accept None, or empty string
Where F is a character which will be repeated
Where G is a string parameter which can accept None, or empty string

There would also be an alignment flag for C and G.

So to recreate the following example:
```mono
------------------------------<-    80 chars    ->------------------------------
--! <any text>
--! <any number of further commented lines
--------------------------------------------------------------------------------
```
We would set the following:
```mono
A = ''
B = '-'
C = '<-  80 chars  ->'
D = '!'
E = ''
F = '-'
G = ''
B.alignment = 'center'
G.alignment = 'center'
```

to recreate this example:
```mono
--!-[begin block comment]--------------------------------------------------
--! <any text>
--! <any number of further commented lines
--!----------------------------------------------------[end block comment]-
```
We would set the following:
```mono
A = '!'
B = '-'
C = '[begin block comment]'
D = '!'
E = '!'
F = '-'
G = '[end block comment]'
C.alignment = 'left'
G.alignment = 'right'
width = 80
```

staerz, 2020-05-14:
Hi @jeremiah-c-leary,

in my opinion, that is a bit more complicated than needed. Let the user decide what the top and bottom line should be entirely and restrict the needed configuration to a minimum.

So I'd rather support your previous suggestion.

m-kru, 2020-05-17:
I have created this issue some time ago. https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/374 . I think restricting comment lines to exactly 80 chars makes sense. Especially if it is needed for doxygen. Making comment only lines length configurable and then adding some kind of dependency in other rules like: *If comment line equals 80 is becoming a bit too complicated for such simple things as comments.

@staerz What about doxygen comments that are indented?
Do they require
```
------------------------------<-    (80 - indentation spaces) chars    ->------------------------------
```
If yes, then I think it is great idea to enforce fixed (80 chars) length of comment only lines.

staerz, 2020-05-18:
Hi @m-kru,

well, you won't make yourself friends with restricting comment lines to (non-negotiable) 80 characters!

As I initially stated, for us it is a habit, that I just try to define a rule to check against, but you will certainly find developers who would want 120 characters or any other number.
But certainly, it would be good to have a rule to check against that, so I'm fully supporting #374!

An important note though is on point 6 of the feature request summary stated above: "except for lines containing the doxygen keyword image": If you look at [the documentation](http://www.doxygen.nl/manual/commands.html#cmdimage), the syntax is

```
\image['{'option'}'] <format> <file> ["caption"] [<sizeindication>=<size>]
```
(for VHDL, it would be
```
--! @image['{'option'}'] <format> <file> ["caption"] [<sizeindication>=<size>]
```
, of course).

If now your file path is very long, or anything else makes this line exceed 80 chars, you're doomed (since doxygen doesn't allow linebreaks in this command [to my knowledge])

So you certainly want this explicit exception. Maybe one could generalise it for any doxygen command found in the line ...

And as said earlier, I don't know for other documentation systems...

So I would go back to the proposal of
```
--AB
--C
--DE
```
Adding another parameter for the accepted line length and some except char (which I'd then set to be '80' and '@', respectively in my case).

m-kru, 2020-06-02:
@staerz can you provide some doxygen documentations?

staerz, 2020-06-02:
Dear @m-kru,

the official doxygen documentation can be found on doxygen's website: https://www.doxygen.nl/manual/ and in particular for VHDL here: https://www.doxygen.nl/manual/docblocks.html#vhdlblocks

Doxygen comments just start with "--!" and it doesn't matter if you have whitespace in front of it.
Hence, by design, you may (and are highly encouraged) to indent the comments with the source code - so this would follow the already existing commenting rules defined by VSG, there's nothing new.

The purpose of this issue (and the discussion here) is to add an exception (/new rule) for an (intentionally) unindented comment block as described above.

Let me know if you need more info.

staerz, 2020-12-19:
Hi @jeremiah-c-leary,

and advancing on this one would also be marvelous ... :wink:

jeremiah-c-leary, 2020-12-20:
Hey @staerz and @m-kru ,

I just pushed three rules to check block comments.  If you get a chance, could you test it out and give me any feedback.

Thanks,

--Jeremy

staerz, 2020-12-21:
Hi @jeremiah-c-leary,

thank you for adding this!

Allow me a few recommendations for the current implementation:
[Doxygen Block Comments](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_block_comments.html#doxygen-block-comment):
- The header and footer should still only start with `---` and not `--!` (in the default case). Usually you don't want Doxygen to print a row of dashes in your documentation. If you compare with my initial feature request, you'll see that I distinguish the usual case and the case where you explicitly want Doxygen to also print the comment block in the documentation. The difference is in how to start the comment. But I agree, VSG has no way to figure out which one to use/is meant to be used (so you cannot fix it) in a particular occurrence. By now I have though reworked my thinking and came to the conviction that the full header in Doxygen wouldn't be needed at all since the comment should be done properly using the indentation and that's it (so if we drop it from the feature request, than, again you can actually fix it).

[Block Comment Rules](https://vhdl-style-guide.readthedocs.io/en/latest/block_comment_rules.html)
- You use "description" here in the header [and footer]. IMHO this is misleading and gives the false impression that this text can be used to give an actual description, although it is a fixed string upon configuration. I'd rather use "header" and "footer" as you also do in [Doxygen Block Comments](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_block_comments.html#doxygen-block-comment) - and maybe use also `<` and `>` over `[` and `]` for consistency.

staerz, 2020-12-21:
More feedback after I tried it out:

I'm using the following configuration:
```json
    "block_comment_001":{
      "disable": false,
      "header_left_repeat": "-",
      "header_string": "<-    80 chars    ->",
      "header_right_repeat": "-",
      "max_header_column": 80,
    },
    "block_comment_002":{
      "disable":false,
    },
    "block_comment_003":{
      "disable":false,
      "max_footer_column": 80,
    },
```
I don't quite understand why the proposed change is shifted:
```
  block_comment_001         | Error      |        475 | Change block comment header to : -------------------------------<-    80 chars    ->-----------------------------
  block_comment_001         | Error      |        485 | Change block comment header to : -----------------------------<-    80 chars    ->-------------------------------
```
Note that neither are actually correct: The first is shifted left by 1, the second right by 1.
Let me know if you see any mistake in my configuration.

Further I noted that any 3 subsequent lines that are commented now are falling under the block comment. I don't think that this is what should happen. I give you an example:

In a port definition (that is commented aligned with the port as should be) you may have:
```vhdl
    --! @brief PCIE SMB clock:
    --! wired but not required for normal operation
    --! @todo golden top has inout, example has input
    pcie_smbclk         : in    std_logic;
```
You'll notice that these are doxygen comments.
So in most cases, I would not want this to be a block comment (which is fully left aligned, i.e. starting at char position 1).

There are a few block comments (for Doxygen mainly) where you would like to add a full description of something, there you would break out of the expected indentation and simply dump the doc. That is then when these block comments come in.

So I think you should update your "block comment detection" to have a comment starting at char 1 and char 1 only. Anything else is a regular comment (that is checked via `commen_010` (and others)).

jeremiah-c-leary, 2020-12-21:
> Further I noted that any 3 subsequent lines that are commented now are falling under the block comment. I don't think that this is what should happen.

My detection logic was to look at the first comment in the group and check if the character after the `--` was not an alphanumeric character.  So in this case the `!` would be detected.  I can put in logic to ignore the `!`

> I'm using the following configuration:

Are the two lines at the same indent level?

My math maybe off.  I think I only tested with comments at an indent of 0.

staerz, 2020-12-21:
Well, that's in deed a tricky question.

Concerning the indentation, I gave the example above, and yes, the indentation is correct, all at the same level. And it's all regular Doxygen comments.

Now again, for me the distinction between a block comment and a block of comments is that the block comment starts at char 1 while the block of comments starts at the current indentation.
With this distinction, there is obviously still the ambiguity when you're at indentation level 0. I think in this case you can do very little. The only correct behaviour that I could think of is to not raise an error but an info (pointing out that possibly using a block comment couldn't be check with certainty).

I thought about it and find it very difficult to come up with a rule to distinguish these 2 via the character that follows the `--`.

Maybe another idea: A "block comment" is a block of its own, so should be standing out, meaning to have a blank line above and below.

Now I think what you could do is the following: Don't start thinking about rules to request a blank line above or below the block comment, but to use that feature as identification: If you encounter "blank line; at least 3 lines of comment; blank line" than you treat it as a block comment.

Regular comments would always be in the line directly above the code - and some developers may find it more appropriate to leave one blank line above the comment (remember, that's what we came to conclude for the rules for "blank line above ... ": A comment is allowed, but then the line above the comment (or comments!) should be blank.

Now if we define a block comment as "blank; (at least 3) comments; blank", then we would also be compatible with all the other rules.

What do you think?

jeremiah-c-leary, 2020-12-21:
I just pushed an update for documentation and fixed the centering algorithm.  Could you run it again and check if the same error occurs?

jeremiah-c-leary, 2020-12-21:
A block comment is a purely arbitrary construct, so it would be up to us to define it.

I have seen block comments used for describing processes:
```vhdl
--+------------------------------
--|  Purpose:  Performs math functions on ....
--|  ....etc...
--+------------------------------
```
also to separate states in a state machine
```vhdl
--+------------------------------
--|  IDLE
--+------------------------------
when IDLE =>

--+------------------------------
--|  READ_FIFO
--+------------------------------
when READ_FIFO =>
```
and before instantiations.  We also can not forget a header block.

> Now if we define a block comment as "blank; (at least 3) comments; blank", then we would also be compatible with all the other rules.

At the moment, I can't think of a reason why that would not work.  It is a little restrictive, but that might not be a bad thing.  It would essentially say:  any group of 3 or more consecutive comments are a block comment and must therefore follow the block_comment rules.  That would mean:
1)  Must have a header
2)  Must have a footer
3)  Must have the comment_left in rule block_comment_002

Is there a case where you would have more than three consecutive comments and not want to make it a block_comment?

staerz, 2020-12-21:
Well, the best counter argument for your proposal is anything related to doxygen that is a regular, but little longer doxygen comment:

For example:
```vhdl
  --! @brief Maximum size of a multi-register
  --!
  --! Maximum size of a multi-register.
  --!
  --! In order to define an array of (pseudo-variable-sized) multi-registers,
  --! the maximum block size must be defined.
  constant AVMM_MREGS_SIZE : positive := G_AVMM_MREGS_SIZE;
```
In order for Doxygen to recognise it properly, you have to spend an extra (Doxygen-empty) line to separate the brief from the detailed description.
The detailed description here includes a line break that is indicated by another (Doxygen-empty) line.

For me this is not yet a block comment.

In one file, where I define a few types and constants, I have this:
```vhdl
--------------------------------------------------------------------------------
--! @brief An array of the stream-type variant of the AVST interface.
--!
--! An array of the stream-type variant of the AVST interface.
--! By default, this array is of unconstrained size.
--!
--! To properly instantiate it, use another constant to constrain it, e.g.
--!
--!```
--!  avst_streams   : t_avst_streams(AVST_CNT - 1 downto 0);
--!```
--! Where `AVST_CNT` is of type positive.
--------------------------------------------------------------------------------
  type t_avst_streams is array (natural range <>) of t_avst_stream;
```

Now I agree (and this is the entire plot of this discussion) that this now is a block comment - and should have the proper header (which here I don't for my own lack of consistency (is rare, but happens :wink: )).

I'm sort of reluctant to indent all these comments to what they should be (i.e. 1 level here) as there is not much gain and I'd rather like to read the full block comment from char 1 on (also to not run into the length limitation early).

On that second example, yes, the header should be updated. And yes, I would, in agreement with my earlier proposal, opt for a blank line above the type definition (that is still compatible with Doxygen).
The ultimate alternative for that comment is to indent it and not have it as a "block comment".

Again I'm sticking to my interpretation that a "block comment" starts on the 1st character of the line (and hence escapes from the expected indentation), while a regular comment, even if longer (over multiple lines, even more than 3), sticks to the expected indentation.

Then it is up to the user to make a decision to change a regular comment section into a block comment or not.

And I think that was the full point of the idea here: The regular comment is already treated well, we simply want to add something that escapes these rules.

jeremiah-c-leary, 2020-12-21:
So is this a block comment?

```vhdl
  constant c_width : integer := 256;

--! @brief An array of the stream-type variant of the AVST interface.
--!
--! An array of the stream-type variant of the AVST interface.
--! By default, this array is of unconstrained size.
--!
--! To properly instantiate it, use another constant to constrain it, e.g.
--!
--!```
--!  avst_streams   : t_avst_streams(AVST_CNT - 1 downto 0);
--!```
--! Where `AVST_CNT` is of type positive.

  type t_avst_streams is array (natural range <>) of t_avst_stream;
```
because:
1)  It starts at the first column
2)  There are more than three consecutive comments
3)  There is a blank line above them
4)  There is a blank line below them

and the block comment should then have the header and footer put on:
```vhdl
  constant c_width : integer := 256;

-------------------------------------------------------------------
--! @brief An array of the stream-type variant of the AVST interface.
--!
--! An array of the stream-type variant of the AVST interface.
--! By default, this array is of unconstrained size.
--!
--! To properly instantiate it, use another constant to constrain it, e.g.
--!
--!```
--!  avst_streams   : t_avst_streams(AVST_CNT - 1 downto 0);
--!```
--! Where `AVST_CNT` is of type positive.
-------------------------------------------------------------------

  type t_avst_streams is array (natural range <>) of t_avst_stream;
```
This could get tricky though, assume we have the following:
```vhdl

--! @brief An array of the stream-type variant of the AVST interface.
--!
--! An array of the stream-type variant of the AVST interface.

  type t_avst_streams is array (natural range <>) of t_avst_stream;
```
how do I know the first line is a header or a comment?
```vhdl

-------------------------------------
--! @brief An array of the stream-type variant of the AVST interface.
--!
--! An array of the stream-type variant of the AVST interface.

  type t_avst_streams is array (natural range <>) of t_avst_stream;
```
I could see something like this happening
```vhdl

----------< 80 columns>----------
-------------------------------------
--! @brief An array of the stream-type variant of the AVST interface.
--!
--! An array of the stream-type variant of the AVST interface.

  type t_avst_streams is array (natural range <>) of t_avst_stream;
```
So I would need some way to detect if a header was attempted.  Maybe that only matters if this rule attempts to fix the block coment.

staerz, 2020-12-22:
Hi @jeremiah-c-leary,

yes, the first would then be a block comment and yes, it should then be formatted as you suggest.

> This could get tricky though, assume we have the following:

Well, only if you intend VSG to fix that, which I think is not necessary here (seen the complexity of the problem).

But, for detecting a block comment header I thought of any (successive) repetition of characters which are not alpha-numericals.
So if the first line is `--.*[:alnum:]\2*` (I hope the regex is correct) then it's a header and should be converted into what is configured to be the header.
For the footer it could be similar.

If you consider the header detection a bit to ambiguous, than I think it's no problem to have no fix but simply have VSG complain.

jeremiah-c-leary, 2020-12-22:
> So if the first line is --.*[:alnum:]\2* (I hope the regex is correct) then it's a header and should be converted into what is configured to be the header.

How about this for a definition of a possible header:

`<whitespace><comment><first_header_character><second_header_character>`

Where
* whitespace =  any number of whitespaces at the beginning of the line (configurable to enforce only left aligned)
* comment = "--"
* first_header_character = any non-alphanumeric character excluding whitespace and exclamation marks to support Doxygen
* second_header_character = any non-alphanumeric character excluding whitespace

So the following would be attempted headers:
```vhdl
--+----
--|------
---------
```
while the following would not:
```vhdl
--!  Doxygen comment
--Some comment
-- Some comment
--| Some comment
```

staerz, 2020-12-22:
I tend to say that this sounds complete.

I'm not sure to get your distinction between `<first_header_character>` and `<second_header_character>`. I think you mix up a character class: alphanumericals are [0-9a-zA-Z]. I think what you mean is basically [:print:] excluding ...

Am I correct? If so, then I agree, that should work.

jeremiah-c-leary, 2020-12-22:
> I think what you mean is basically [:print:] excluding ...

I think the regexp would almost be: `^\s*--[:punct:][:punct:]` where the first `[:punct:]` is missing the exclamation mark.  I am not sure how to write that in regex.

staerz, 2020-12-22:
You can use negative look ahead:

`grep -P '^\s*--(?!\!)[[:punct:]]+'` does the trick for me.

jeremiah-c-leary, 2020-12-22:
Hey @staerz ,

I had some updates to the header/footer detection logic.  I also added a `allow_indenting` option.  It defaults to `True`, so you will need to set it to `False`.

--Jeremy

staerz, 2020-12-22:
Hi @jeremiah-c-leary,

I'll test your new updates shortly. While trying to get there (I still have a bunch of `comment_010` to fix) I found this interesting edge case:

The line
```vhdl
-- vsg_off comment_010
```
itself throws an error on `comment_010` :rofl:

I think this should not happen, right?

The use case here is that I really simply have a block of code that I temporarily commented out, and I would know that `comment_010` would complain, so I want to (temporarily) mask that.

The lines following that line are then not complained on, which is good.

I'm not sure how you'd have to catch this edge case on your side...

jeremiah-c-leary, 2020-12-22:
> itself throws an error on comment_010

That is funny.  I think I might know how to fix it.

staerz, 2020-12-22:
And another edge case concerning `comment_010`:

Somewhere deep in a process:
```vhdl
            -- Reset if required
            if rst_mregs_rw_i(i) = '1' then
              mregs_rw(i) <= MREGS_RW_INIT(i);
            -- Write particular data in address memory upon write request (if found)
            -- for the sub-part of the multi-register
            elsif avmm_master_i.write = '1' then
```

`comment_010` complains about the 2 last comments as it want it to be indented by 1 level. I think the comment indentation logic needs to look ahead instead of considering the current indentation level!

I had found this already for libraries, but didn't mention it...

jeremiah-c-leary, 2020-12-22:
The rule `if_032` should adjust the indent level for those comments.  I will look into it.

staerz, 2020-12-22:
Hi @jeremiah-c-leary,

so it's not only the `if_032`, it seems to be a general thing:

In an architecture I have this:
```vhdl
  gen_delay : case d_depth generate

    -- No delay of #sig_i
    when 0 =>
      delay_reg(0) <= sig_i;

    -- Delay #sig_i by one clock cycle
    when 1 =>

      proc_singledelay : process (clk)
      begin
        if rising_edge(clk) then
          if en_i = '1' then
            delay_reg(0) <= sig_i;
          else
            delay_reg(0) <= delay_reg(0);
          end if;
        end if;
      end process;

    -- Delay #sig_i by multiple clock cycles
    when others =>
```
Guess what:
```
  comment_010               | Error      |        106 | Indent level 5
  comment_010               | Error      |        113 | Indent level 5
  comment_010               | Error      |        118 | Indent level 5
```

staerz, 2020-12-23:
Hi @jeremiah-c-leary,

alright, so apart from these, `block_comment_001` is off by 1 when checking the header. It is suggesting
```vhdl
-------------------------------<-    80 chars    ->-----------------------------
```
while it should be
```vhdl
------------------------------<-    80 chars    ->------------------------------
```
(offset by 1 to the right).

I'm using the following configuration:
```json
    "block_comment_001":{
      "disable": false,
      "header_left" : "-",
      "header_left_repeat": "-",
      "header_string": "<-    80 chars    ->",
      "header_right_repeat": "-",
      "max_header_column": 80,
      "allow_indenting": false,
    },
```

jeremiah-c-leary, 2020-12-23:
Hey @staerz ,

I believe I fixed the comment_010 issue with my latest push.

I am working on the center alignment.

jeremiah-c-leary, 2020-12-23:
Hey @staerz ,

I pushed an update to fix the centering.  When you get a chance, could you check it out.

Thanks,

--Jeremy

staerz, 2020-12-23:
Hi @jeremiah-c-leary,

I confirm that now the header centering is correct, at least for my case, no more complaints about it.

I'll see about the remaining few complaints that I get and see where they come from and report back to you.

staerz, 2020-12-23:
Hi @jeremiah-c-leary,

I would suggest that we close this issue as it is addressed. There are now a few remainders on the comments in general, but that mostly concerns `comment_010` - should I re-open #502 or do you want me to open a new ticket?

jeremiah-c-leary, 2020-12-23:
Hey @staerz ,

I would open a new ticket, thanks.

closed 2020-12-23 by jeremiah-c-leary