jeremiah-c-leary/vhdl-style-guide #374: New rule proposal - maximum block comment length.

m-kru, 2020-04-06:
Restricting maximum line length in VHDL files is usually not so simple as lines in VHDL tends to be much longer than in programming languages. However it makes sense to restrict the maximum length of block comments (probably 80 should be the default) as it makes it easier to look through the code or read the documentation using `cat` or `less` commands.

```vhdl
----------------------------------------------------------------
-- This
-- is
-- block
-- comment.
----------------------------------------------------------------

-- This
-- is
-- also
-- block
-- comment.
```

jeremiah-c-leary, 2020-05-03:
This seems like a reasonable rule.

Pushing to release 2.1.0...

jeremiah-c-leary, 2020-12-20:
Hey @m-kru ,

I think I have this covered when I updated issue #373.  This is also covered by the `length_001` rule which checks the length of any line.

If you get a chance, could you test out the new `block_comment` rules?

Otherwise, I will close this as part of the 3.0.0 release.

Thanks,

--Jeremy

jeremiah-c-leary, 2020-12-26:
Hey @m-kru ,

I am going to close this as issue #373 was recently closed.

--Jeremy

closed 2020-12-26 by jeremiah-c-leary